JMRI/JMRI #6540: Deprecations in RailCom code

bobjacobsen, 2019-02-09:
After PR #6493 was merged, [Jenkins](http://builds.jmri.org/jenkins/job/Development/job/Deprecations/) has started showing [warnings](http://builds.jmri.org/jenkins/job/Development/job/Deprecations/1927/warnings4Result/) about the use of deprecated methods in RailCom:

[image]

(I haven't confirmed that it was #6493 that did it, it was just about that time)

Should those methods really be deprecated?  If so, should we update this code?

pabender, 2019-02-09:
That depreciation was intentional.

The RailCom interface was duplicating a part of the LocoAddress interface.  Also, with refactoring the RailCom interface so that it extends the new addressed IdTag interface I want to remove the gerDccLocoAddress in favor of the getLocoAddress specified by the new interface.

I’ll get those warnings cleaned up.

pabender, 2019-02-10:
These have all been [fixed now](
http://builds.jmri.org/jenkins/job/Development/job/Deprecations/1929/warnings4Result/).

closed 2019-02-10 by pabender