JMRI/JMRI #6552: Type checked memory variables

danielb987, 2019-02-11:
I would like to add type checked memory variables, DigitalIOMemory, AnalogIOMemory and StringIOMemory, there these are type checked. A DigitalIOMemory would only be allowed to have the values ON and OFF, an AnalogIOMemory would only be allowed to have an decimal value and a StringIOMemory would be allowed to have any string value but no other type of object.

I'm thinking of adding three tables for these and changing the Tools->Tables menu so that "Memory Variables" is moved to a sub menu "Memory Variables" together with the tables for "Digital Memory", "Analog Memory" and "String Memory".

The background for this is that I'm working on a more generic version of Logix, called NewLogix, which will support DigitalIO, AnalogIO and StringIO, and there I would like to be able to store the result of expressions in a type safe way.

Any thoughts about this?

bobjacobsen, 2019-02-13:
A next-generation Logix sounds great!  One suggestion:  don’t call it “new Logix”.  That kind of naming leads to complications and un-clarity in lots of places.  Please call it Logix2 or LogixNG or some other completely-novel word.

As to typed memories: Type-checking isn’t really a user value. They just want memories to store stuff.  Adding complexity to the user interface (where do I have to look for that memory?) is viewed as a cost by users, even though developers would view it as a simplification. And even developers have (slowly) learned that generics and templates, where available in the language, are a better technical solution than having lots of CompoundNoun class names.

Type information is still valuable, but users are going to consider it a _configuration_ item, not a type. So I recommend:
- add a visible content-type indication to the Memory class
- that type is set when something is assigned to the Memory value
- trying to set a value of an incompatible (in the Java sense) type once the type has been set is an error; you have to null the value out first.
- make that type appear in a column in the table

Then, if you want to do more, we could add custom (i.e. by type) editors and cell-drawers for as many types as we want.

Bob

> I would like to add type checked memory variables, DigitalIOMemory, AnalogIOMemory and StringIOMemory, there these are type checked. A DigitalIOMemory would only be allowed to have the values ON and [...]

pabender, 2019-02-14:
> On Feb 13, 2019, at 11:46 AM, Bob Jacobsen <notifications@github.com> wrote:
> As to typed memories: Type-checking isn’t really a user value. They just want memories to store stuff. Adding complexity to the user interface (where do I have to look for that memory?) is viewed as a cost by users, even though developers would view it as a simplification. And even developers have (slowly) learned that generics and templates, where available in the language, are a better technical solution than having lots of CompoundNoun class names.

I totally agree with everything @bobjacobsen says above.

Also, on type checking and the user experience, remember that many of our users are already writing scripts in jython, which is a dynamically typed language, so the users don’t ever think about typing.

Paul

KenC57, 2019-02-15:
However, I find that a number of times I've got to write scripts with a
try/catch thing as if it is an object I have to treat it one way and if it
is text, another way instead. Usual case is that by default it seems, JMRI
tries to keep all memory values when you do a save. One memory feature I'd
love is an option to say it is NOT to have its value saved with the panel.
It would save a lot of initialization scripts flushing things.

danielb987, 2019-02-15:
@KenC57 
Are you using the same memory for different things? @bobjacobsen suggested that the type of the memory is set when the memory is set, and if the same memory is used for different things, this change will break current scripts.

Adding a checkbox "Save value in panel file?" should be simple. But before doing that, I need to be sure of what I should do.

Maybe it's better to let LogixNG to handle the error checking than adding new features to Memory. But adding a configuration to not save the value of a particular memory to the panel file would be nice.

danielb987, 2019-02-18:
A suggested solution is in PR #6572.

danielb987, 2019-02-19:
I'm closing this issue as a result of the discussion in PR #6572.

Thanks for all the comments.

closed 2019-02-19 by danielb987