JMRI/JMRI #6561: Request better sizing on the DecoderPro CV pane

bobjacobsen, 2019-02-13:
From a JMRIusers post by "Jay":

> To the developers, would it be possible to increase the size of CV tab window? I programmed a Soundtraxx decoder today, scrolling through all of the CV's was a pain. The window took up less that 1/2 of the tab space. Just a suggestion as to my experience.

devel-bobm, 2019-02-16:
I don't suppose we have any specific information on how "Jay" would like the huge number of available CVs to be displayed which would be "better" than what we have?  

The most obvious change would be to increase the number of visible rows.  But blindly doing that will fall apart on smaller displays.  

The next most obvious change would be to try to figure out the number of rows to display, dynamically at run-time.  While this might be simple (speculation!) on non-Windows machines, it most definitely becomes problematic on Windows machines where font size reported by JAVA is not necessarily an accurate representation of how Windows renders the font.  This problem occurs when users like me have forced Windows to present fonts at larger sizes account too-small text on high-DotsPerInch displays (and perhaps for other reasons, too).

bobjacobsen, 2019-02-16:
> I don't suppose we have any specific information on how "Jay" would like the huge number of available CVs to be displayed which would be "better" than what we have?

As a start, the CV table scroll area could resize itself to fill the available area in the pane. 

When that was written, a long time ago, Java 1.2 Swing had very limited tools for controlling sizing. The CV area ended up constrained, because it would otherwise force all the planes to be too large. There are better tools available now.

closed 2021-04-14 by stale[bot]