JMRI/JMRI #6568: LaunchJMRI not always working

mattharris, 2019-02-15:
Since 4.15.3, the change introduced in #6504 seems to have broken launches on 64-bit machines with only a 32-bit Java installed.

Further details in user forum post https://groups.io/g/jmriusers/message/156948

bobjacobsen, 2019-02-15:
Would it be helpful to add something to the release note in the meantime?  Perhaps

> Note to 64-bit Windows users:  This release may fail to start if you have only the 32-bit version of Java installed.  Please install the 64-bit version of Java to use this JMRI version.  We're working on getting this fixed in a future version of JMRI.

mattharris, 2019-02-15:
Already done - see https://github.com/JMRI/website/pull/307

mmosher5501, 2019-02-15:
Perhaps also add to release note:
Work around for 64bit Windows OS with 32 bit Java
In the shortcut Target box
change "C:\Program Files (x86)\JMRI\LaunchJMRI.exe" apps....
to "C:\Program Files (x86)\JMRI\LaunchJMRI.exe" /32bit apps....

mattharris, 2019-02-15:
Something along those lines is coming in https://github.com/JMRI/website/pull/308

Thanks for the suggestion.

mattharris, 2019-02-20:
Just an update to say that I'm most of the way through getting something ready for this, just needs some further testing in a suitable test environment which I need some time to setup.

bobjacobsen, 2019-03-05:
A jmriusers note that might have useful info:

> (All installations involved complete un-installation of all previous instances of JMRI and  removal of all relevant JMRI folders)
> 
> I re-installed JMRI 4.14 - all shortcuts work perfectly.
> I installed JMRI.4.15.2ish+jenkins+20190105T1348Z+Rcc7e9c5 (build supplied by Paul Bender 5/1/19 relating to issue with Lenz Compact)- all shortcuts work perfectly.
> I installed JMRI.4.15.3+R7949343 (version suggested by Steve Todd ref. small RPi query) -  shortcuts do not work.
> I installed JMRI.4.15.4ish+jenkins+20190301T0348Z+R954342d (as suggested in your message 1/3/19) -  shortcuts do not work.
> I have tried the same procedures as 1-4 above on a notebook running WIN7. - All shorcuts work perfectly.
> 
> Also, all shortcuts seem to be working fine when installed on RPi. 
> 
> The issue seems to be with WIN10.
> 
> Checking properties for the shortcuts in all iterations of JMRI (1-4 above), the relevant targets are shown as  "C:\Program Files (x86)\JMRI\LaunchJMRI.exe" apps.PanelPro.PanelPro and "C:\Program Files (x86)\JMRI\LaunchJMRI.exe" apps.gui3.dp3.DecoderPro3.
> 
> Opening shortcuts for 3 & 4 above in a sandbox, as I said before, I get the " No command line parameter..." error message.
>

closed 2019-03-09 by bobjacobsen