JMRI/JMRI #6569: Extended accessory address encoding

Johan-Dahlberg, 2019-02-15:
Hi,

I'm not sure whether this really is an JMRI issue or not - I guess it mostly is a matter of
poor specification by NMRA...

I have built my own signal decoder to be controlled as a DCC extended accessory.
The DCC packet decoding is implemented as in NMRA S-9.2.1 section "Extended Accessory Decoder Control Packet Format" (which leaves everything up to the interpreter...) and S-9.3.2 section 6.1.
In the latter it is specified that the a simple linear addressing should be used "1 0 A10 A9 A8 A7 A6 A5 0 0 A4 A3 A2 0 A1 A0 1 0 0 0 0 X X X X X".

However, as I have tried controlling my signal from JMRI (using the Z21 loconet driver) I find that JMRI encodes the address similar to the basic accessory address, i.e. "1 0 A7 A6 A5 A4 A3 A2 0 0 !A10 !A9 !A8 0 A1 A0 1".

Now to the point - which address encoding method is "correct"?
1. Am I the only one to interpret the extended accessory decoder address
according to S-9.3.2, i.e. should I obey the address encoding as JMRI provides today?

or 

2. Should perhaps JMRI offer a way to provide both the existing address encoding method,
 plus the one described in S-9.3.2?

Best regards,
/Johan

bobjacobsen, 2019-02-15:
JMRI's code for this can be found in `jmri.NmraPacket` class (See the [Javadoc](http://jmri.org/JavaDoc/doc/jmri/NmraPacket.html) and [code](https://github.com/JMRI/JMRI/blob/master/java/src/jmri/NmraPacket.java) )  There are a number of formats implemented there to talk to equipment found on various layouts. All of them are consistent with NMRA S-9.2.1 (and S-9.3.2 has no bearing, as the part you quote was just dicta added without approval long after various equipment was out in the field).  JMRI was built to work with what people have, and historically has rejected the NMRA's political attempts to push manufacturers around.

The format you've chosen is not the most common implementation.  But it is available as an alternative in JMRI. In the JMRI user interface to define a Signal Head, check the box to say you want to use an "OffSet Address":

[image]

There's more information on that in the "DCC Signal Decoder" section [on this page](http://jmri.org/help/en/package/jmri/jmrit/beantable/SignalAddEdit.shtml)

I don't think there's anything here for JMRI to change, except perhaps to add additional documentation.

pabender, 2019-02-15:
@Johan-Dahlberg one caveat to this....

I have not verified that raw DCC packets sent to the Z21’s LocoNet connection make it to the rails.

I know the trick that sometimes works on XpressNet systems for sending raw packets doesn’t work on the Z21.

( in my opinion, a major deficiency in the XpressNet protocol is the lack of a formal raw DCC packer command... )

dheap, 2019-02-16:
@Johan-Dahlberg As @bobjacobsen has pointed out, JMRI [code](https://github.com/JMRI/JMRI/blob/master/java/src/jmri/NmraPacket.java)  and [Javadoc](http://jmri.org/JavaDoc/doc/jmri/NmraPacket.html) provides for two interpretations of the extended accessory decoder addressing scheme, both for [Ops Mode Programming](http://jmri.org/help/en/html/apps/DecoderPro/CreateDecoderAdvanced.shtml#accessoryops) and for [Signal Head creation](http://jmri.org/help/en/package/jmri/jmrit/beantable/SignalAddEdit.shtml), the choice of encoding format being governed by the "Offset Address" checkbox.

See for example [accSignalDecoderPkt](http://jmri.org/JavaDoc/doc/jmri/NmraPacket.html#accSignalDecoderPkt-int-int-) and [altAccSignalDecoderPkt](http://jmri.org/JavaDoc/doc/jmri/NmraPacket.html#altAccSignalDecoderPkt-int-int-)
- The NCE Light-It signal decoder uses the interpretation selected by checking "Offset Address".
- Other signal decoder brands differ (others may offer examples).

@bobjacobsen has shown a screenshot of the Add Signal Head dialog. Here is a screenshot of the Single CV Programmer dialog:
[image]

But (as @pabender has pointed out) JMRI preferentially uses raw DCC packets for all Signal Operations and there may be an issue sending these when using the Z21 LocoNet driver.

Johan-Dahlberg, 2019-02-16:
Thank you all for your thoughts on this matter!

@bojacobsen, well, the problem is that NMRA S-9.2.1 can be interpreted almost anyway you like,
as there is does not tell how each bit in the address field shall be assigned.
S-9.3.2 definitely does relate to this issue, as it clearly specifies how each bit is to be interpreted,
at least in case when the extended accessory decoder is to implement Railcom for feedback.
Well, a signal mast doesn't have much information to send back to the command station,
except to tell that it has accepted the command...

And the format specified by S-9.3.2. is not among the supported ones, as far as I can read from
the source code. And the alternative encoding is only available for "signal heads", not "signal masts".
Only the latter seem to make any sense on non-US layouts. (But maybe I am missing some
configuration which would add different naming to the aspects, as can be done for the signal masts)

@pabender, don't worry - the Z21 Loconet encoding seems to be correct,
at least for the few extended accessory decoder addresses I have tried.

@dheap, thanks for pointing out the encoding portions of the code. (Google helped me
find it while I was debugging my signal decoder code...)
This is the discussion I wanted to start, to hear if I was the only one ever trying to implement
the address decoding according to S-9.3.2 - and it seems I might be.

Thus, for the time being I decide to change to the interpretation of S-9.2.1 that is preferred by JMRI.

Thanks again,
/Johan

closed 2019-02-16 by Johan-Dahlberg

dheap, 2019-02-16:
@Johan-Dahlberg a couple of corrections are in order:
> And the alternative encoding is only available for "signal heads", not "signal masts".

1. It is not yet available for signal masts if you choose DCC Signal Mast Decoder. There is an open issue for that #5880, waiting for someone with the time/inclination/skills.
2. It is available for signal masts if you choose Signal Head Controlled Mast and use one or more predefined signal head(s) built using DCC Signal Mast Decoder(s).

> Thus, for the time being I decide to change to the interpretation of S-9.2.1 that is preferred by JMRI.

JMRI currently offers a choice of two different interpretations of S-9.2.1, both of which are known to be used by signal decoders "out there". If you feel there is a need for provision of yet another interpretation _(I haven't studied S9.3.2 to check if it differs from both of the options we already offer)_ you are welcome to contribute to extensions the JMRI j`mri.NmraPacket` class code and submit a Pull Request.

If you decide to contribute, life is a lot easier for all if you follow our guidelines:
http://jmri.org/help/en/html/doc/Technical/index.shtml
http://jmri.org/help/en/html/doc/Technical/gitdeveloper.shtml
and linked pages.