JMRI/JMRI #6573: Suggested Manual Revision Multiple Profiles

MikeHeintzman, 2019-02-18:
[JMRI Multi Profile Setup.docx](https://github.com/JMRI/JMRI/files/2877032/JMRI.Multi.Profile.Setup.docx)

dheap, 2019-02-19:
A couple of thoughts:
- There's some possibility of confusion between User Files Sharing (between Profiles and/or Computers) and Profile Sharing (between Computers). Either can be used or both can coexist at the same time. Maybe a reference to [Profile Directory Tutorial](http://jmri.org/help/en/html/doc/Technical/ProfileFileStructure.shtml) (or a section thereof - I can easily add some anchors to that page to allow reference to specific sections) would be helpful.
- The existing documentation does not cover the Search Paths (adding and setting default) tab. This is necessary when sharing profiles between computers (and a reference to the Sharing Profiles section of  the [Profile Directory Tutorial](http://jmri.org/help/en/html/doc/Technical/ProfileFileStructure.shtml) may be useful.
- We don't actually have any documentation of the "New" and "Copy" buttons, nor the relative merits of either.

These are just some suggestions you may want to run with. If not, I'll probably just put what you have written up and worry about additions/changes later.

Thanks again for doing this.

MikeHeintzman, 2019-02-19:
Thanks for the input. I was trying to figure out whether it was better to insert references to the Profile Directory Tutorial and the Dropbox page and another spot that escapes me, or putting references to this in those spots, without becoming too convoluted. Let me take another look, and maybe add a bit about New and Copy.

dheap, 2019-02-20:
> @MikeHeintzman without becoming too convoluted.

Agreed. I'm prone to adding too much information...

MARCFOUR, 2019-02-20:
A little clarification here for NEW Users might be in order.  Windows user.

By default JMRI does not open the 'Set Active Profile'  Selector menu. It opens the last-used profile which would be the System default profile it created (My_JMRI_Railroad).  The User must go into Preferences and Select the proper option in 'Config Profiles' to have the 'Show Profile Selector for' displayed for the seconds defined (Count down).

By default, DecoderPro and PanelPro do not act in the same manner. DecoderPro will load the last_used_profile  where as PanelPro will display the "Set Active Profile Selector" and it's My_JMRI_Railroad default for 10 seconds.

Creating a New Profile:   If the User has not selected the option to display the " User Profile Selector " in Preferences,  they will not be presented with the option to create NEW profile at start up. They would have to go into Preferences, Config Profiles and create 'NEW" from within Preferences.   A reboot is required for the newly created  PROFILE to take effect and open the Startup Wizard.  This is a bit different than going thru the 'Set Active Profile' selector, create NEW profile as it will go directly into the Startup Wizard when the users hits the "OK" button. 

Would it not be best to put the DropBox references into an Advanced setup section of this Help.

Marc

KenC57, 2019-02-20:
A couple of corrections to Marc's notes:

1. If there is only one profile, the default is to not show the profile
selector. This is true for either PanelPro or DecoderPro.

2. If there is more than one profile, the default is to show the profile
selector. This is true for either PanelPro or DecoderPro.

3. A reboot is not required for a newly created profile, a restart is
required.

-ken c

MARCFOUR, 2019-02-21:
Ken, with all due respect, I do not make these up as I go along, I test them.

1 . If there is only one profile, the default is to not show the profile selector. PanelPro, with only one profile, WILL show the Set Active Profile selector.  I tested it before writing it.

2. If there is more than one profile, the default is to show the profile selector.. IF THE OPTION HAS BEEN SELECTED to display the 'Set Active Profile Selector'.  This is true for both PanelPro and DecoderPro.

3. It depends where the profile was created. if it was from within the PREFERENCES, then a restart is required. If the profile is created within the Set Active Profile Selector, then it will go direct to the Setup Wizard and does not require a restart.  By "reboot" I really meant restart, my error.

MARCFOUR, 2019-02-21:
PanelPro has the 10 second counter and "Set Active Profile" selector by default. DecoderPro does not and
requires the user go in and change the Config Profiles option away from "Automaticaly start application with last used profile " and over to "Show Profile selector" or he will not see the "Set Active Profile" selector

KenC57, 2019-02-21:
I just tested with 4.15.3 and both begin with 'last profile selected'. I did
this by removing the JMRI directory (ok, saved mine by rename). So it was
starting fresh. DecoderPro gives the Startup Wizard, but not PanelPro.
PanelPro keeps the 'last selected', it did not change because it saw more
profiles.

Yet once there was the second profile created (via the PanelPro startup),
then I went and use DecoderPro. It popped up the profile selector. And the
'use last' had change to offering the profile selector.

Trying SoundPro, it seemed to act more like the PanelPro. But when I did a
new and got a third profile, on restart I was back to the 'last selected'.

So I'm surprised by the inconsistencies. I had thought more of the profile
managing was consistent between the startups.

MikeHeintzman, 2019-02-21:
Unless there’s a technical reason for the difference, might it not be better for common functions like this to operate consistently across the platform? I’m guessing it would make tweaking the Manuals easier.

MARCFOUR, 2019-02-21:
Ken, I was also running 4.15.3  but under Win10.  There has to be some explanation for the difference in end results.  I also deleted the User\JMRI folder (renamed to JMRI_backup)  so I started from a blank slate. 

DecoderPro would load the default profile. PanelPro would show the Profile selector.  

Preferences, Config Profiles  of DP shows Auto Start as active (selected). PanelPro: Preferences, Config Profiles show "Show profile selector" .  Even if I create multyple profiles they still react the same until I manualy change the  DecoderPro, Config Profiles to "Show Profile Selector".

I am not trying to be a pest, just trying to add to the Suggested Revisions Mike is planning. I look at it with the eyes of a new user.  I find it very hard to come up with a clear game plan, My hat is off to anyone who works/keeps at this till the end. 

Marc

KenC57, 2019-02-21:
I think one difference is the startup wizard only works from DecoderPro. So
that gives a considerable different path when it starts with a blank profile
and the wizard gets triggered. Part of me thinks the wizard would be a help
for the PanelPro users too.
-ken c

MikeHeintzman, 2019-02-21:
I don’t mind plugging away at it, but I think that someone(s) with more technical expertise than me needs to determine if a) this differentiation is the way it’s supposed to be; b) that’s the way to leave it; or c) should it be changed to be one way or the other in a future revision.
My background, among other things, is writing user manuals for technically challenging software. I can try to write it either way, but my objective was to write it as simply as possible for new users. I like the idea of having a Basic and Advanced setup, and maybe using a format similar to the one used in the OperationsPro manual.

MikeHeintzman, 2019-02-24:
I made some changes incorporating all of the comments. I tried to add a section on the differences in the default startups. Please take a look and see if it says what it's supposed to say. Thanks.
[JMRI Multi Profile Setup 2.docx](https://github.com/JMRI/JMRI/files/2898330/JMRI.Multi.Profile.Setup.2.docx)

MARCFOUR, 2019-02-25:
downloaded and browsed.. After three or more system crashes of this aging computer, I gave up.  There were a few niggly bits I found .  There are a few more items that end up in a PROFILE  apart user files, throttle and roster. Things like Panels, backup Panels, any Tables, Imported  decoder definitions,  Decoder Index rebuilds.  I don't know if you want to mention these.  

Also the bit about .JMRI suffix. That is added by JMRI. They way it is state it is as if the user must add it. That will just confuse and end up with .JMRI.JMRI .  

I also found another DEFAULTS variation if a user starts a new setup via PanelPro. versus DecoderPro.   The Setup Wizard throws another curve at the defaults mentioned.  This will be another cause for confusion for a new user. 

I would take the UPGRADE warnings and group them together in last paragraph.

MikeHeintzman, 2019-02-25:
I incorporated your suggestions, other than the additional Default variation. I'm not sure I can add it without making it even more confusing. Until somebody takes a crack at the code to make how Profiles are handled between the apps that may just have to go under an unexpected 'Feature', 
My old desktop computer died a few weeks ago necessitating taking my hard saved train money to buy a new one and try to save what I could from it (and even though the backups were on a separate drive they decided to commit seppuku in sympathy). When the weather here in Oregon clears up I intend do do a Terry Pratchett on it and tun it down with a steam roller.

[JMRI Multi Profile Setup 2.docx](https://github.com/JMRI/JMRI/files/2902611/JMRI.Multi.Profile.Setup.2.docx)

MikeHeintzman, 2019-02-25:
I hit the wrong button. How do I Unclose this thing?

closed 2021-04-14 by stale[bot]