JMRI/JMRI #6578: In 4.15.3+R7949343  Edit->Preferences->Defaults cannot be modified

gbglacier, 2019-02-19:
In 4.15.3+R7949343 When one goes to Edit->Preferences->Defaults with CPP++ interface  I cannot change ANY "Default" settings.
I double checked and these were changeable in 4.12. 
This also appears to be the case with the QSI programmer interface I tried.
I am hoping this is the right place to report this.

bobjacobsen, 2019-03-30:
@gbglacier Is this still happening in current development builds?  There's been some work in this area.

gbglacier, 2019-04-02:
Hi Bob,
Yes, I just reconfirmed this, I cannot change any default setting for a DCC++ configuration under JMRI.4.15.4+R0d44598 but can under JMRI.4.12+Rb6a9bb1.
'Hope this helps. Best regards, Geoff

closed 2021-04-14 by stale[bot]