JMRI/JMRI #6579: Deadlock when starting Panel Pro

rm5248, 2019-02-19:
When attempting to start PanelPro, I get a deadlock in Java waiting for a monitor.  Stack dump attached.
[stack.txt](https://github.com/JMRI/JMRI/files/2882398/stack.txt)

I noticed this on the test release(4.15.3) and in current master.

rm5248, 2019-02-20:
Note: I have just seen this issue with Decoderpro 3 as well.  It didn't happen for a while, and then once I added a new profile it started happening.  Now that it is happening, I can't get it to run properly(even removing the .jmri folder doesn't fix it).

bobjacobsen, 2019-02-20:
Looking at it, but without being able to recreate it this is going to be slow.

Can you confirm that you're using Java 8 181-b13?

rm5248, 2019-02-20:
I'm using Debian's build of OpenJDK: 8u181-b13-2~deb9u1.  I have not tried with the Oracle JDK at all, but that is something that I can test as well if needed.

java version output:
openjdk version "1.8.0_181"
OpenJDK Runtime Environment (build 1.8.0_181-8u181-b13-2~deb9u1-b13)
OpenJDK 64-Bit Server VM (build 25.181-b13, mixed mode)

bobjacobsen, 2019-02-21:
Thanks for the info.  I see where the problem is coming from. So far, the only way I have found to fix it will delay startup, in some cases (like people with slow railroad computers) significantly.  So I'd like spend some time coming up with a better fix.

Issue #3678, which doesn't come from the same cause, recommends trying the Java 9 JDK.  (You could also try Java 11 by now)

svatoun, 2019-02-25:
The stacktrace shows that `init prefs` thread illegally accessess the UI, while only EDT thread may manipulate with the UI hierarchy.  No matter which JDK is used, the code which performs UI operations from non-UI thread is broken and will deadlock or eventually corrupt data, since most of the lightweight components are not thread-safe in any way.

bobjacobsen, 2019-02-25:
It's a bit more subtle than that.  You'll notice that both the main and init prefs threads are working to construct the TabbedPreferences window.  

It is generally OK to _construct_ a Swing/AWT Component off the Swing/AWT thread. It's only once it's first been accessed by Swing/AWT, usually but not always when first set visible, that it can no longer be safely accessed off the Swing/AWT thread.

The problem here is that they _deadlock_ doing construction, as both are working int the same Component tree in the Preferences window.

bobjacobsen, 2019-03-03:
@rm5248 Could you try the development build in http://builds.jmri.org/jenkins/job/Development/job/Builds/3226/ or later? I think the changes in #6621 should fix your problem, and those are included in that test build.  Thanks.

rm5248, 2019-03-10:
@bobjacobsen It seemed to work at least the first time, so Panelpro did at least come up.  However, I then went and tested it a bit and found another deadlock after I added a (non-existent) Xpressnet connection.  I've attached the newest stack trace(which also includes all the debug output as printed out in netbeans).
[stack.txt](https://github.com/JMRI/JMRI/files/2950080/stack.txt)

I also couldn't find the build on Jenkins to download, which is why I ran it through Netbeans.

PR #6621 (2019-03-03): Fix startup deadlock in preferences
This is intended to fix #6579, Deadlock when starting PanelPro.

As preliminary matters, it includes:
- Move the `TabbedPreferences*`classes from `apps.gui3` to a new `apps.gui3.tabbedpreferences` package to localize their mutual content.   This required import/classname changes in a number of other classes, some of which were in the `jmri` tree:  That indicates a [packaging problem](http://jmri.org/help/en/html/doc/Technical/IntroStructure.shtml) that can be addressed some other time.
- Improved the error handling in several places if classes and/or service [... cut]
closed 2021-04-14 by stale[bot]