JMRI/JMRI #6581: DCS52 (Zephyr Express) Support

Jonny-vdv, 2019-02-20:
Digitrax has released a new version of the Zephyr DCC Starter Set. This controller is not yet supported in JMRI. Is this currently being worked on for the next update?

devel-bobm, 2019-02-20:
JMRI does not currently have "explicit" support for DCS52.  At some time in the future, JMRI will likely include an explicit configuration for the DCS52.  It is unclear when this will be integrated, as it depends on a developer's level of interest, availability of hardware, and availability of development time.

However, any JMRI release with support for the PR3 and the DCS51 will successfully work with the DCS52 in either of two JMRI modes.

- If interested only in decoder programming via JMRI and the DCS52, configure the "Connection" for a "System connection" of "LocoNet PR3" connection with a "Command station type" set to "PR3 standalone programmer".  This setup allows JMRI to use the DCS52 "Programming" connections to program decoders, and may be used even when the DCS52 is operating as a command station to control trains.  In this mode, JMRI cannot communicate with "LocoNet", so JMRI cannot operate trains, configure LocoNet accessories, or communicate with sensors and turnouts and signals.

- If interested in JMRI operating trains and/or configuring LocoNet accessories and/or working with LocoNet-connected "panels", configure a "Connection" for a "System connection" of "LocoNet PR3" connection with a "Command station type" set to "DCS51".  This setup allows JMRI to communicate with LocoNet.  And JMRI can communicate with the "programming track" for decoder CV programming.

devel-bobm, 2019-03-01:
The next test release of JMRI should contain updates which allow a JMRI "connection" to specify the "connection type" as
- PR4
- DCS240 USB interface
- DCS52 USB interface.

For each of these, the user may configure the connection as working with some command station, or working only with the programming track (i.e. as a "Standalone programmer"), or working with a "Standalone LocoNet".

As such, that forthcoming JMRI Test Release will no longer require specification of PR3 as the "interface type".

If you would like to try out the new features, you may install one of the "intermediate" Development builds, which can be found at [http://jmri.tagadab.com/jenkins/job/Development/job/Builds/](url) .  Any build at or after build number 3218 should include the related code changes.

Jonny-vdv, 2019-03-01:
Thanks for the information. I have my layout set up using the DCS51 option, which is fine for running locos, but it will be nice to get the full functionality from JMRI.

closed 2019-03-01 by Jonny-vdv