JMRI/JMRI #6582: JMRI 4.15.3  Preferences, Unable to delete 2nd Connection

MARCFOUR, 2019-02-20:
Preferences. I created a second Loconet Connection to test (Loconet2) in a Profile.  Rebooted JMRI.
After testing tried to delete 2nd connection to remove Loconet2 via the RED X.  

Preferences screen hangs. Unable to exit Preferences screen. Unable to kill Preferences screen. Unable to terminate DecoderPro. Unable to terminate DecoderPro via the task bar. 

Had to go into WIn10 task Manager and kill  JAVA binary as Task manager shows no activity for the JAVA binary and  DecoderPro all entries, Preferences.

Reboot JMRI  and still unable to delete the 2nd connection.

Tried to delete the 1st connection, same result. Preferences screen hangs.

Marc

MARCFOUR, 2019-02-20:
The Session log shows 
" ptionhanldler.UncaughtExcaptionHandler. Thread has stopped : Loconet transmit handler "  for each attempt to delete connection.

devel-bobm, 2019-02-21:
I have seen a possibly-related problem: If I have a functioning profile, and try to change its "system connection" setting (say, from "LocoBuffer-USB" to "PR3"), JMRI locks up as soon as I select the new "system connection" setting.  

Here's the last two lines in the log:

```
2019-02-20 21:04:15,328 PanelPro.PanelPro                     INFO  - Main initialization done [main]
2019-02-20 21:04:22,753 ptionhandler.UncaughtExceptionHandler INFO  - Thread has stopped: LocoNet transmit handler [LocoNet transmit handler]
```

I have not been able to find any hints by looking at the Netbeans debugger thread trace when the problem occurs.  (I am new to that feature, and I am unfamiliar with what I should expect to see for the various JMRI and Java threads at the time when the lockup occurs.)
 
@bobjacobsen, it seems that this problem is related to the `loconet.LnPacketizer.dispose()` method.  I can prevent the JMRI lockup if I comment-out the whole `LnPacketizer.dispose()` method.  I've tried a small amount of fiddling within the `LnPacketizer.dispose()` method without any improvement.

devel-bobm, 2019-02-21:
Further info - the problem remains if I comment-out the `xmtThread` portion of `LnPacketizer.dispose()`.

The problem disappears if I comment-out the `rcvThread` portion of `LnPacketizer.dispose()`.

The problem disappears if I only comment-out the code:
```
            try {
                rcvThread.join();
            } catch (InterruptedException e) { log.warn("unexpected InterruptedException", e);}
```

bobjacobsen, 2019-02-21:
I've got a bypass for the problem that's being seen (unable to delete).

Unfortunately, that fix doesn't really fix the underlying issue as it leaves the port still in use.    That in turn doesn't allow you to delete and recreate a connection without having to restart in between.

Fixing that in a reliable way is going to be much harder. I'm not sure it even makes sense to try, as we'll probably just generate new ways to disappoint users.  It might be better to just figure out a way to disable (and tooltip) controls so that people have to restart after deleting connections before adding any.

danielb987, 2019-02-21:
One possibility is that then a user clicks "Delete", the program shows a dialog that tells the user that in order to delete, the program must be restarted. And that that dialog box has the buttons "Cancel" and "Delete connection and restart PanelPro".

icklesteve, 2019-02-21:
See #6303 
Whenever a connection is changed, it goes unstable and isn't immediately obvious to the user. 
Even if user clicks Cancel, connection is changed.
Would support in this area, ie working dialogue / forced restart.

KenC57, 2019-02-21:
Only issue I see in forcing the restart is making sure the user has the
option to save anything else they were doing, like the panel file. Since we
seem to be wary of trying to track changes, maybe prompting them to consider
doing the save before fiddling in the preferences since that could result in
a restart.

bobjacobsen, 2019-02-24:
It seems likely that requests are just going to get piled onto this, so I'm going to back off from working on it.

bobjacobsen, 2019-03-30:
Is this still happening in current development builds?  There's been some work in this area.

MARCFOUR, 2019-03-30:
4.15.5 does not display the " deleting second connection " error.
 
What I have noticed, and in this dev release also, is the 2 connections ending up on the same device. I have some examples of changing  the system connections ( Ex: invert Loconet1 and Loconet2) and the end result is the "Edit Connections" window popping up and both connections end up on same device.  I get around this by first deleting a connection instead of an out right change as a  change fails.  At first I suspected a typo on my part but further testing showed I could replicate the issue. And Dev Rel  4.15.5 shows the same outcome. 

I have screen captures of the PREFERENCES SETUP, changes and end result.

closed 2021-06-16 by stale[bot]