JMRI/JMRI #6586: Why the subtle differences between DecoderPro and PanelPro

MARCFOUR, 2019-02-21:
Items like Recreate Decoder Index.  In DecoderPro it is found under ACTIONS. In PanelPro it is found under DEBUG.  

Recreate Roster Index:  In DecoderPro it is in both FILE as Rebuild Roster and under ACTIONS as Recreate Roster Index. I believe these are one and the same.  In PanelPro  it is under DEBUG : Rebuild Roster and with nothing regarding this under FILE.  

Those assisting others can assume the user is running PanelPro and suggest a fix that is not to be found in DecoderPro because they are in the habit of working with PanelPro.  Example : run Recreate Roster under Debug suggested to someone running DecoderPro will simply confuse them. 

Would it not be better to have some consistency.

pabender, 2019-03-22:
The basic answer is that DecoderPro focuses on programming decoders and PanelPro does not.

The locations of things diverged when the DecoderPro3 was created years ago.  For a while we distributed both DecoderPro and DecoderPro3.  Eventually DecoderPro3 just became DecoderPro.

There are things you can't do, or can't easily do in DecoderPro that you can do in PanelPro.  This is intentional.

We used to get the question "why so many programs?" because DecoderPro and PanelPro were the same except for the buttons on the main program window.

MARCFOUR, 2019-03-22:
Thanks Paul.

closed 2019-03-22 by MARCFOUR