JMRI/JMRI #6598: Accessory decoder addressing for lights

brookerg, 2019-02-25:
Hi, 
I am using JMRI 4.14 with DCC++.
Using Geoff B's decoders as accessory decoders I have setup:--
- A decoder for 7 turnouts using the DCC++ base station configuration starting at accessory address 1.
- A decoder with accessory address of 40 for lights and setup JMRI  Tools/Table/lights.
The issue I found was when I used 40 as the start address for the lights, the JMRI traffic monitor showed the address/subaddress sent as "a 9 3 1". I was expecting it to send "a 10 3 1" Obviously no lights worked.
When I used 44 as the starting address, JMRI sent "a 10 3 1" and all lights worked.
No other tables have been setup.
How is it that address 44 (not 40 as I expected) can convert to address/subaddress 10 03?
Regards
Graeme Brooker

dheap, 2019-02-25:
Graeme,

> On 25 Feb 2019, at 5:02 PM, brookerg <notifications@github.com> wrote:
> 
> Using Geoff B's decoders as accessory decoders I have setup:--
> 
> A decoder for 7 turnouts using the DCC++ base station configuration starting at accessory address 1.
> A decoder with accessory address of 40 for lights and setup JMRI Tools/Table/lights.
> The issue I found was when I used 40 as the start address for the lights, the JMRI traffic monitor showed the address/subaddress sent as "a 9 3 1". I was expecting it to send "a 10 3 1" Obviously no lights worked.
> When I used 44 as the starting address, JMRI sent "a 10 3 1" and all lights worked.
> No other tables have been setup.
> How is it that address 44 (not 40 as I expected) can convert to address/subaddress 10 03?

Basic Accessory (Turnout) Decoders are defined to start at 1, have a 9 bit decoder address starting at 1 and a two bit channel address (range 0-3). As a result Turnout 1 has a decoder address of 1 and a a channel address of 0, while Turnout 40 has a decoder address of 10 and a a channel address of 3.

Extended Accessory (Signal) Decoders are defined to start at 1, and have an 11 bit decoder address. Because of the similarity in the DCC packet bit allocation and the lack of specificity in NMRA S9.2.1, manufacturers have interpreted it two different ways:

1) Signal 1 has a decoder address of 0 and a a channel address of 0, while Signal 40 has a decoder address of 9 and a a channel address of 3.

2) Signal 1 has a decoder address of 1 and a a channel address of 0, while Signal 40 has a decoder address of 10 and a a channel address of 3.

In JMRI we allow for this by having a checkbox for "Offset Address" in Add DCC Signal Head and the Single CV Programmer.

However JMRI uses Basic Accessory (Turnout) commands to drive Lights defined in the Lights Table.

From what you have posted regarding the JMRI traffic monitor it would seem that DCC++ may be using interpretation (2) for Accessory Decoders? That's odd.

Do you have another DCC system you can use for checking Geoff B's decoders?

Dave in Australia

brookerg, 2019-02-25:
Dave,
Thanks for your reply.
I don't have another DCC system or, in fact, even a model railroad setup. This is all being done for a busy son and young grandson and their setup.

Decoder #1 for points / turnouts has CV1 and CV9 written with  1 and 0 respectively and Arduino DCC++ has the eeprom address starting at 1 0. JMRI then sends 'T' commands. All good.

Decoder #2 for lights has "Accdec...17..led..ino" loaded from Geoff B with address 40. JMRI sends 'a' commands from the Lights table. And address/subaddress 10 / 03 is what lights the first light. (Then I use Logix to control the lights based on turnout position.)

Thats all consistent with your point #2.
But in my case, it seems that an offset address is required for the starting address.

It works but would I be better using the Signal Heads table as those signals can be positioned on a Panelpro panel for on screen control by an exuberant young grandson.

I've used the Lights table even though they are home made signals because I couldn't wrap my mind around the addressing and multitude of types in Signal Heads and, in any case, my signals are loosely based on Qld rail signalling and I couldn't find how to implement those.

Regards
Graeme Brooker

From: Dave Heap
Sent: Monday, 25 February, 5:51 pm
Subject: Re: [JMRI/JMRI] Accessory decoder addressing for lights (#6598)
To: JMRI/JMRI
Cc: brookerg, Author

Graeme,

> On 25 Feb 2019, at 5:02 PM, brookerg <notifications@github.com> wrote:
>
> Using Geoff B's decoders as accessory decoders I have setup:--
>
> A decoder for 7 turnouts using the DCC++ base station [...]

Basic Accessory (Turnout) Decoders are defined to start at 1, have a 9 bit decoder address starting at 1 and a two bit channel address (range 0-3). As a result Turnout 1 has a decoder address of 1 and a a channel address of 0, while Turnout 40 has a decoder address of 10 and a a channel address of 3.

Extended Accessory (Signal) Decoders are defined to start at 1, and have an 11 bit decoder address. Because of the similarity in the DCC packet bit allocation and the lack of specificity in NMRA S9.2.1, manufacturers have interpreted it two different ways:

1) Signal 1 has a decoder address of 0 and a a channel address of 0, while Signal 40 has a decoder address of 9 and a a channel address of 3.

2) Signal 1 has a decoder address of 1 and a a channel address of 0, while Signal 40 has a decoder address of 10 and a a channel address of 3.

In JMRI we allow for this by having a checkbox for "Offset Address" in Add DCC Signal Head and the Single CV Programmer.

However JMRI uses Basic Accessory (Turnout) commands to drive Lights defined in the Lights Table.

From what you have posted regarding the JMRI traffic monitor it would seem that DCC++ may be using interpretation (2) for Accessory Decoders? That's odd.

Do you have another DCC system you can use for checking Geoff B's decoders?

Dave in Australia

dheap, 2019-02-25:
I've set up a DCC++ Simulator connection.

Throwing Turnout 40 shows this in the DCC++ Traffic Monitor window:
```
Accessory Decoder Cmd: 
	Address: 9
	Subaddr: 3
	State: ON
```
This seems inconsistent with what I would expect, I would have expected:
```
Accessory Decoder Cmd: 
	Address: 10
	Subaddr: 3
	State: ON
```
Same results with Light 40.

If i try a DCC Signal Decoder Signal Head with address 40 (and Offset Address), then attempt to change aspect, the System Console shows multiple NPEs.

I have a DCC Packet Analyser, but no access to DCC++ hardware, so I can't check actual track packets, but I would expect them to be incorrect.

I'm reluctant to dive into the JMRI DCC++ code without having access to, or knowledge of DCC++ hardware.

dheap, 2019-02-25:
> @brookerg Decoder #1 for points / turnouts has CV1 and CV9 written with 1 and 0 respectively and Arduino DCC++ has the eeprom address starting at 1 0. JMRI then sends 'T' commands. All good. Decoder #2 for lights has "Accdec...17..led..ino" loaded from Geoff B with address 40. JMRI sends 'a' commands from the Lights table. And address/subaddress 10 / 03 is what lights the first light. (Then I use Logix to control the lights based on turnout position.)

I think Graeme is onto something there. It seems to me that the calculation iv `DCCppMessage.makeAccessoryDecoderMsg(address, activate)` is the culprit.

brookerg, 2019-02-25:
Having a spare Arduino, I setup a backup clean DCC++ system. No Arduino eeprom records, no other JMRI entries exceot for 4 JMRI tabke/light entries with address beginning at 40. JMRI still sends "a 9 3 1"
(I thought it might have been due to the main system having previous, now deleted, entries and imperfect table indexing )
Graeme

dheap, 2019-02-25:
@brookerg I've just emailed you offline, with information about a private test build of JMRI I'd like you to try. We'll bring it back to this forum when we see the results of that.

Dave

dheap, 2019-02-26:
#6601 addresses this. I don't have access to DCC++ hardware so would appreciate real-hardware testing. @brookerg has done some testing.

dheap, 2019-03-01:
This has been resolved via #6601.

closed 2019-03-01 by dheap