JMRI/JMRI #6600: Add check that every JUnit test class is run during tests

danielb987, 2019-02-25:
I have several times been biten by forgotten to add my new test to PackageTest.java, or forgotten to add mypackage.PackageTest to the parent PackageTest.java.

This raises the question whenether there may be more tests written that is not run during tests. Maybe it could be a good idea to add a test that checks that every test class is actual included in their PackageTest, and that every PackageTest is included in the parent PackageTest. Unless this is difficult or time consuming to implement.

rhwood, 2019-02-26:
I think the CI servers actually exclude the PackageTest classes and don’t use them, so it might be better to make the ant test targets run every test instead of relying on PackageTests to organize them.

danielb987, 2019-02-26:
If I forget to add a test class to PackageTest, it's not used for the coverage test. That's how I found this problem. I created a new class and a test for that class, but still had "coverage" of 0 for that class.

But I agree that it would be good if all test classes are used without the need of PackageTest.

pabender, 2019-02-26:
Travis currently ignores the PackageTest classes.  Appveyor uses them because it still uses ant.

If you want to check to verify you have all the tests in the file, you can run the alltestcoveragecheck target ( I may have the target name wrong, I am doing this from memory.. )

pabender, 2019-02-26:
We will not be using Maven on Appveyor any time soon.  Ant is our official build tool, and there is still an annoying issue that crops up every once in a while where the NCE tests fail inexplicably and every time that crops up I think about moving Travis back to ant.

pabender, 2019-02-26:
The actual ant target you want to use is “alltest-coveragecheck”.

pabender, 2019-02-26:
reading @rhwood's suggestion again, I see the request is to make ant behave more like maven.  In the past that wasn't possible, but perhaps that has now changed.  I will investigate.

pabender, 2019-02-26:
So it looks like the ant JUnit plugin can now be used to start tests based on a pattern.

There is an impact on how test results are saved, which might impact the configuration of some of the reports on jmri.tagadab.com.

bobjacobsen, 2019-02-26:
> There is an impact on how test results are saved, which might impact the configuration of some of the reports on jmri.tagadab.com.

I’m happy to handle that aspect if needed.

As to whether to make this change, I don’ t have an a-priori opinion.  If we can _entirely_ get away from PackageList files, so that we can entirely remove them, while still supporting how people currently work, then moving in this direction is probably a good thing.  

One case to consider: I use Ant to build and ./runtest.csh to test _a_ _lot_.   I often test an entire package with 

./runtest.csh jmri.jmrit.roster.PackageList

(I have some aliases defined to save some typing, but that’s the underlying mechanism)

Can we construct a way to do run subsets of the tests via pattern?  If so, that might be even more powerful that PackageList, and make the transition worth  the effort.

pabender, 2019-02-26:
> One case to consider: I use Ant to build and ./runtest.csh to test _a_ _lot_. I often test an entire package with 

> ./runtest.csh jmri.jmrit.roster.PackageList 

This is also a concern for me, because my normal mode of operation is to run the package tests for the package I am currently working on (and sometimes the parent of the package I am working on) before issuing a PR and letting the CI engines tell me if i broke something outside of the package I am working on.

pabender, 2019-02-26:
> Can we construct a way to do run subsets of the tests via pattern? If so, that might be even more powerful that PackageList, and make the transition worth the effort.

Possibly.  The mechanism allows you to specify the path to the source, we might be able to parameterize that path....

rhwood, 2019-02-26:
I know ant is preferred, but maven allows patterns on the command line (for example `mvn '-Dtest=jmri.foo.**.*Test' test` runs all tests in the jmri.foo package and sub-packages; `mvn '-Dtest=jmri.foo.*Test' test` runs all tests in the jmri.foo package _without_ sub-packages; `mvn '-Dtest=jmri.foo.FooTest#testBar' test` runs the testBar method of FooTest)

bobjacobsen, 2019-02-26:
Happy to look at Maven for local use, but am having trouble with 
```
mvn '-Dtest=jmri.jmrit.usscctc.StationTest' test
```
which (after `ant realclean` to get a consistent starting point) does some stuff and then:
```
[ERROR] Failed to execute goal org.apache.maven.plugins:maven-compiler-plugin:3.6.1:compile (default-compile) on project jmri: Compilation failure
[ERROR] cannot access jmri.util.JmriJFrame$3
[ERROR]   bad class file: /Users/jake/Documents/Trains/JMRI/projects/JMRI/target/classes/jmri/util/JmriJFrame$3.class
[ERROR]     bad enclosing method attribute for class jmri.util.JmriJFrame$3
[ERROR]     Please remove or make sure it appears in the correct subdirectory of the classpath.
```

More info, including version info, in the attached file.  If this is a trivial mistake on my part, a hint or reference to help would be appreciated.

[log.txt](https://github.com/JMRI/JMRI/files/2906648/log.txt)

pabender, 2019-02-26:
I have been able to use mvn to run the cucumber tests by themselves.  I think that was the last thing I used the mvn single target for.

I tried to run logalltest using batchtest and that failed to produce useable results... though I think I may have missed kicking out the HeadlessTest and it started executing that.

I have my local machine here running the ant headlesstest target with batchtest. now. If that turns out ok, I'll try logalltest again.

(it just happens that logalltest and headlesstest were already using the ant junit plugin....)

bobjacobsen, 2019-02-26:
Switching to JDK11 (after trying Maven 3.6, which didn't help) got me temporarily past that so I can try it.  Comparing the two methods, Maven is significantly slower and more verbose (see below).  Is there a way to create a specific maven task that's more targeted at a rapid test cycle?

```
JMRI/projects/JMRI/ % mvn '-Dtest=jmri.jmrit.BundleTest' test
[INFO] Scanning for projects...
[INFO] 
[INFO] ---------------------------< org.jmri:jmri >----------------------------
[INFO] Building JMRI 4.15.4-SNAPSHOT
[INFO] --------------------------------[ jar ]---------------------------------
[INFO] 
[INFO] --- build-helper-maven-plugin:3.0.0:timestamp-property (timestamp-property) @ jmri ---
[INFO] 
[INFO] --- jacoco-maven-plugin:0.8.2:prepare-agent (prepare-agent) @ jmri ---
[INFO] argLine set to -javaagent:/Users/jake/.m2/repository/org/jacoco/org.jacoco.agent/0.8.2/org.jacoco.agent-0.8.2-runtime.jar=destfile=/Users/jake/Documents/Trains/JMRI/projects/JMRI/target/jacoco.exec -Xmx1536m -Duser.language=en -Duser.region=US
[INFO] 
[INFO] --- javacc-maven-plugin:2.6:jjtree-javacc (default) @ jmri ---
[INFO] Skipping - all parsers are up to date
[INFO] 
[INFO] --- buildnumber-maven-plugin:1.4:create-metadata (generate) @ jmri ---
[INFO] ShortRevision tag detected. The value is '7'.
[INFO] Executing: /bin/sh -c cd '/Users/jake/Documents/Trains/JMRI/projects/JMRI' && 'git' 'rev-parse' '--verify' '--short=7' 'HEAD'
[INFO] Working directory: /Users/jake/Documents/Trains/JMRI/projects/JMRI
[INFO] 
[INFO] --- maven-resources-plugin:2.6:resources (default-resources) @ jmri ---
[INFO] Using 'UTF-8' encoding to copy filtered resources.
[INFO] Copying 2261 resources
[INFO] Copying 1 resource
[INFO] 
[INFO] --- maven-compiler-plugin:3.6.1:compile (default-compile) @ jmri ---
[INFO] Nothing to compile - all classes are up to date
[INFO] 
[INFO] --- build-helper-maven-plugin:3.0.0:add-test-source (add-test-source) @ jmri ---
[INFO] Test Source directory: /Users/jake/Documents/Trains/JMRI/projects/JMRI/java/acceptancetest/step_definitions added.
[INFO] 
[INFO] --- maven-resources-plugin:2.6:testResources (default-testResources) @ jmri ---
[INFO] Using 'UTF-8' encoding to copy filtered resources.
[INFO] Copying 366 resources
[INFO] 
[INFO] --- maven-compiler-plugin:3.6.1:testCompile (default-testCompile) @ jmri ---
[INFO] Nothing to compile - all classes are up to date
[INFO] 
[INFO] --- maven-surefire-plugin:2.20:test (default-test) @ jmri ---
[INFO] 
[INFO] -------------------------------------------------------
[INFO]  T E S T S
[INFO] -------------------------------------------------------
[INFO] Running jmri.jmrit.BundleTest
[INFO] Tests run: 4, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.325 s - in jmri.jmrit.BundleTest
[INFO] 
[INFO] Results:
[INFO] 
[INFO] Tests run: 4, Failures: 0, Errors: 0, Skipped: 0
[INFO] 
[INFO] ------------------------------------------------------------------------
[INFO] BUILD SUCCESS
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  7.797 s
[INFO] Finished at: 2019-02-26T09:24:32-08:00
[INFO] ------------------------------------------------------------------------

% ./runtest.csh jmri.jmrit.BundleTest
JUnit version 4.12
....
Time: 0.079

OK (4 tests)
```

(including an `ant tests` step increases the `runtest` time to 3.6 seconds, which is closer to the `mvn` time but still a factor of two faster, and I don't always need to recompile when working on a test cycle)

pabender, 2019-02-26:
I'm beginning to think moving to batchtest is a bad idea.

If I run all the tests without forking the JVM, i get a significant number of errors, mostly in the GUI tests.

If I fork the JVM, I still get some more errors, but the tests take a lot longer.  To give you an idea, I started headlesstest before I left for a 90 minute class, and it wasn't finished when I got back... though I did have an error which caused all of the tests to run twice (I had a typo in an exclude...)

rhwood, 2019-02-27:
Do you have the junit reloading parameter set to true (default) or false?

pabender, 2019-02-27:
@rhwood, I am sure that was set to the default.  Will have to check in the morning.

pabender, 2019-02-27:
setting reloading to false seems to help with errors when run with fork set to yes.  It does not help with runtime.  Headlesstest took over an hour to complete.

With fork set to no, more tests fail than succeed.

rhwood, 2019-02-28:
Concerning maven, the version of maven is not completely relevant (as long as its > 3.3.9) since the pom.xml defines the version of the code ("plugins") used for most of the tasks. Those need to be updated individually.

There has been a long standing conflict between maven and Eclipse generated code that I don't entirely grasp, but it seems to cause maven to do excessive compilation. I'll try to look into that again.

Concerning the original concern here, does the coverage target in ant not fail completely with "class X is not covered" or something similar (the coverage targets on the CI servers appear to be specific about that)?

pabender, 2019-02-28:
> Concerning the original concern here, does the coverage target in ant not fail completely with "class X is not covered" or something similar (the coverage targets on the CI servers appear to be specific about that)?

The ant `alltest-coveragecheck` Target will fail if the coverage targets are not met.

There is a certain amount ( however small ) of cost associated with the PackageTest files.  If I can get the ant targets to behave with the batchtest option in place I will.

pabender, 2019-03-01:
Just in case someone wants to pursue this further, I am attaching the results of 'ant logalltest' and the modified build.xml (with a .txt extension, since github is complaining...) that was used to create this.

In this case, 'ant logalltest' took 190 minutes to complete way too long for normal operation....

[testoutput.txt](https://github.com/JMRI/JMRI/files/2919808/testoutput.txt)
[build.txt](https://github.com/JMRI/JMRI/files/2919811/build.txt)

rhwood, 2019-03-02:
I'm getting runtimes for the _headlesstest_ target comparable to master with the attached 
[build.xml.txt](https://github.com/JMRI/JMRI/files/2922577/build.xml.txt), but one run got me "missing NceMessage" errors like in #4911.

Due to some strange decision on my part not to cut and paste the batchtest element from the _headlesstest_ target to the _logalltest_ target, I've been running that with duplicate tests, so don't have runtimes for that yet.

The significant difference from what @pabender posted is that I added the argument `forkmode="perBatch"` property to the `junit` element in those targets (and excluded the FailTest from running).

rhwood, 2019-03-03:
I'm getting persistent timeouts on the _logalltest_ target, which leads me to think there is some ant junit / batchtest / GUI interaction that is proving problematic, especially given the errors with thread locks in the logs, since the _headlesstest_ is running flawlessly with batchtest.

rhwood, 2019-03-03:
I found some stack overflow comments that suggest the presence of a JUnit 3 test may cause ant's junit task to run everything as a JUnit 3 test; we still have at least 183 JUnit 3 tests (based on output of `grep -r "import.*junit.*TestCase" java/test | wc -l`)

pabender, 2019-03-03:
@rhwood thanks for picking up the torch on this one.  I think this is generally the right direction to go.  

I do wish we could figure out the cause of those NceMessage related errors, which I have always found strange because it is a runtime linking complaint about one of our classes.

rhwood, 2019-03-03:
@pabender I'm working on #6619 as a precursor to returning to this.

I am beginning to think that the errors in the tests are stemming from the tests run by ant not running on the same class of thread (i.e. AWT or whichever thread runs main) as the tests run by AllTest, since the overwhelming majority seem to be from a deadlock in TabbedPreferences (which may also be a real problem per #6579).

bobjacobsen, 2019-03-03:
#6621 (from the bobjacobsen/preferences-deadlock branch) is a reworking of the synchronization of TabbedPreferences et al.  It may improve the situation with test deadlocks.

rhwood, 2019-03-14:
While working on #6619, I did find that at least one test class was not included in a PackageTest suite, and that including this test (since fixed) broke many tests that ran after it because it changed static value that did not get unloaded (and the change was not reverted in `tearDown()`). I don't know if some of the failures noted earlier were related to the mere fact that tests not normally run as part of AllTest are now being run under ant, but weren't earlier (this also points to an interesting difference between AllTest and the maven tests--changing static fields under AllTest propagates through subsequent tests, but does not propagate through the maven tests.

danielb987, 2020-05-26:
PackageTest classes is now removed so this issue is resolved.

closed 2020-05-26 by danielb987