JMRI/JMRI #6603: Access to coveralls is randomly broken

dsand47, 2019-02-26:
I get a 504 timeout trying to access the coveralls report for PR #6576.  This has been happening for several days with 2 different builds.  I can sometimes access reports for other pull requests.

I am trying to see the coverage for jmri.jmrit.ctc and its components.

pabender, 2019-02-27:
Unfortunately, there isn’t much we can do about this because we don’t run coveralls and the issue is n their end.

You do have the option of running the coverage checks yourself locally.  The `ant alltest-coveragereport` target will generate a coverage report for you on all the tests.  The report itself is a set of web pages.

dsand47, 2019-02-27:
The coveragereport option works great.  An additional advantage is that since it is runs on my local machine, I can run it as needed.  Thanks.

closed 2019-02-27 by dsand47