JMRI/JMRI #6608: Bad Javadoc links to other projects

danielb987, 2019-03-02:
I noticed that Travis gives three warnings:

```
3 warnings
[WARNING] Javadoc Warnings
[WARNING] javadoc: warning - Error fetching URL: http://static.javadoc.io/com.fasterxml.jackson.core/jackson-databind/2.8.11/
[WARNING] javadoc: warning - Error fetching URL: https://static.javadoc.io/org.openlcb/openlcb/0.7.22/
[WARNING] /home/travis/build/danielb987/JMRI/java/src/jmri/jmrix/configurexml/AbstractStreamConnectionConfigXml.java:11: warning: empty <p> tag
[WARNING] * <p>
[WARNING] ^
```

The last warning is simply two lines with `<p>` so it's easy to fix and I will do a PR for it. But I'm unsure about the other two.

The link https://static.javadoc.io/org.openlcb/openlcb/0.7.22/ gives a xml file with the message `AccessDenied`. But the link https://javadoc.io/doc/org.openlcb/openlcb/0.7.22 gives the Javadoc for openlcb. Has javadoc.io changed the links on their web page and if so, there should we change the links in our code?

danielb987, 2019-03-02:
I found the links to `static.javadoc.io` in `pom.xml`. But the question is if these links should be changed.

bobjacobsen, 2019-03-02:
The configuration for building the Javadoc is in build.xml and pom.xml

I don't know of a correct place to link to for either project.  An older version of jackson.core is available here: https://fasterxml.github.io/jackson-core/javadoc/2.2.0/  

Part of the problem is that the target link has to provide a package-list file (ala http://jmri.org/JavaDoc/doc/package-list) that the Javadoc processor can read.  In a couple places, we've had to pull those into the JMRI distribution for that to work.

bobjacobsen, 2019-03-02:
Also, see the javadoc target in build.xml for some examples and a StackOverflow link.

danielb987, 2019-03-02:
Thanks @bobjacobsen 
Both PRs are now merged so closing this issue.

closed 2019-03-02 by danielb987