JMRI/JMRI #6613: GitHub reporting security issue with lodash

bobjacobsen, 2019-03-02:
GitHub's security scanners are [reporting an issue with lodash](https://github.com/JMRI/JMRI/network/alert/web/app/package-lock.json/lodash/open) recommending that it be updated to version 4.17.11 or later.

> CVE-2018-16487
> low severity
> Vulnerable versions: < 4.17.11
> Patched version: 4.17.11
> A prototype pollution vulnerability was found in lodash <4.17.11 where the functions merge, mergeWith, and defaultsDeep can be tricked into adding or modifying properties of Object.prototype.

It JMRI, 'lodash' appears only in the web/app tree. At the top level of that, the `web/app/package.json` file specifies the version as 4.17.5:

```
  "dependencies": {
    "angular": "1.6.9",
    "angular-patternfly": "^4.15.1",
    "angular-resource": "1.6.9",
    "angular-route": "1.6.9",
    "angular-translate": "2.17.0",
    "angular-translate-handler-log": "2.17.0",
    "angular-translate-loader-partial": "2.17.0",
    "angular-websocket": "2.0.1",
    "angularjs-datatables": "^0.5.9",
    "lodash": "4.17.5",
    "moment": "2.21.0"
  }
```

We distribute a `web/app/node_modules/lodash` directory that seems (from the README.md file) to contain version 4.17.4

But there are also `package.json` files further down that refer to 4.17.4, and `package-lock.json` files that refer to all sorts of things.

It would be great if somebody who understands this area of the code could confirm that what's required is just:
- update the content of the `web/app/node_modules/lodash` directory
- update all the `package.json` files

in which case I'm happy to do the update.  Thanks in advance.

rhwood, 2019-03-02:
I’ll have a PR tomorrow.

rhwood, 2019-03-02:
Should be addressed by #6614

bobjacobsen, 2019-03-03:
@rhwood Thank you!  Seems to have fixed the `lodash` issue.

It's moved on to flagging `bootstrap`, [requesting 3.4 or later](https://github.com/JMRI/JMRI/network/alert/web/app/package-lock.json/bootstrap/open).  Is that one of the updates that has to wait, as described in #6614?

rhwood, 2019-03-03:
@bobjacobsen I don't get to see the alerts (I don't have whatever permission is required), but since the `bootstrap` package is not listed in the file `web/app/package.json` that package is brought in as a dependency of the `angular-patternfly` package, which we need to wait on as described in #6614.

rhwood, 2020-05-12:
I think this has been resolved.

closed 2020-05-12 by rhwood