JMRI/JMRI #6631: NullPointerException thrown when editing track

svatoun, 2019-03-04:
I experimented with `Edge Connectors`. I'm not sure where the data got corrupted, so I'll try to describe the seqence of operation. 
1/ I had an existing track which ended with a bumper.
2/ removed the bumper
3/ changed the endpoint at the track's end (former bumper) into an anchor.
4/ connected the edge connector with the anchor using track segment, started dragging at the edge anchor

Now when I edit track, NPE is thrown:
```
java.lang.NullPointerException
	at jmri.Path.toString(Path.java:375)
	at jmri.jmrit.display.layoutEditor.LayoutBlock.lambda$updateBlockPaths$3(LayoutBlock.java:838)
	at java.util.ArrayList$ArrayListSpliterator.forEachRemaining(ArrayList.java:1382)
	at java.util.stream.ReferencePipeline$Head.forEach(ReferencePipeline.java:580)
	at jmri.jmrit.display.layoutEditor.LayoutBlock.updateBlockPaths(LayoutBlock.java:837)
	at jmri.jmrit.display.layoutEditor.LayoutBlock.updatePaths(LayoutBlock.java:728)
	at jmri.jmrit.display.layoutEditor.TrackSegment.updateBlockInfo(TrackSegment.java:650)
	at jmri.jmrit.display.layoutEditor.LayoutTrackEditors.editTracksegmentDonePressed(LayoutTrackEditors.java:381)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
	at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:252)
	at java.awt.AWTEventMulticaster.mouseReleased(AWTEventMulticaster.java:289)
```

geowar1, 2019-03-05:
On Mar 4, 2019, at 3:02 PM, Svata Dedic <notifications@github.com> wrote:
> I experimented with Edge Connectors. I'm not sure where the data got corrupted, so I'll try to describe the seqence of operation.
> 1/ I had an existing track which ended with a bumper.
> 2/ removed the bumper
> 
Deleting an End Bumper should delete the track segment that it's attached to. Or do you mean the track segments end bumper decoration?

(It's been in my queue to re-think the UI for end bumper decorations… it makes more sense for them to be in the End Bumper's popup menu than the track segments.)
> 3/ changed the endpoint at the track's end (former bumper) into an anchor.
> 
How do you change something you just deleted?
> 4/ connected the edge connector with the anchor using track segment, started dragging at the edge anchor
> 
I can't get this far… When I remove the bumper the track segment disappears.
> Now when I edit track, NPE is thrown:
>
> java.lang.NullPointerException
> 	at jmri.Path.toString(Path.java:375)
> 	at [...]

-- 
Enjoy,
George Warner
eMail, Message, or FaceTime: <geowar1@mac.com>

svatoun, 2019-03-05:
Sorry, misleading description. Removed bumper == `Right click on track -> Decorations -> End Bumpers -> None`.
Feel free to ignore this and the other bug I've filed yesterday, I was just recording my "session" with JMRI, will try to fix those myself (and learn a little bit more along the way).

geowar1, 2019-11-04:
@svatoun I've moved the arrows and bumpers to the anchors (specifically Edge Connectors and End Bumper track segments) popup menus. See PR #6662. If you cannot duplicate this issue now please close. Otherwise post a comment to that effect here and I'll re-investigate.

geowar1, 2019-12-22:
What's the status of this? Can we close? Or is there something for me to investigate?

svatoun, 2019-12-22:
sorry, cannot reproduce now. Sorry for the inconvenience.

closed 2019-12-22 by svatoun