JMRI/JMRI #6633: UI inconsistencies in Layout Editor

svatoun, 2019-03-04:
- memory variable and signal show tooltips with names/IDs when hovering mouse over them. Turnouts and tracks do not.
- Edit vs. Properties in popup menu; should be unified for all objects
- Delete vs. Remove in popup menu

(will add as I encounter other ones)

geowar1, 2019-03-05:
On Mar 4, 2019, at 3:25 PM, Svata Dedic <notifications@github.com> wrote:
> memory variable and signal show tooltips with names/IDs when hovering mouse over them. Turnouts and tracks do not.
Tooltips can only be attached to subclasses of JComponent… like memory variable & signals. Turnouts and tracks are not subclasses of JComponent. :(
(There may be a different way to do tooltips for LayoutTracks… I'll look into it.) My "solution" was to add more properties to the LayoutTrack popup menus.
The down side is that you have to be in Edit mode… Of course that's an up side also… (I wouldn't necessarily want all that info popping up during a session.)
> Edit vs. Properties in popup menu; should be unified for all objects
Not 100% what you mean here… Most track popups start with properties… some of which you can modify… other s not. Those properties are followed by commands that can be executed on that track type (usually including "Edit" and "Delete" (or "Remove").
> Delete vs. Remove in popup menu
Yeah, not sure why both are used… or if anything is actually different between one vs the other…
It looks like LayoutTrack sub-classes all use "Delete" and JComponent sub-classes all use "Remove".
Note: JComponent is part of java/swing (not JMRI) so we can't change anything it does (or what strings it uses).
LayoutTracks sub-classes could be changed to use "Remove" instead of "Delete". Thoughts?
> (will add as I encounter other ones)
> 
Please do… my head is too close to the code and I don't always notice some of these things. ;-)
-- 
Enjoy,
George Warner
eMail, Message, or FaceTime: <geowar1@mac.com>

svatoun, 2019-03-05:
General: feel free to ignore, for some while, I should have marked the bugs as 'my backlog' somehow to indicate I am willing at least to investigate & suggest the fix (but not immediately, I've tried to play with the editor)

re.: Delete / Remove and Edit / Properties: just naming consistency. 
Strictly technically while track segment is Deleted (since it does not exist anywhere else), Memory is Removed (since it exists still in Memories table), but for "normal user" naming should be IMHO consistent. I would suggest to use Delete everywhere since, again technically, you're not deleting the Memory itself, but just a viewer on it.

The same with Properties / Edit. There may be technical differencies, but under the radar for most regular users. I don't think any one is better, but they should be the same.

closed 2021-04-14 by stale[bot]