JMRI/JMRI #6634: Changing Block user names can break Layout Blocks

dsand47, 2019-03-04:
Track components on LE panels can have a block name assigned.  This name is actually the user name of a layout block.  The user name of the layout block is also the user name of the related block object.  This establishes the link between the layout block and block.

When a block is renamed, the link is broken.  See #2599.

I have identified several possible solutions.
1. Add pre-change validation to jmri.jmrit.beantable.beanedit.BeanEditAction#renameBean before accepting the change with appropriate feedback.  This does require adding some block specific logic to the class.
2. Add a setUserName (overriding setUserName in AbstractNamedBean) to jmri.Block which does the validation with possible rejection.
3. Add a listener to block renames in jmri.jmrit.display.layoutEditor.LayoutBlock to validate the change and notify the user of potential problems.
4. Add a listener as in 3, but rename the layout block and rename all block references in all layout editor panels.

This problem also applies to **Remove User Name**.

Any comments, recommendations or other solutions?

bobjacobsen, 2019-03-05:
> Track components on LE panels can have a block name assigned. This name is actually the user name of a layout block. The user name of the layout block is also the user name of the related block [...]

Is there a specific reason that the names have to be the same?  If not, there’s a possible 5th solution:

- LayoutBlock references its associated Block via a NamedBeanHandle<Block> data member. When constructed, it finds and attaches to the Block with the same user name, but after that it references through the NamedBeanHandle member even if the Block is renamed.  (The user name of the Block is saved and loaded as needed)

dsand47, 2019-03-05:
> Is there a specific reason that the names have to be the same? If not, there’s a possible 5th solution: - LayoutBlock references its associated Block via a NamedBeanHandle<Block> data member. When [...]

I did consider a variation on that idea.  The potential issue is that from a user perspective, the block name assigned to a track component  **_is an entry_** in the block table.  The user has no direct visibility to the existence of the layout block.  The primary purpose of the layout block is to provide the track, occupied, and extra colors, along with memory variables for old style block tracking.

bobjacobsen, 2019-03-05:
> The potential issue is that from a user perspective, the block name assigned to a track component  is an entry in the block table. The user has no direct visibility to the existence of the layout [...]

Thanks!  (I’m not a LayoutEditor user, so appreciate you taking the time to explain)

Sounds like a good case for inverting the dependency:

- Have the LayoutBlock set a listener on the Block
- When the username property of the Block changes, the Block fires a notification to its listeners
- The LayoutBlock responds by renaming itself.

geowar1, 2019-03-05:
On Mar 4, 2019, at 4:39 PM, Bob Jacobsen <notifications@github.com> wrote:
> Thanks! (I’m not a LayoutEditor user, so appreciate you taking the time to explain)
>
> Sounds like a good case for inverting the dependency:
>
> - Have the LayoutBlock set a listener on the Block
> - When [...]

The LayoutBlock already has a listener… currently it's only used to update any attached memory variable.

I do see it getting called when the block assigned to the LayoutBlock is renamed:

2019-03-04 19:17:08,009 layoutEditor.LayoutBlock              INFO  - handleBlockChange(java.beans.PropertyChangeEvent[propertyName=UserName; oldValue=SF-Yard; newValue=SF-YardX; propagationId=null; source=IB37]) [AWT-EventQueue-0]

So this should be QED. I'll put it in my queue. ;-)

The problem now is that all the LayoutTracks's that use LayoutBlocks should also be notified when their LayoutBlocks are renamed. 
Probably need to:
- Have the LayoutTrack (sub-classes) set listeners on any LayoutBlocks (when they're assigned)
- When the username property of the LayoutBlock changes, the LayoutBlock fires a notification to its listeners
- The LayoutTrack (sub-classes) respond by renaming any affected LayoutBlock name variables.

It's always bugged me that LayoutBlocks are saved (inconsistently) by both reference and by name in the LayoutTrack sub-classes… perhaps all the names should be removed and only accessed indirectly via the LayoutBlock (AbstractNamedBean) getUserName method.

AFAICT the names are only used by the XML loading code before calling the setObjects method… Which may not be necessary since I changed most (if not all) of the LayoutEditor XML loading to do two-pass loading instead of using redundant XML entries or relying on the order of things in the XML file.

-- 
Enjoy,
George Warner
eMail, Message, or FaceTime: <geowar1@mac.com>

dsand47, 2019-03-05:
Here is the current scope of this project.

There are 2 situations that have to be addressed.  The Block user name is **changed** or it is **removed**.  This happens in either **BeanEditAction** for table row edits or **BeanTableDataModel** for the table row user name context menu.  If a related layout block with a use count greater than zero exists, the remove option has to be prevented, while the rename can proceed if confirmed by the user.  If a layout block does not exist, the block user name change or removal proceeds without interruption.

The block property change listener in LayoutBlock is modified to include block user name changes.  It then fires the UserName property change so that the track components are notified.

The track components are modified to listen for layout block user name changes.

The front end changes insure that the user is aware of the nature of the change that is being requested.  The back end changes insure that the displayed block name on the layout editor panel continues to match the related block by way of the layout block.

dsand47, 2019-03-05:
The layout track block references should be changed to named bean handles.  Other references, such as turnouts, signal masts and signal heads already use named bean handles.

This eliminates the need for listeners on the track components.

geowar1, 2019-03-05:
Do you mean that LayoutBlocks should be NamedBeanHandles?
Or just their references:
```
    protected LayoutBlock block = null;    // Change this...
    protected NamedBeanHandle<LayoutBlock> block = null;    // ...to this?

```

dsand47, 2019-03-05:
If you look at LayoutTurnout, the bean references (turnouts, signals, etc.) are named bean handles except for layout block.

I don't see LayoutBlock trying to refer to Block as a named bean handle since LayoutBlock is a bean and its user name has to change at the same time as the Block user name.

geowar1, 2019-03-05:
So just change the references to LayoutBlocks in the LayoutTracks sub-classes to be NamedBeanHandle<LayoutBlock>'s?

geowar1, 2019-03-05:
Hey @dsand47, I'd appreciate a code review on #6637. Thanks. ;-)

geowar1, 2019-03-11:
Fixed via PR #6637. Verify and close? TIA.

dsand47, 2019-03-12:
As of 4.15.4, block renames are propagated to the related layout block and are visible to the LE track objects.

If a block has a related layout block, attempts to remove the block user name are not permitted.  For renames, the user is notified that their LE track block assignments are also being changed.  See #6669 .

closed 2019-03-12 by dsand47