JMRI/JMRI #6635: Handing parameters in Javadoc

bobjacobsen, 2019-03-05:
This issue is for discussion of handling parameters in Javadoc; see some background discussion in PR #6628

What should our common practice be on documenting parameters in Javadoc?

How the Javadoc processor handles parameters without explicit Javadoc `@param` lines has changed over time.  Currently, with the Ant options we use for building for the web, this code example:

```
        /** 
         * Top comment for testMe(int p1, int p2, int p3)
         *
         * @param p1 comment
         * @param p2
         */
        public void testMe(int p1, int p2, int p3) {}
```

throws a warning during building
```
  [javadoc] foo.java:12: warning: no description for @param
  [javadoc]          * @param p2
```
and shows this Javadoc:

[image]

Note that `p3` doesn't appear at all!  This is a change from prior versions of the Javadoc processor.  

p1, p2, p3 are _terrible_ method names.  It's certainly better to have descriptive ones:

```
    void remove(T item)
```

seems completely self explanatory to me.  (If you want to say `item` can't be null, there are annotations for that)  In cases like that, I'm not sure that adding 

```
    * @param item The item to remove
```
wouldn't be all that better than committing it _except_ that Javadoc wouldn't otherwise show anything about `item`.  So there's some value in adding the blah `@param`.  But things like:
```
  * @param address - address
```
(which also results in duplicate - characters) seems to actually take away from the quality of the Javadoc; they just look dumb to me. Perhaps other people feel differently.

I wish that there was a way of getting back the old Javadoc processor. Perhaps somebody has written a custom processor, or there's a magic flag that does it, but I don't know of one.

Absent that, what's the best approach? What do we recommend people do as they touch the code?

- One thing we could do is to do a systematically pass to put in `@param foo ?` defaults, then turn on checking in CI to prevent new undocumented parameters. We could then eventually drive the current number to zero (Randall has already done a lot to reduce the number, but there are quite a few left that would need to be updated one-by-one) Would that be a good thing?

- `% grep -r "@param[ ]*[a-zA-Z01-9]*[ ]*-" java/src/ | wc` finds 298 `@param` lines with a "-" after the variable name.  We could update those, and then add a CI scan for that. (The regular expression also has to get a bit smarter, as it'll miss some things, but this was a first attempt)

I care more about having a consensus, so that people aren't _undoing_ what others _are_ _working_ _on_, rather than any specific approach.  So I hope we can find one.

pabender, 2019-03-05:
@rhwood mentioned in #6628 that the build is already set to error if the number of warnings exceeds a threshold.

If we set the threshold lower, that will help prevent adding new parameters that are undocumented. without documenting the existing parameters with something meaningless.

This is a bit like our testing thresholds which are in place not to prevent new untested ( or undertested ) code, but to prevent large additions of untested code or turning off existing tests.

Overall I agree that we need to do a better job of documenting methods with Javadoc.

Incidentally, we should also look at how many of the warnings are generated from overriding methods.  We can use inheritdoc to reduce the numbers further.

rhwood, 2019-03-06:
I'm for documenting everything.

In the case of `remove(T item)`, I'd think that needs to be documented to discuss what happens if `item` is not present, is only the first instance of `item` removed or all instances, and what happens if `item` is null (yes, there are annotations for that, but users writing jython scripts or third parties relying on a compiled jmri.jar have no access to the annotations, since the annotations are removed at compile time).

bobjacobsen, 2019-03-06:
Not sure what you're saying about annotations.  This code:
```
        /**
         * Do big processing on the the input string.
         * <p>Always returns the result, although it might be small instead of big.
         * @param startingString The string to use as a starting point, but you knew that, didn't you?
         */
         @Nonnull
         public String doSomethingBigToString(@Nonnull String startingString) {}
```
generates this for the Javadoc summary:
[image]

which doesn't show the annotations, but the Method Detail section does show them:

[image]

That "The string to use ..." form used to be recommended by Sun, though I haven't checked recently:  Start with a capital letter, the first noun should be the (lower-case, as it refers to an object) type of the parameter, followed by what it's use for.  Given that the type is now right before that, and that the goal is to have useful names, the whole thing seems redundant to me (hence the text in my example), but maybe that's still the right thing to do.

I also tend to thing that the description of the what the method does belongs to the comment block that describes what the method does.  How to handle internal state (i.e. parameter object not present) isn't a property of the parameter, it's a property of the method.  When documenting

```
  public int computeFunctionWithTwoInputs(int a, int b)
```

where do you discuss that what happens depends on whether a is bigger or smaller than b?  That b is used twice if a == 0?  That these are used together with information already in the object?  If you have to describe any holistic behavior, it seems better to me to describe all the behavior in the same place: the method comment.

silverailscolo, 2019-03-06:
I agree on this (time permitting)
And economically stripping out the dashes using a ``grep`` improves readability in Javadoc (and stumbling over it during coding).

Egbert

> Op 6 mrt. 2019, om 11:48 heeft Randall Wood <notifications@github.com> het volgende geschreven:
> I'm for documenting everything.
> 
> In the case of remove(T item), I'd think that needs to be documented to discuss what happens if item is not present, is only the first instance of item removed or all instances, and what happens if item is null (yes, there are annotations for that, but users writing jython scripts or third parties relying on a compiled jmri.jar have no access to the annotations, since the annotations are removed at compile time)
>

danielb987, 2019-03-07:
I'm working on a new peice of code and have some methods like this:

```
/**
 * Set an analog value.
 */
public void setValue(float value);
```
I have omitted the javadoc for the parameter since I have thought that it doesn't add any value. But I'm happy to add it if desired. If so, what should I write as javadoc?

Is this the way to go?
```
/**
 * Set an analog value.
 *
 * @param value the value
 */
public void setValue(float value);
```

pabender, 2019-03-07:
@danielb987

Without context I cannot say if that is adequate.

As an example, if the value is constrained, you need to add something about the constraints and what happens if they are violated.

danielb987, 2019-03-07:
I'm working on a new generation Logix. Instead of Conditionals, it has Expressions and Actions, there Expressions ask a question and Actions do something.

The example above is pulled from the interface AnalogAction which sets an analog value. The entire interface is below.

There is no constraint of the value to the method setValue() except that it must be a valid float, which is ensured by the compiler. However, it may be desired to not accept NaN as value.

```
package jmri.jmrit.logixng;

import jmri.NamedBean;

/**
 * A LogixNG analog action.
 * 
 * @author Daniel Bergqvist Copyright 2018
 */
public interface AnalogAction extends NamedBean, Base {

    /**
     * Set an analog value.
     */
    public void setValue(float value);
    
}
```

The AnalogExpression looks like this:
```
package jmri.jmrit.logixng;

import jmri.NamedBean;

/**
 * Analog expression is used in LogixNG to answer a question that can give
 * an analog value as result.
 * 
 * @author Daniel Bergqvist Copyright 2018
 */
public interface AnalogExpression extends NamedBean, Base {
    
    /**
     * Evaluate this expression.
     * 
     * @return the result of the evaluation
     */
    public float evaluate();
    
}
```

rhwood, 2019-03-07:
> I also tend to thing that the description of what the method does belongs to the comment block that describes what the method does. How to handle internal state (i.e. parameter object not present) isn't a property of the parameter, it's a property of the method. When documenting

I was under the impression annotations are not included in Javadocs; my mistake. However, I also see that annotations have been added that putting illegal constraints on an overridden method signature—I’d rather discourage the breaking of a class's contract than rely on annotations for documentation.

rhwood, 2019-03-07:
In @danielb987's examples, the discouragement of using NaN as a value should be documented (and why) and in the case of the `evaluate()` method what happens when an expression is evaluated and has an error (like divide by zero or multiplying Float.MAX_VALUE * 2 could trigger)?

closed 2021-04-14 by stale[bot]