JMRI/JMRI #6636: Is EasyDCC broken?

bobjacobsen, 2019-03-05:
There have been a couple comments on jmriusers that perhaps EasyDCC is communications is broken in recent versions.  [Earlier today](https://groups.io/g/jmriusers/message/157607):

> I too am having difficulty with EasyDCC and JMRI interface. As with Peter, this has worked in the past to the point where I was able to run WiThrottle on my cell phone. I am able to get a response from the Command Station when I send a "V" command via the JMRI EasyDCC Command window. However, when I attempt to read a decoder, one that can be read and programmed from the command station, I see nothing in the EasyDCC Command Monitor window. Rick Beaber has been helping and posted a message to this group which so far has gone unanswered. I am using the same computer (windows 7) but reloaded to factory specs. Since I am receiving information back from the Command Station with the "V" (version) command, I am comfortable that I have the cable and connection working properly. In Rick's post, he commented that there might be an issue in the JMRI EasyDCC section. He normally programs with Sprog but set his connect back to EasyDCC and received the same results as I have.

bobjacobsen, 2019-03-05:
An [earlier post](
https://groups.io/g/jmriusers/message/157456):

> Dan Foltz was having problems using "Identify" with his EasyDCC system and a loco on the programming track.  I also tried, and I find I'm having the same problem, so was wondering if anyone else had run into this. Maybe we've found a bug?
> 
> After verifying connectivity to the EasyDCC command station (using the version command), clicking "Identify" results in an immediate exception, programmer in use.  Dan was using 4.14, and I'm using 4.15.2.  Watching the Command Monitor window, it does not appear that any commands are being sent to the EasyDCC command station after the version command. The line below that says ERROR- reply complete is my sending the "V" (version) command, and the response was as expected, so not sure why it says error.  After that is when I tried the "Identify" command.
> 
> The system log:
> 
> 2019-03-01 13:36:51,250 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
> 2019-03-01 13:36:51,275 util.Log4JUtil                        INFO  - This log is appended to file: C:\Users\Trains\JMRI\log\messages.log [main]
> 2019-03-01 13:36:51,276 util.Log4JUtil                        INFO  - This log is stored in file: C:\Users\Trains\JMRI\log\session.log [main]
> 2019-03-01 13:36:51,290 apps.AppsBase                         INFO  - DecoderPro version 4.15.2+Rfda0762 starts under Java 1.8.0_201 on Windows 10 x86 v10.0 at Fri Mar 01 13:36:51 PST 2019 [main]
> 2019-03-01 13:36:51,729 gui3.Apps3                            INFO  - Starting with profile DecoderPro_3.3e0bff5e [main]
> 2019-03-01 13:36:52,099 node.NodeIdentity                     INFO  - Using jmri-l0_RJbCEbhNiaacFaMaaaf-3e0bff5e as the JMRI Node identity [AWT-EventQueue-0]
> 2019-03-01 13:36:52,296 xml.AbstractSerialConnectionConfigXml INFO  - Starting to connect for "EasyDCC via Serial" [main]
> 2019-03-01 13:36:54,021 serialdriver.SerialDriverAdapter      INFO  - COM4 port opened at 9600 baud, sees  DTR: true RTS: true DSR: false CTS: false  CD: false [main]
> 2019-03-01 13:36:54,722 util.FileUtilSupport                  INFO  - File path program: is C:\Program Files (x86)\JMRI\ [main]
> 2019-03-01 13:36:54,723 util.FileUtilSupport                  INFO  - File path preference: is C:\Users\Trains\JMRI\ [main]
> 2019-03-01 13:36:54,724 util.FileUtilSupport                  INFO  - File path profile: is C:\Users\Trains\JMRI\DecoderPro_3\ [main]
> 2019-03-01 13:36:54,724 util.FileUtilSupport                  INFO  - File path settings: is C:\Users\Trains\JMRI\ [main]
> 2019-03-01 13:36:54,725 util.FileUtilSupport                  INFO  - File path home: is C:\Users\Trains\ [main]
> 2019-03-01 13:36:54,726 util.FileUtilSupport                  INFO  - File path scripts: is C:\Program Files (x86)\JMRI\ [main]
> 2019-03-01 13:37:17,142 jmrix.AbstractMRTrafficController     ERROR - reply complete in unexpected state: 10 was O
>  in class easydcc.EasyDccTrafficController [easydcc.EasyDccTrafficController Receive thread]
> 2019-03-01 13:39:49,579 symbolicprog.CvValue                  WARN  - Exception during CV read: java.lang.NullPointerException [AWT-EventQueue-0]
> 2019-03-01 13:39:49,582 mentation.VerifyWriteProgrammerFacade INFO  - programmer already in use by CvValue _num=1 _cvName= [AWT-EventQueue-0]
> 2019-03-01 13:39:49,584 symbolicprog.CvValue                  WARN  - Exception during CV read: jmri.ProgrammerException: programmer in use [AWT-EventQueue-0]
> 2019-03-01 13:39:49,585 mentation.VerifyWriteProgrammerFacade INFO  - programmer already in use by CvValue _num=1 _cvName= [AWT-EventQueue-0]
> 2019-03-01 13:39:49,586 symbolicprog.CvValue                  WARN  - Exception during CV read: jmri.ProgrammerException: programmer in use [AWT-EventQueue-0]
> 2019-03-01 13:39:49,586 mentation.VerifyWriteProgrammerFacade INFO  - programmer already in use by CvValue _num=1 _cvName= [AWT-EventQueue-0]
> 2019-03-01 13:39:49,587 symbolicprog.CvValue                  WARN  - Exception during CV read: jmri.ProgrammerException: programmer in use [AWT-EventQueue-0]
> 2019-03-01 13:39:49,590 mentation.VerifyWriteProgrammerFacade INFO  - programmer already in use by CvValue _num=1 _cvName= [AWT-EventQueue-0]
> 2019-03-01 13:39:49,590 symbolicprog.CvValue                  WARN  - Exception during CV read: jmri.ProgrammerException: programmer in use [AWT-EventQueue-0]
> 2019-03-01 13:39:49,591 mentation.VerifyWriteProgrammerFacade INFO  - programmer already in use by CvValue _num=1 _cvName= [AWT-EventQueue-0]
> 2019-03-01 13:39:49,592 symbolicprog.CvValue                  WARN  - Exception during CV read: jmri.ProgrammerException: programmer in use [AWT-EventQueue-0]
> 2019-03-01 13:39:49,592 mentation.VerifyWriteProgrammerFacade INFO  - programmer already in use by CvValue _num=1 _cvName= [AWT-EventQueue-0]
> 2019-03-01 13:39:49,592 symbolicprog.CvValue                  WARN  - Exception during CV read: jmri.ProgrammerException: programmer in use [AWT-EventQueue-0]
> 2019-03-01 13:39:49,593 mentation.VerifyWriteProgrammerFacade INFO  - programmer already in use by CvValue _num=1 _cvName= [AWT-EventQueue-0]
> 2019-03-01 13:39:49,593 symbolicprog.CvValue                  WARN  - Exception during CV read: jmri.ProgrammerException: programmer in use [AWT-EventQueue-0]
> 2019-03-01 13:39:49,594 mentation.VerifyWriteProgrammerFacade INFO  - programmer already in use by CvValue _num=1 _cvName= [AWT-EventQueue-0]
> 2019-03-01 13:39:49,594 symbolicprog.CvValue                  WARN  - Exception during CV read: jmri.ProgrammerException: programmer in use [AWT-EventQueue-0]
> 
> Any help would be appreciated.

tom1971, 2019-03-31:
I have the same thing happing to me, went back to JMRI.4.12+Rb6a9bb1 and it works

RBeaber, 2019-05-14:
The latest version (4.15.5), I can confirm this is still an issue.  Everything except decoder programming on a programming track works fine.  But trying to program on the programming track always results with this error:

2019-05-14 10:55:13,739 mentation.VerifyWriteProgrammerFacade INFO  - programmer already in use by CvValue _num=1 _cvName= [AWT-EventQueue-0]
2019-05-14 10:55:13,739 symbolicprog.CvValue                  WARN  - Exception during CV read: jmri.ProgrammerException: programmer in use [AWT-EventQueue-0]

I would be more than happy to help anyone troubleshoot this.  

Rick Beaber
rbeaber@gmail.com

dheap, 2019-05-14:
@RBeaber Try turning off the Verify Writes option in Preferences->Roster->Programmer.

bobjacobsen, 2019-05-17:
Another trace from a posting:
```
2019-05-16 18:39:50,980 util.Log4JUtil INFO - ****** JMRI log ******* [main]
2019-05-16 18:39:50,996 util.Log4JUtil INFO - This log is appended to file: C:\Users\Grandpa Tom\JMRI\log\messages.log [main]
2019-05-16 18:39:50,996 util.Log4JUtil INFO - This log is stored in file: C:\Users\Grandpa Tom\JMRI\log\session.log [main]
2019-05-16 18:39:51,011 apps.Apps INFO - PanelPro version 4.15.6+Red1ceb2 starts under Java 1.8.0_211 on Windows 10 amd64 v10.0 at Thu May 16 18:39:51 EDT 2019 [main]
2019-05-16 18:39:53,675 apps.Apps INFO - Starting with profile My_JMRI_Railroad.3f512bac [main]
2019-05-16 18:39:53,769 node.NodeIdentity INFO - Using jmri-nEDRBH5YJ-NkhWaUtZlTOa-3f512bac as the JMRI storage identity for profile id 3f512bac [AWT-EventQueue-0]
2019-05-16 18:39:53,878 xml.AbstractSerialConnectionConfigXml INFO - Starting to connect for "EasyDCC via Serial" [main]
2019-05-16 18:39:54,542 serialdriver.SerialDriverAdapter INFO - COM3 port opened at 9600 baud, sees DTR: true RTS: true DSR: false CTS: false CD: false [main]
2019-05-16 18:39:56,613 util.FileUtilSupport INFO - File path program: is C:\Program Files (x86)\JMRI\ [main]
2019-05-16 18:39:56,613 util.FileUtilSupport INFO - File path preference: is C:\Users\Grandpa Tom\JMRI\My_JMRI_Railroad.jmri\ [main]
2019-05-16 18:39:56,613 util.FileUtilSupport INFO - File path profile: is C:\Users\Grandpa Tom\JMRI\My_JMRI_Railroad.jmri\ [main]
2019-05-16 18:39:56,613 util.FileUtilSupport INFO - File path settings: is C:\Users\Grandpa Tom\JMRI\ [main]
2019-05-16 18:39:56,613 util.FileUtilSupport INFO - File path home: is C:\Users\Grandpa Tom\ [main]
2019-05-16 18:39:56,613 util.FileUtilSupport INFO - File path scripts: is C:\Program Files (x86)\JMRI\jython\ [main]
2019-05-16 18:39:56,942 PanelPro.PanelPro INFO - Main initialization done [main]
2019-05-16 18:40:02,855 withrottle.FacelessServer INFO - Published ZeroConf service for 'my jmri railroad._withrottle._tcp.local.' on 192.168.0.100:12090 [WiThrottleServer]
2019-05-16 18:40:05,650 withrottle.FacelessServer INFO - Published ZeroConf service for 'my jmri railroad._withrottle._tcp.local.' on fe80:0:0:0:6886:5bc5:c274:e288%eth0:12090 [WiThrottleServer]
2019-05-16 18:40:05,650 withrottle.FacelessServer INFO - Creating new WiThrottle DeviceServer(socket) on port 12090, waiting for incoming connection... [WiThrottleServer]
2019-05-16 18:41:14,271 symbolicprog.CvValue WARN - Exception during write CV '29' to '32' [AWT-EventQueue-0]
java.lang.NullPointerException
at jmri.jmrix.AbstractMRTrafficController.sendMessage(AbstractMRTrafficController.java:320)
at jmri.jmrix.easydcc.EasyDccTrafficController.sendEasyDccMessage(EasyDccTrafficController.java:91)
at jmri.jmrix.easydcc.EasyDccProgrammer.writeCV(EasyDccProgrammer.java:67)
at jmri.implementation.AddressedHighCvProgrammerFacade.writeCV(AddressedHighCvProgrammerFacade.java:68)
at jmri.implementation.MultiIndexProgrammerFacade.writeCV(MultiIndexProgrammerFacade.java:241)
at jmri.jmrit.symbolicprog.CvValue.write(CvValue.java:417)
at jmri.jmrit.symbolicprog.EnumVariableValue.writeAll(EnumVariableValue.java:472)
at jmri.jmrit.symbolicprog.EnumVariableValue.writeChanges(EnumVariableValue.java:454)
at jmri.jmrit.symbolicprog.tabbedframe.PaneProgPane.executeWrite(PaneProgPane.java:807)
at jmri.jmrit.symbolicprog.tabbedframe.PaneProgPane.nextWrite(PaneProgPane.java:1022)
at jmri.jmrit.symbolicprog.tabbedframe.PaneProgPane.writePaneChanges(PaneProgPane.java:958)
at jmri.jmrit.symbolicprog.tabbedframe.PaneProgPane.lambda$2(PaneProgPane.java:373)
at javax.swing.AbstractButton.fireItemStateChanged(Unknown Source)
...
at javax.swing.JToggleButton$ToggleButtonModel.setPressed(Unknown Source)
at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(Unknown Source)
at java.awt.AWTEventMulticaster.mouseReleased(Unknown Source)
at java.awt.Component.processMouseEvent(Unknown Source)
at javax.swing.JComponent.processMouseEvent(Unknown Source)
at java.awt.Component.processEvent(Unknown Source)
...
at java.awt.LightweightDispatcher.retargetMouseEvent(Unknown Source)
at java.awt.LightweightDispatcher.processMouseEvent(Unknown Source)
at java.awt.LightweightDispatcher.dispatchEvent(Unknown Source)
at java.awt.Container.dispatchEventImpl(Unknown Source)
...
at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
at java.awt.EventDispatchThread.run(Unknown Source)
2019-05-16 18:41:14,287 symbolicprog.CvValue WARN - Exception during write CV '29' to '32' [AWT-EventQueue-0]
java.lang.NullPointerException
```

bobjacobsen, 2019-05-17:
I can recreate this last one, so will work on it this weekend.

RBeaber, 2019-05-17:
@dheap Unfortunately, turning off the Write Verify did not have any effect.  

Thank you!

-Rick

bobjacobsen, 2019-05-26:
I think PR #6995 fixed this.  There are installers at http://jmri.tagadab.com/jenkins/job/Development/job/Packages/ now.  @RBeaber could you try this is if you get a chance? Thanks

RBeaber, 2019-05-31:
Bob:
     Thank you very much!!  I was able to identify, read and write CV's with no issues.  It looks fixed to me!  Sorry it took so long to test it, I had some trouble getting the Jenkins server to talk nicely.  Seems OK now.
     Thanks again!

-Rick

closed 2019-05-31 by bobjacobsen