JMRI/JMRI #6659: AllTest running too long

bobjacobsen, 2019-03-11:
Recently (past few days?)  both AppVeyor and Travis graphical have taken to running too long.  Jenkins is seeing this too on jobs that run AllTest.

'ant alltest' on my laptop, which used to run in ~30 minutes, now runs for as long as I let it.

Running specific tests seems to point to:
```
./runtest.csh jmri.jmrit.logix.PackageTest
```

as (at least one) something that runs forever. In that state, a thread dump doesn't show a `main` thread!  
[thread.txt](https://github.com/JMRI/JMRI/files/2950869/thread.txt)

Anybody know what this might be?

bobjacobsen, 2019-03-11:
`git bisect` narrowed this down to commit e7d516d2d1e8eebd10bc63551071bb84c07945e3
```
commit e7d516d2d1e8eebd10bc63551071bb84c07945e3
Author: Randall Wood <297232+rhwood@users.noreply.github.com>
Date:   Mon Mar 4 06:34:12 2019 -0500

    test: Do not use JUnitCore.main()
    
    Use JUnitCore.runClasses() instead, per JUnit docs.
```

The changes are here:  
[changes.txt](https://github.com/JMRI/JMRI/files/2950989/changes.txt)  (From `git diff 96ce34f8e2010b42ed77bb48a890be2b0275577f...e7d516d2d1e8eebd10bc63551071bb84c07945e3`)

Since I've been running this test via `./runtest.csh`, the most relevant part seems to be:
```
diff --git a/java/test/jmri/util/junit/TestClassMainMethod.java b/java/test/jmri/util/junit/TestClassMainMethod.java
index 231ff294d2..40b67b1246 100644
--- a/java/test/jmri/util/junit/TestClassMainMethod.java
+++ b/java/test/jmri/util/junit/TestClassMainMethod.java
@@ -2,6 +2,8 @@ package jmri.util.junit;
 
 import java.lang.reflect.*;
 
+import org.junit.runner.JUnitCore;
+
 /**
  * Main method to launch a JUnit test class
  *
@@ -25,16 +27,21 @@ public class TestClassMainMethod {
         className = className.replace("..",".");    
         
         try {
-            // first try to find a main in the class
             Class<?> cl = Class.forName(className);
-            Method method = cl.getMethod("main", String[].class);
-            method.invoke(null, new Object[] {new String[] { /* put args here */ }});
-        } catch (InvocationTargetException e) {
-            // main threw an exception, report
-            System.err.println(e);
-        } catch (ClassNotFoundException | NoSuchMethodException | IllegalAccessException e) {
-            // failed, now invoke manually
-            org.junit.runner.JUnitCore.main(className);
+            // first try to find a main in the class
+            try {
+                Method method = cl.getMethod("main", String[].class);
+                method.invoke(null, new Object[] {new String[] { /* put args here */ }});
+            } catch (InvocationTargetException e) {
+                // main threw an exception, report
+                System.err.println(e);
+            } catch (NoSuchMethodException | IllegalAccessException e) {
+                // failed, now invoke manually
+                JUnitCore.runClasses(cl);
+            }
+        } catch (ClassNotFoundException e) {
+            // log error
+            System.err.println("Unable to locate class " + className);
         }
     }
 }
```

rhwood, 2019-03-11:
I did that because [JUnit documentation](https://github.com/junit-team/junit4/wiki/Test-runners) suggested using `runClasses()` instead of `main()` was the correct method for launching JUnit tests.

rhwood, 2019-03-11:
I think this immediate issue is fixed in 461367a in #6657

rhwood, 2019-03-11:
And 6009514 restores the JUnit default text output for tests in `runtest.csh`. Note that we do now have the option of implementing our own output formatter.

PR #6657 (2019-03-11): Minor fixes and improvements for JSON servers
- JSON Sensors show inverted state
- Fix schemas for JSON sensor, time, and turnout types
- Additional tests for JSON sensor, time, and turnout types
- Additional tests for JSON utility services
- Address all warnings flagged by Eclipse in `jmri.server` packages
- Fix a number of tests (approximately 169) that were leaving Panels in place after calling `dispose()` on them (discovered when adding tests for JSON utility services)
- Add some tests to their package suite
- Fixes #6656 (found while working on tests for JSON time type)
- Fixes #6659 (int [... cut]
closed 2019-03-11 by pabender