JMRI/JMRI #6660: Reporters Table can be inaccurate

devel-bobm, 2019-03-11:
Seen in 4.15.4 (first pre-release build):
The "Reporters" table does not always reflect "absence" events as reported by a LocoNet transponding device.
Testing with real hardware usually does not reflect transponding "absence" events, but sometimes does.

I ran a "LocoNet Simulator" connection to send sent a couple of transponding messages using Send "LocoNet Packet".  Messages were:

```
D0 20 06 7D 03 77
D0 00 06 7D 03 57
```

LocoNet Monitor decodes these as follows:

```
[D0 20 06 7D 03 77]  Transponder address 3 (short) (or long address 16003) present at LR7 () (BDL16x Board ID 1 RX4 zone D or BXP88 Board ID 1 section 7 or the BXPA1 Board ID 7 section).
[D0 00 06 7D 03 57]  Transponder address 3 (short) (or long address 16003) absent at LR7 () (BDL16x Board ID 1 RX4 zone D or BXP88 Board ID 1 section 7 or the BXPA1 Board ID 7 section).
```

The "Reporters" table only reflects the first of the two messages:

[image]

devel-bobm, 2019-03-11:
Some additional LocoNet message examples:

```
[D0 21 1E 0A 78 62]  Transponder address 1400 (long) present at LR159 () (BDL16x Board 10 RX4 zone H). 
[D0 01 1E 0A 78 42]  Transponder address 1400 (long) absent at LR159 () (BDL16x Board 10 RX4 zone H). 
[D0 21 18 0A 78 64]  Transponder address 1400 (long) present at LR153 () (BDL16x Board 10 RX4 zone E). 
[D0 01 18 0A 78 44]  Transponder address 1400 (long) absent at LR153 () (BDL16x Board 10 RX4 zone E).  
[D0 21 12 0A 78 6E]  Transponder address 1400 (long) present at LR147 () (BDL16x Board 10 RX4 zone B).
[D0 01 12 0A 78 4E]  Transponder address 1400 (long) absent at LR147 () (BDL16x Board 10 RX4 zone B).
[D0 21 10 0A 78 6C]  Transponder address 1400 (long) present at LR145 () (BDL16x Board 10 RX4 zone A).
[D0 01 10 0A 78 4C]  Transponder address 1400 (long) absent at LR145 () (BDL16x Board 10 RX4 zone A).
[D0 00 7F 49 39 20]  Transponder address 9401 (long) absent at LR128 () (BDL16x Board ID 8 RX4 zone H).
[D0 20 7F 49 39 00]  Transponder address 9401 (long) present at LR128 () (BDL16x Board ID 8 RX4 zone H).
[D0 20 0B 00 03 07]  Transponder address 3 (short) present at LR12 () (BXP88 Board ID 2 section 4 or the BXPA1 Board ID 12 section).
[D0 00 0B 00 03 27]  Transponder address 3 (short) absent at LR12 () (BXP88 Board ID 2 section 4 or the BXPA1 Board ID 12 section).
[D0 20 08 00 03 04]  Transponder address 3 (short) present at LR9 () (BDL16x Board ID 1 RX4 zone E or BXP88 Board ID 2 section 1 or the BXPA1 Board ID 9 section).
[D0 00 08 00 03 24]  Transponder address 3 (short) absent at LR9 () (BDL16x Board ID 1 RX4 zone E or BXP88 Board ID 2 section 1 or the BXPA1 Board ID 9 section).
[D0 20 09 00 03 05]  Transponder address 3 (short) present at LR10 () (BXP88 Board ID 2 section 2 or the BXPA1 Board ID 10 section).
[D0 00 09 00 03 25]  Transponder address 3 (short) absent at LR10 () (BXP88 Board ID 2 section 2 or the BXPA1 Board ID 10 section).
[D0 20 0A 00 03 06]  Transponder address 3 (short) present at LR11 () (BDL16x Board ID 1 RX4 zone F or BXP88 Board ID 2 section 3 or the BXPA1 Board ID 11 section).
[D0 00 0A 00 03 26]  Transponder address 3 (short) absent at LR11 () (BDL16x Board ID 1 RX4 zone F or BXP88 Board ID 2 section 3 or the BXPA1 Board ID 11 section).
[D0 20 0B 00 03 07]  Transponder address 3 (short) present at LR12 () (BXP88 Board ID 2 section 4 or the BXPA1 Board ID 12 section).
[D0 00 0B 00 03 27]  Transponder address 3 (short) absent at LR12 () (BXP88 Board ID 2 section 4 or the BXPA1 Board ID 12 section).
[D0 20 0C 00 03 00]  Transponder address 3 (short) present at LR13 () (BDL16x Board ID 1 RX4 zone G or BXP88 Board ID 2 section 5 or the BXPA1 Board ID 13 section).
[D0 00 0C 00 03 20]  Transponder address 3 (short) absent at LR13 () (BDL16x Board ID 1 RX4 zone G or BXP88 Board ID 2 section 5 or the BXPA1 Board ID 13 section).
[D0 20 0D 00 03 01]  Transponder address 3 (short) present at LR14 () (BXP88 Board ID 2 section 6 or the BXPA1 Board ID 14 section).
[D0 00 0D 00 03 21]  Transponder address 3 (short) absent at LR14 () (BXP88 Board ID 2 section 6 or the BXPA1 Board ID 14 section).
[D0 20 0E 00 03 02]  Transponder address 3 (short) present at LR15 () (BDL16x Board ID 1 RX4 zone H or BXP88 Board ID 2 section 7 or the BXPA1 Board ID 15 section).
[D0 00 0E 00 03 22]  Transponder address 3 (short) absent at LR15 () (BDL16x Board ID 1 RX4 zone H or BXP88 Board ID 2 section 7 or the BXPA1 Board ID 15 section).
[D0 20 0F 00 03 03]  Transponder address 3 (short) present at LR16 () (BXP88 Board ID 2 section 8 or the BXPA1 Board ID 16 section).
[D0 00 0F 00 03 23]  Transponder address 3 (short) absent at LR16 () (BXP88 Board ID 2 section 8 or the BXPA1 Board ID 16 section).
[D0 20 0C 00 03 00]  Transponder address 3 (short) present at LR13 () (BDL16x Board ID 1 RX4 zone G or BXP88 Board ID 2 section 5 or the BXPA1 Board ID 13 section).
[D0 00 0C 00 03 20]  Transponder address 3 (short) absent at LR13 () (BDL16x Board ID 1 RX4 zone G or BXP88 Board ID 2 section 5 or the BXPA1 Board ID 13 section).
```

pabender, 2019-03-11:
> Seen in 4.15.4 (first pre-release build):
> The "Reporters" table does not always reflect "absence" events as reported by a LocoNet transponding device.
> Testing with real hardware usually does not [...]

I see the exit, which is the last of the two messages.  That is correct behavior here.

pabender, 2019-03-11:
Is your concern that “last report” doesn’t show the enter message?

That isn’t something we are going to be able to change for two reasons:
1) the tag itself is storing the enter/exit as an attribute.  There is only one tag, so both reports are the same
2) the reporter api requires that if there is a current report, the last report and current report are equal.

devel-bobm, 2019-03-11:
@pabender - you're right, the image shows "exit".  It turns out that the display changed between the time I sent the message and the time I captured the screen-shot.

After a repeated test, I find that the "exit" case did not show up in the Reporters Table - until after I switched focus away from JMRI and back.  And then I re-sent the entry message, and it didn't show up in the Reporters Table until after I switched focus away and back to JMRI. I do not know if it changed at the switch away or the switch back.

Then I sent the "exit" message, and switched away and the Reporters Table display didn't change, then switched back and the Reporters Table display didn't change.  And opened a new application and switched back and the Reporters Table display didn't change.

I've waited a couple of minutes since sending the "exit" message and the Reporters Table display still shows "enter".  I've sent the "exit" message again, and the Reporters Table still shows "enter".

There is nothing in the Console log other than the LnReporter "debug" messages I've enabled, which shows the "exits" and "enter" debug messages.  Those debug messages seem consistent with the LocoNet messages I've sent.

devel-bobm, 2019-03-11:
Does this console log help any?  It is from a separate run, where I've only sent the "entry" message once and the "exit" message once.  The "Reporters" table shows "3 enter" for both "Report" and "Last Report.

```
2019-03-11 09:01:09,283 managers.AbstractReporterManager      DEBUG - new Reporter: LR7 null [AWT-EventQueue-0]
2019-03-11 09:01:09,293 loconet.LnReporter                    DEBUG - new Reporter 7 [AWT-EventQueue-0]
2019-03-11 09:01:09,323 loconet.LnReporter                    DEBUG - Tag: 3 enter [AWT-EventQueue-0]
2019-03-11 09:01:09,323 implementation.AbstractIdTagReporter  DEBUG - Notify: LR7 [AWT-EventQueue-0]
2019-03-11 09:01:09,323 implementation.AbstractIdTagReporter  DEBUG - Tag: 3 enter [AWT-EventQueue-0]
2019-03-11 09:01:09,323 implementation.AbstractIdTagReporter  DEBUG - Seen here: LR7 [AWT-EventQueue-0]
2019-03-11 09:01:14,343 loconet.LnReporter                    DEBUG - Tag: 3 exits [AWT-EventQueue-0]
2019-03-11 09:01:14,343 implementation.AbstractIdTagReporter  DEBUG - Notify: LR7 [AWT-EventQueue-0]
2019-03-11 09:01:14,343 implementation.AbstractIdTagReporter  DEBUG - Tag: 3 exits [AWT-EventQueue-0]
2019-03-11 09:01:14,343 implementation.AbstractIdTagReporter  DEBUG - Previous reporter: LR7 [AWT-EventQueue-0]
2019-03-11 09:01:14,343 implementation.AbstractIdTagReporter  DEBUG - Current report was: 3 exits [AWT-EventQueue-0]
2019-03-11 09:01:14,343 implementation.AbstractIdTagReporter  DEBUG - Seen here: LR7 [AWT-EventQueue-0]
2019-03-11 09:01:14,343 loconet.LnReporter                    DEBUG - Tag: 3 exits [AWT-EventQueue-0]
2019-03-11 09:01:14,343 implementation.AbstractIdTagReporter  DEBUG - Notify: LR7 [AWT-EventQueue-0]
2019-03-11 09:01:14,343 implementation.AbstractIdTagReporter  DEBUG - Tag: 3 exits [AWT-EventQueue-0]
2019-03-11 09:01:14,343 implementation.AbstractIdTagReporter  DEBUG - Previous reporter: LR7 [AWT-EventQueue-0]
2019-03-11 09:01:14,353 implementation.AbstractIdTagReporter  DEBUG - Current report was: 3 exits [AWT-EventQueue-0]
2019-03-11 09:01:14,353 implementation.AbstractIdTagReporter  DEBUG - Seen here: LR7 [AWT-EventQueue-0]

```

devel-bobm, 2019-03-11:
@pabender,
My problem is that the Reporters Table is behaving inconsistently, per my comments above.  
- It does not reliably show "exit" and "entry" events.  
- It sometimes changes the Reporters Table display to the correct state when focus shifts to another application.
- At other times, changing focus has no effect on display of a "not yet reported" state change event - the previously-reported state is maintained.

bobjacobsen, 2019-03-11:
This sounds like the Table isn't being told there's a difference.  

`jmrit.beantable.BeanTableDataModel` has a `propertyChange` method (line 130 or so) that gets `PropertyChange` events from all the table's beans, and decides whether to fire table refreshes.

- Are all the Reporter value changes going there?
- Do they carry the `getPropertyName()` and `getSource()` values needed?

pabender, 2019-03-11:
Ok, I think I understand what is happening.

Because the enter and exit are for the same object, the property change for the report values isn’t sent.  That is actually correct behavior, though it doesn’t show what you want it to in the table.

To force the table to update correctly, there are at least a couple of options:

1. The quick and dirty potion is to just set the report to null before sending the new report.
2. A ( perhaps better, but perhaps not ) way of doing this might be to have the reporter table listen to property changes on the IDTag objects in the reports.

I’ve implemented the first of these in PR #6661.

bobjacobsen, 2019-03-11:
> Because the enter and exit are for the same object, the property change for the report values isn’t sent.

I don’t think I understand the statement above. It’s routine for the same object to send PropertyChange events when just the _value_ of that property changes. Am I misunderstanding sone about ‘value’?

devel-bobm, 2019-03-11:
FYI - I am not a frequent "reporter" user; I simply saw that LocoNet transponding activities were not being reported in the same way that they were when I added the BXP88 and BXPA1 transponding features to the pre-existing BDL168+RX4 transponding features.  I figure that this new behavior would be problematic for existing users of JMRI Reporters with LocoNet Transponding.

pabender, 2019-03-11:
Sent from my iPad

> On Mar 11, 2019, at 10:42 AM, Bob Jacobsen <notifications@github.com> wrote:
> > Because the enter and exit are for the same object, the property change for the report values isn’t sent.
> 
> I don’t think I understand the statement above. It’s routine for the same object to send PropertyChange events when just the _value_ of that property changes. Am I misunderstanding sone about ‘value’?

The value the reporter is reporting is the tag.

The tag maintains the enter and exit property so it can be used in reports.  In other words, The tag has properties that change, but the tag ( which is being reported by the reporter ) does not actually change.  

We fire a property change on the report value when the property changes in the tag, but if the listener is looking superficially at “did the object change”, it sees the new tag and the old tag are the same, so it may not react to the change.  I.e. the listener on the reporter table isn’t looking for the change in the tag property but for a change in the actual tag.

pabender, 2019-03-11:
> FYI - I am not a frequent "reporter" user; I simply saw that LocoNet transponding activities were not being reported in the same way that they were when I added the BXP88 and BXPA1 transponding [...]

For what it’s worth, we implemented the toString on the TranspondingTag objects and immediately deprecated it so that existing users would have time to adjust to the new behavior.  The old behavior had the LocoNet reporters reporting Strings, which are hard to use in other parts of JMRI ( such as operations ).

pabender, 2019-03-11:
@devel-bobm I just merged #6661.  Can you verify this solves the problem for you.

devel-bobm, 2019-03-11:
The behavior from #6661 _seems_ to be consistent with the behavior I remember from previous JMRI behavior.  Thanks!

We will have to rely upon user input to validate this approach.

devel-bobm, 2019-03-11:
P/R #6661 seems to restore the timely and accurate "Reporters Table" reporting of LocoNet transponding messages, without "missing" any reports from LocoNet messages.  I believe that P/R #6661 addresses the issue adequately and thus am closing this issue.

closed 2019-03-11 by devel-bobm