JMRI/JMRI #6667: Exception loading TCS decoder definition with current head of master

pabender, 2019-03-12:
I am getting the following exception when trying to load a decoder definition for a TCS IMF4 decoder from the January 2012 family:

     [java] 320465 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
     [java] java.lang.NullPointerException
     [java] 	at jmri.jmrit.symbolicprog.FnMapPanel.addOutputElements(FnMapPanel.java:392)
     [java] 	at jmri.jmrit.symbolicprog.FnMapPanel.configOutputs(FnMapPanel.java:352)
     [java] 	at jmri.jmrit.symbolicprog.FnMapPanel.<init>(FnMapPanel.java:160)
     [java] 	at jmri.jmrit.symbolicprog.tabbedframe.PaneProgPane.pickFnMapPanel(PaneProgPane.java:2228)
     [java] 	at jmri.jmrit.symbolicprog.tabbedframe.PaneProgPane.newColumn(PaneProgPane.java:1535)
     [java] 	at jmri.jmrit.symbolicprog.tabbedframe.PaneProgPane.<init>(PaneProgPane.java:268)
     [java] 	at jmri.jmrit.symbolicprog.tabbedframe.PaneProgFrame.newPane(PaneProgFrame.java:1216)
     [java] 	at jmri.jmrit.symbolicprog.tabbedframe.PaneProgFrame.readConfig(PaneProgFrame.java:960)
     [java] 	at jmri.jmrit.symbolicprog.tabbedframe.PaneProgFrame.loadProgrammerFile(PaneProgFrame.java:790)
     [java] 	at jmri.jmrit.symbolicprog.tabbedframe.PaneProgFrame.<init>(PaneProgFrame.java:509)
     [java] 	at jmri.jmrit.symbolicprog.tabbedframe.PaneServiceProgFrame.<init>(PaneServiceProgFrame.java:57)
     [java] 	at jmri.jmrit.symbolicprog.tabbedframe.PaneProgAction$2.startProgrammer(PaneProgAction.java:106)
     [java] 	at jmri.jmrit.symbolicprog.CombinedLocoSelPane.openKnownLoco(CombinedLocoSelPane.java:585)
     [java] 	at jmri.jmrit.symbolicprog.CombinedLocoSelPane.openButton(CombinedLocoSelPane.java:563)
     [java] 	at jmri.jmrit.symbolicprog.CombinedLocoSelPane$5.actionPerformed(CombinedLocoSelPane.java:268)
     [java] 	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
     [java] 	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
     [java] 	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
     [java] 	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
     [java] 	at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:252)
     [java] 	at java.awt.AWTEventMulticaster.mouseReleased(AWTEventMulticaster.java:289)
     [java] 	at java.awt.Component.processMouseEvent(Component.java:6539)
     [java] 	at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
     [java] 	at java.awt.Component.processEvent(Component.java:6304)
     [java] 	at java.awt.Container.processEvent(Container.java:2239)
     [java] 	at java.awt.Component.dispatchEventImpl(Component.java:4889)
     [java] 	at java.awt.Container.dispatchEventImpl(Container.java:2297)
     [java] 	at java.awt.Component.dispatchEvent(Component.java:4711)
     [java] 	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4904)
     [java] 	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4535)
     [java] 	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4476)
     [java] 	at java.awt.Container.dispatchEventImpl(Container.java:2283)
     [java] 	at java.awt.Window.dispatchEventImpl(Window.java:2746)
     [java] 	at java.awt.Component.dispatchEvent(Component.java:4711)
     [java] 	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:760)
     [java] 	at java.awt.EventQueue.access$500(EventQueue.java:97)
     [java] 	at java.awt.EventQueue$3.run(EventQueue.java:709)
     [java] 	at java.awt.EventQueue$3.run(EventQueue.java:703)
     [java] 	at java.security.AccessController.doPrivileged(Native Method)
     [java] 	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
     [java] 	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:84)
     [java] 	at java.awt.EventQueue$4.run(EventQueue.java:733)
     [java] 	at java.awt.EventQueue$4.run(EventQueue.java:731)
     [java] 	at java.security.AccessController.doPrivileged(Native Method)
     [java] 	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
     [java] 	at java.awt.EventQueue.dispatchEvent(EventQueue.java:730)
     [java] 	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
     [java] 	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
     [java] 	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
     [java] 	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
     [java] 	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
     [java] 	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

This is happening both with a previously existing decoder definition and trying to open a new new decoder definition.

I also tried several nearby decoder definitions from the list, and they also fail to load.

@bobjacobsen, could this be related to your recent changes related to the Function panel?

bobjacobsen, 2019-03-12:
PR #6670 should fix this. Sorry for the trouble.

pabender, 2019-03-12:
Thanks @bobjacobsen, that fixed it.

closed 2019-03-12 by pabender