JMRI/JMRI #6685: Jenkins fails test w non-ASCII characters

bobjacobsen, 2019-03-16:
[Jenkins Jacoco job is now failing](http://jmri.tagadab.com/jenkins/job/Development/job/JaCoCo/2028) due to non-ASCII characters in a test that passes CI. Is this a Jenkins configuration issue?

```
   [javac] <http://builds.jmri.org/jenkins/job/Development/job/JaCoCo/ws/java/test/jmri/util/zeroconf/ZeroConfServiceManagerTest.java>:189: error: unmappable character for encoding ASCII
   [javac]         Assert.assertEquals("Lots of emojii", "xn--358haaaa8nbbbb", ZeroConfServiceManager.hostName("????????????????????????????????????????"));
   [javac]                                                                                                                                             ^
   [javac] <http://builds.jmri.org/jenkins/job/Development/job/JaCoCo/ws/java/test/jmri/util/zeroconf/ZeroConfServiceManagerTest.java>:192: error: unmappable character for encoding ASCII
   [javac]                 ZeroConfServiceManager.hostName("My ???????? Railroad My ???????? Railroad My ???????? Railroad My ???????? Railroad My ???????? Railroad My ???????? Railroad"));
   [javac]                                                     ^
   [javac] <http://builds.jmri.org/jenkins/job/Development/job/JaCoCo/ws/java/test/jmri/util/zeroconf/ZeroConfServiceManagerTest.java>:192: error: unmappable character for encoding ASCII
   [javac]                 ZeroConfServiceManager.hostName("My ???????? Railroad My ???????? Railroad My ???????? Railroad My ???????? Railroad My ???????? Railroad My ???????? Railroad"));
   [javac]                                                      ^
```

rhwood, 2019-03-16:
Is it only the Jacoco job that is failing or other jobs as well?

rhwood, 2019-03-16:
Although https://www.linkedin.com/pulse/how-resolve-utf-8-encoding-issue-jenkins-ajuram-salim is specific to Windows, it suggests that maybe the LANG environment variable needs to set.

bobjacobsen, 2019-03-16:
@rhwood thank you. LANG and LANGUAGE are set, but there seems to be others involved too. Working on it.

bobjacobsen, 2019-03-17:
@rhwood Thanks for the hint. I think it’s fixed, am running a full set to check.

bobjacobsen, 2019-03-17:
Now fixed via Jenkins config and update.

closed 2019-03-17 by bobjacobsen