JMRI/JMRI #6687: Lockup when changing LocoNet "system connection" setting

devel-bobm, 2019-03-17:
Users on jmriusers list report JMRI lockup when changing LocoNet "system connection" setting.  See [https://groups.io/g/jmriusers/message/157968](url).

icklesteve, 2019-03-17:
Is this the same issue as #6303 ?

bobjacobsen, 2019-03-17:
This is recent enough that if somebody can reproduce it, `git bisect` should locate the change that caused it. 

Can anybody reliably recreate this?

bobjacobsen, 2019-03-17:
I haven't been able to recreate it exactly as described.  A similar-but-not-exactly-the-same lockup can be traced back via `git bisect` to 
```
986a084d5944d03285fa8c7bfd6d3da992832fe0 is the first bad commit
commit 986a084d5944d03285fa8c7bfd6d3da992832fe0
Author: Bob Jacobsen <jacobsen@mac.com>
Date:   Mon Dec 31 11:57:15 2018 -0800

    more consistent LocoNet dispose handling

:040000 040000 2fe42a5f2f32151551bbb55b11c8782b0fcea9aa 57e3a7fd9bcd02488d29b0557e41bc6284098756 M	java
```
so I'm working on that now.

devel-bobm, 2019-03-17:
@bobjacobsen ,The lockup I see (Win7-64) can be traced (via `git bisect`) to the same commit as you note above.

devel-bobm, 2019-12-02:
This problem no longer exists; presumably closed with #6691.  Closing

closed 2019-12-02 by devel-bobm