JMRI/JMRI #6688: LocoNet Receive thread looping seems "uninterruptable"

devel-bobm, 2019-03-17:
Several `while` loops within `LnPacketizer.RcvHandler.run()` and `LnPacketizer.readByteProtected()` do not provide any mechanism for graceful termination.  

This appears to cause problems for `LnPacketizer.dispose()`:
```
    public void dispose() {
        if (xmtThread != null) {
            xmtThread.stop(); // interrupt not sufficient?
            try {
                xmtThread.join();
            } catch (InterruptedException e) { log.warn("unexpected InterruptedException", e);}
        }
        if (rcvThread != null) {
            rcvThread.interrupt();
            try {
                rcvThread.join();
            } catch (InterruptedException e) { log.warn("unexpected InterruptedException", e);}
        }
        super.dispose();
    }
```
The unprotected `while` loops seem to prevent `rcvThread.interrupt()` from doing anything useful, such as exiting the `RcvHandler.run()` method.  With the `RcvHandler.run()` still operating, this appears to prevent `rcvThread.join()` from joining.

This problem may relate to #6687.

devel-bobm, 2019-03-17:
Setting the purejava `serialPort's` `enableReceiveTimeout()' parameter may also help with interruptability of `LnPacketizer.readByteProtected()`.

bobjacobsen, 2019-03-17:
The proximate cause underlying #6687 is actually in JTermios, which underlies the PureJavaComm library we're now using:

```
     [java] "AWT-EventQueue-0" #14 prio=6 os_prio=31 tid=0x00007fdc7b252000 nid=0x1220f in Object.wait() [0x0000700010de9000]
     [java]    java.lang.Thread.State: WAITING (on object monitor)
     [java] 	at java.lang.Object.wait(Native Method)
     [java] 	at java.lang.Thread.join(Thread.java:1245)
     [java] 	- locked <0x00000007996275a8> (a java.lang.Thread)
     [java] 	at java.lang.Thread.join(Thread.java:1319)
     [java] 	at jmri.jmrix.loconet.LnPacketizer.dispose(LnPacketizer.java:500)
     [java] 	at jmri.jmrix.loconet.LocoNetSystemConnectionMemo.dispose(LocoNetSystemConnectionMemo.java:453)
     [java] 	at jmri.jmrix.loconet.pr3.PR3SystemConnectionMemo.dispose(PR3SystemConnectionMemo.java:194)
     [java] 	at jmri.jmrix.AbstractPortController.dispose(AbstractPortController.java:63)
     [java] 	at jmri.jmrix.AbstractSerialPortController.dispose(AbstractSerialPortController.java:319)
     [java] 	at jmri.jmrix.AbstractSerialConnectionConfig.dispose(AbstractSerialConnectionConfig.java:541)
     [java] 	at jmri.jmrix.JmrixConfigPane.selection(JmrixConfigPane.java:280)
     [java] 	at jmri.jmrix.JmrixConfigPane.lambda$new$115(JmrixConfigPane.java:211)
     [java] 	at jmri.jmrix.JmrixConfigPane$$Lambda$176/398917306.actionPerformed(Unknown Source)
     [java] 	at javax.swing.JComboBox.fireActionEvent(JComboBox.java:1258)
...
     [java] "LocoNet receive handler" #21 daemon prio=10 os_prio=31 tid=0x00007fdc79c57000 nid=0x7c07 runnable [0x0000700010eee000]
     [java]    java.lang.Thread.State: RUNNABLE
     [java] 	at jtermios.macosx.JTermiosImpl$C_lib_DirectMapping.select(Native Method)
     [java] 	at jtermios.macosx.JTermiosImpl.select(JTermiosImpl.java:376)
     [java] 	at jtermios.JTermios.select(JTermios.java:474)
     [java] 	at purejavacomm.PureJavaSerialPort$2.read(PureJavaSerialPort.java:882)
     [java] 	at java.io.DataInputStream.read(DataInputStream.java:149)
     [java] 	at jmri.jmrix.loconet.LnPacketizer.readByteProtected(LnPacketizer.java:196)
     [java] 	at jmri.jmrix.loconet.LnPacketizer$RcvHandler.run(LnPacketizer.java:234)
     [java] 	at java.lang.Thread.run(Thread.java:745)
```
The top trace is the AWT/Swing thread action that's responding to the user change of configuration by trying to shut down the existing LocoNet connection.  The second is that connection hung _inside_ the PJC read.

The `interrupt()` was issued, but nothing has happened.  (Instrumenting `LnPacketizer.readByteProtected(..)` confirms that)

We're setting up the port to timeout and loop every 10msec:
```
locobuffer.LocoBufferAdapter          DEBUG - Serial timeout was observed as: 10 enabled: true threshold: 0 enabled: false [main]
```
but that doesn't seem to be happening, at least on macOS.

devel-bobm, 2019-03-17:
It seems to me that this is caused by two things: 
- `java.io.read()` (over a purejava stream) does not return on a thread interrupt unless a ReceiveTimeout is set to be non-zero; `LnPacketizer.readByteProtected()` is stalled and cannot allow the thread to terminate;
- the various `while` loops in `LnPacketizer.RcvHandler.run()` are not sensitive to Thread `interrupt()`, so the thread cannot be terminated by `rcvThread.interrupt()`.

When I hack `AbstractSerialPortController.configureLeadsAndFlowControl()` with 'serialPort.enableReceiveTimeout(1000)`, and I hack the various `while(true)` loops in `LnPacketizer.readByteProtected()` and `LnPacketizer.RcvHandler.run()`, to be sensitive to `Thread.currentThread().isInterrupted()`, I can prevent the lock-up at `LnPacketizer.dispose()` issuance of `rcvThread.join()`.

FYI - I don't see why the timeout of 10 in `LocoBufferAdapter` doesn't work (i.e. lockup happens) when I only hack in the while loop interrupt sensitivity, but when I implement a timeout of 1000 in `AbstractSerialPortController.configureLeadsAndFlowControl()`, the lockup doesn't happen.  Perhaps there is some sequence-of-events problem that ends up zeroing-out the previously set timeout of 10...

devel-bobm, 2019-03-17:
@bobjacobsen,

A proof-of-concept solution for this, based on the rough description above, can be found in a branch of my github repository as `devel-bobm:dbm-debugLocoNetConnectionChangeLockup`.

It solves the preferences panel LocoNet "system connection change causes lockup" problem.  It is possible that it might also address some need of @Ignore within LocoNet tests.

devel-bobm, 2019-03-17:
Rolling this type of change into master would require changes for the "variant" packetizers under jmrix.loconet.

If found useful, it may make sense to roll this type of change into other system types, since some were coded using the LocoNet packetizer as a model.

bobjacobsen, 2019-03-18:
Have you tried `serialPort.enableReceiveTimeout(1000)` with real hardware?

The reason for the smaller value was to get characters in without long waits; RXTX wouldn't forward characters until the timeout expired. Perhaps PJC has a different interpretation of that parameter.

devel-bobm, 2019-03-18:
@bobjacobsen,

No, I have not tried real hardware.  This was just as "proof of concept" that "Timeout" could help solve the "interruptability" issues in low-level purejava and java methods that seemed to contribute to the lockup.

devel-bobm, 2019-03-18:
@bobjacobsen,

Life is going to prevent me from working on this in any predictable way for at least a few days.  I may get a chance to look into it soon, but probably not for several days...

devel-bobm, 2019-03-28:
@bobjacobsen,

> Have you tried serialPort.enableReceiveTimeout(1000) with real hardware?
>
> The reason for the smaller value was to get characters in without long waits; RXTX wouldn't forward characters until the timeout expired. Perhaps PJC has a different interpretation of that parameter.

I have confirmed that this solution does not work in-system.  I find that any character generated by some other LocoNet device seems to get thru in a timely fashion, but any TX character from JMRI seems to be delayed by the "ReceiverTimeout" delay.

As such, any approach which increases the timeout beyond `serialPort.enableReceiveTimeout(10)` is likely to be unacceptable.  I am closing this issue and deleting my personal repository branch `devel-bobm:dbm-debugLocoNetConnectionChangeLockup` .

closed 2019-03-28 by devel-bobm