JMRI/JMRI #6693: Connection type list is no longer in desired order in preferences

bobjacobsen, 2019-03-17:
The set of possible connections (at least under Digitrax LocoNet) in the preferences used to be in a specific order:

[image]

At some recent point, it switched to alpha order (which is not a good one; many of those connection types are uncommon, and should be further down):

[image]

The problem occurred sometime between v4.15.3 and v4.15.4. Unfortunately, `git bisect` gets tangled up in non-functional commits when trying to work between those two.  It's best guess is f17466f6d51a67c0182feafa919db4e31742b7b1 as the first bad commit, but that doesn't seem to be right.  The only change that seems related is:
```
--- a/java/src/jmri/jmrix/loconet/LnStreamConnectionTypeList.java
+++ b/java/src/jmri/jmrix/loconet/LnStreamConnectionTypeList.java
@@ -11,7 +11,7 @@ import org.openide.util.lookup.ServiceProvider;
  * @author Kevin Dickerson Copyright (C) 2010
   *
  */
-@ServiceProvider(service = StreamConnectionTypeList.class)
+@ServiceProvider(service = ConnectionTypeList.class)
 public class LnStreamConnectionTypeList implements jmri.jmrix.StreamConnectionTypeList {
 
     public static final String DIGITRAX = "Digitrax"; // NOI18N
```
but I don't really see why this might be relevant.

pabender, 2019-03-17:
I don’t see how that change could have impacted this.  The PR that was part of was adding connection confit options to the remote serial port on an XBee.  

The specific change you show above was making it so the Stream connections don’t show in the regular connection list.

bobjacobsen, 2019-03-18:
I agree it's unexpected, but I've confirmed that's the change that causes it (after a clean compile).  

It looks to me like having `LnStreamConnectionTypeList` serve the `ConnectionTypeList` is causing confusion with the same service provided by `LnConnectionTypeList`.  The order is set in the `getAvailableProtocolClasses()` method of `LnConnectionTypeList`.  `LnStreamConnectionTypeList` `@Overrides` that without calling the original (super) method.

pabender, 2019-03-18:
@bobjacobsen, LnStreamConnectionTypeList does not inherit from LnConnectionTypeList.  LnStreamConnectionTypeList only implements the StreamConnectionTypeList interface.

In other words, there is no method in super to call.  Also, if there was a supper method to call calling it would defeat the purpose of the separate interface.  This is being used so that the stream interface din’t Appear in the normal list of connections and only stream interfaces appear in the context sensitive locations where they can be used.  ( a stream connection has to be used in conjunction with an IOStream )

bobjacobsen, 2019-03-18:
Clearly, I don’t understand what stream connections are supposed to be. Never-the-less, changing that one line is interfering with the regular connection configuration process. Can we just put it back? What configuration setup uses it?

pabender, 2019-03-18:
Right now It is very specifically being used by the XBee Node configuration to configure connections on a remote XBee node.

pabender, 2019-03-18:
I would have to go back through the entire PR that this was a part of to see if we can change it back.  There was a reason I changed it in the first place.

pabender, 2019-03-18:
I think we might be able to remove the StreamConnectionTypeList methods altogether.  Part of the changes in the PR in question included code to filter the normal list and the list used by the XBee code.

Let me experiment and verify.

pabender, 2019-03-18:
@bobjacobsen see #6698.

I did a quick check.  The XBee configuration correctly only shows the Stream connection options and the Loconet options for a standard connection appeared to be in the specified order with the stream connection option omitted.

bobjacobsen, 2019-03-18:
@pabender Thanks!  That #6698 PR looks good, at least in my tests (without real hardware, just dummy serial ports)

PR #6698 (2019-03-18): remove jmri.jmrix.StreamConnectionTypeList
Also move connections defined in such lists to the standard ConnectionTypeList classes.

Fixes #6693
closed 2019-03-18 by pabender