JMRI/JMRI #6718: Layout Editor File -> Store menu item behaves differently?

bobjacobsen, 2019-03-22:
From jmriusers:
> On Mar 21, 2019, at 10:21 PM, Don Weigt <dweigt47@gmail.com> wrote:

> Do you mean the Save Panels command in the Layout Editor window? It would be great if that always did save the table changes. But, I'm pretty sure it doesn't always.

> I tried very conscientiously to use the Save Panels command as I was editing the turnout table a few weeks ago. When I closed and reopened JMRI, the table changes hadn't been saved. It seemed clear to me that wasn't enough to save the tables associated with the LE panel I was working on. That's when I found the other way of saving the tables from the Panel Pro window.

it looks like the store menu item in the Layout Editor's File menu invokes `StoreXmlUserAction`, while (maybe) everywhere else invokes `jmri.configurexml.SaveMenu`.

`StoreXmlUserAction` comes with the warning:
```
 * Note that this does not store preferences, configuration, or tool information
 * in the file. This is not a complete store! See {@link jmri.ConfigureManager}
 * for information on the various types of information stored in configuration
 * files.
```

That means that this store isn't storing the tables!  

- Should we keep that?  If so, should be rename it to make that clear?
- Or should we make it use the same code as all the others?

Pugwash1, 2019-03-22:
We always use the file menu store  in the LE, and it saves everything. We ould have noticed!!
However, if you edit a line in a table, and don't move focus somewhere else, the change doesn't always work.

KenC57, 2019-03-22:
I never noticed a 'Store' in the LE. I would think using the typical store
would be the expected behavior.
-ken c

Pugwash1, 2019-03-22:
I tried the LE File->Store and the Tables File->Store->configuration and panels
They both finish up executing StoreAll in configureXMLManager, that store everything except "Preferences", and a few odds and ends. Reading the original thread, he says he used PanelPro->Edit->Preferences->Save, which doesn't save panels.....or tables..... Was the user using Tables->Store->Configurations, confusing configuration & preferences? Its all very odd.

bobjacobsen, 2019-03-22:
So, at a minimum, it sounds like I should:
- Fix the comments in `StoreXmlUserAction` and kin to describe what they really do
- Change LayoutEditor to use the everybody-else convention of  `jmri.configurexml.SaveMenu`

Anything else? 

Will put together a PR.

Pugwash1, 2019-03-22:
@bobjacobsen That's the crux of it. The fact that if you change a combo box in tables and then do store,without clicking anywhere else, the combo box field is not changed is something to do with ValueEditor.java which saves text fields without pressing enter, but doesn't service other types. I'm not too sure how its supposed to work.

closed 2021-04-14 by stale[bot]