JMRI/JMRI #6723: json websockets no longer processing changes from client

mstevetodd, 2019-03-25:
Sometime after 4.15.3, the json websocket service began ignoring incoming change requests. Example, can be tested on http://jmri.mstevetodd.com/json: 
`{"type":"sensor","data":{"name":"LS149","state":2}}`
should change the state of sensor LS149 to 2 (active). It does not. Also tried with turnout:
`{"type":"turnout","data":{"name":"LT93","state":2}}`
Note: these commands properly return the current state, so if already set to 2, try again with 4 (inactive).
This problem was first reported here: https://groups.io/g/jmriusers/topic/30750047

rhwood, 2019-03-25:
If you add an explicit method to those messages does it then work as expected?

rhwood, 2019-03-26:
Actually, the JSON WebSockets are behaving appropriately; absent an explicit request, performing a GET (an idempotent action) *is and has been* the default message behavior for some time. That sensors and turnouts no longer treat GETs as POSTs is the correct behavior; the web pages need to be updated to handle that.

mstevetodd, 2019-03-26:
Really? This syntax has been in use since you changed panel.js and wrote jquery.jmri.js to use websockets instead of xmlio many years ago. All of the examples shown here: http://jmri.org/help/en/html/web/JsonServlet.shtml show this syntax.
Can you provide an example of the explicit message syntax you're saying is now the default? Is this syntax in use in the JMRI code anywhere?

rhwood, 2019-03-26:
The default method has been a GET since 4.11.3, and was undefined before then (meaning that different types behaved differently, which could result in inconsistent behavior). The Javadocs for the JSON servers read that adding the `method` value for POST is optional, but even before 4.11.3, that was inconsistently true, so the docs have been inconsistent with implementation (i.e., either correct or wrong depending on the `type` value in the JSON message) for some time. I'll update the web client, release notes, and Javadocs, and write some tests to ensure that since the behavior is now consistent, it remains consistent (GETs should be idempotent).

PR #6777 (2019-04-18): Improve JSON services documentation and other enhancements
This was originally intended to just improve the JSON services documentation and fix the JSON jQuery Javascript library, but has expanded a little beyond that:
- [x] Consolidate JSON protocol documentation in one place to remove contradictory documentation
- [x] Make JSON jQuery library follow JSON protocol documentation (fixes #6723, fixes #6765)
- [x] Ensure JSON services do not listen to unrequested objects (fixes #6735)
- [x] Allow property change listeners for NamedBeans to listen to only specific properties
- [x] Address some war [... cut]
closed 2019-04-18 by bobjacobsen