JMRI/JMRI #6728: AbstractManager.register doesn't give an exception or warning if system name exists

danielb987, 2019-03-27:
Due to a bug in my code, I tried to register a new bean with an existing system name and therefore it never got stored in the manager, which caused a strange error.

I have now found my bug and knows why my code generated the wrong system name. But I think that it would be a good idea if AbstractNamedBean.register() would give some notification that the new bean cannot be registed in the manager due to a  duplicate system name.

The preferred solution for me would be throwing an InvalidArgumentException if the system name is already registered in the manager, but maybe it's a bad idea?

bobjacobsen, 2019-03-27:
This came up recently:  https://jmri-developers.groups.io/g/jmri/message/736

I think it's fine to throw the exception; I'd prefer that it also log.error as this is a pretty severe error in the code.

danielb987, 2019-03-27:
#6729 tries to fix this. But there are errors that I'm not sure on how to fix.

rhwood, 2019-03-27:
However, it is not fine to throw an exception if registering the same NamedBean twice; we already allow that and test for that situation and there is no valid reason not to continue to allow that (the same way HashSets allow the same object to be added twice without changing the set the second time without error).

danielb987, 2019-03-30:
#6729 is merged so closing this.

closed 2019-03-30 by danielb987