JMRI/JMRI #6737: Repetition of signal sections in Layout Editor .xml

Paul-T, 2019-03-28:
I am developing my panel for eventual automation;  so far the file is around 7500 lines, as seen in Notepad++

Ever since I added working Signal Heads I have a frequent error on reload after other minor modifications which were done in the GUI and saved.

Typically the error is:
Parse error  bla bla at line 2000 duplicate signal head MH2  bla bla.
On examining the  .xml, I find not only Sig 2 duplicate entry but all 8 signals in a block.
Easily edited out but still annoying.

This has happened a dozen or more times but is not exactly reproducible as not all mods trigger this effect.  Thus it is difficult to report it as a bug due to inconsistent behaviour. It may be caused by some error of mine. 
I note that Martin Booker saw the same issue in 4.12 as reported on the Forum thread of same name.
I am using JMRI 4.15.4  but same error in versions over last 3 months.  Java 1.8.0_191.   Profile is for Merg CBus ethernet connection

bobjacobsen, 2019-03-28:
Could you attach a copy of the file with the duplicates please? You might need to add a .txt to the name

Paul-T, 2019-03-29:
Here is a sample file with the fault
[Pseudo Dartmouth_Mk45.xml.txt](https://github.com/JMRI/JMRI/files/3021436/Pseudo.Dartmouth_Mk45.xml.txt)

bobjacobsen, 2019-03-29:
Exactly what error do you see with that file?  I don't see any "MH2" references in it, and only one occurrence of "Sig 2".  (I don't have a similar enough configuration to load the file and see that way)

bobjacobsen, 2019-03-29:
Was able to cobble something together and find that 'Sig2 T&amp;DR Up Distant' is duplicated.

The '&amp' is HTML coding for '&'.  Not entirely sure why that's happening, but it might be related.

Do you ever see other signal logic duplicated?  I.e. see some that don't have a '&' in them?

Paul-T, 2019-03-29:
_Do you ever see other signal logic duplicated? I.e. see some that don't have a '&' in them?_

I had only recently started adding Signal logic in that file so it is incomplete. I have not looked for logic duplicates. When the signal elements duplication is removed, the file loads to JMRI without parse errors

bobjacobsen, 2019-03-30:
Thanks.  I've found one thing that can cause this and the fix for that is in PR #6745, will be in the next test release and next (number 3224 or later) development build here: http://jmri.tagadab.com/jenkins/job/Development/job/Packages/  But that might not be the only thing, so I'd appreciate you keeping an eye out.

Paul-T, 2019-03-30:
Thanks Bob.

Downloaded 3224 and will test over next few days.

I have many mods to do, including signal masts and SML so that will give 
it a thrashing.

closed 2021-04-14 by stale[bot]