JMRI/JMRI #6739: New decoder Definition

kestrelmc, 2019-03-29:
Could someone please supply decoder definition for ESU LokSound v5.00
Thank you
Terry

dheap, 2019-03-29:
> @kestrelmc Could someone please supply decoder definition for ESU LokSound v5.00

This is in-hand but it is not a small job. The number of CVs has _more than doubled_ from the V4 series, many existing CV numbers have been changed, new features have been made added and the special Java code required to create and display the ESU Function Map will require significant rewriting.

I will make it available as soon as possible, but it certainly won't be in the next test release. The documentation has only recently been released by ESU  and I don't yet have access to V5 hardware for testing.

kestrelmc, 2019-03-29:
Thanks Dave

 

I look forward to it thank you.

I have an Auscision Paybus with NO DOCUMENTATION as well an Auscision 442 that has minimum documentation that does not easily lend itself to alteration that both have v5 decoders in them.

 

Terry

kestrelmc, 2019-03-29:
Dave

 

If it is of any help I can test it models I have with v5 decoders.

 

Terry

tualatin, 2019-04-15:
It is very useful, when you add support for Piko SmartDecoder, thank you!

silverailscolo, 2019-04-15:
@tualatin asking about a new Piko decoder has nothing to do with this issue on ESU LokSound (which is also just a request). Better join issue #6469. You could offer to assist in testing the definition that I might adapt from similar Piko decoder models, since you own one, I guess.

closed 2021-02-23 by silverailscolo