JMRI/JMRI #6741: Adding a multi IO device

danielb987, 2019-03-29:
For background, see the device discussed in https://github.com/JMRI/JMRI/issues/6238#issuecomment-444668260 and forward.

I'm thinking of creating an Arduino project where the Arduino is connected to LocoNet and has some digital I/O, some analog I/O and some string I/O. For example, a couple of ADC-converters and a text display.

In JMRI, I'm thinking of creating the interface `jmri.MultiIOCard` and the concrete class `jmri.jmrix.loconet.LnMultiIOCard`.

Each MultiIOCard will be a named bean and there will be a manager for them and a beantable. And each I/O will have their own bean, for example turnout, sensor, AnalogIO or StringIO.

I want this to be configurable by the user, so that the user can download the source code for the Arduino and configure the card for their own purpose, or for a manufacture to create a configuration for their card. And that the user selects the type of card when creating the MultiIOCard bean in the beantable, like the user selects the type of signal when the user adds a signal head.

For the configuration of the types of cards, I'm thinking of an xml file that looks something like this (written in meta-code):
```
<multi_io_device>
    <name> {a name of this multi_io_device type} </name>
    <protocol> {loconet / ....} </protocol>
    <start_addresses>
        <address>
            <name> Bit data </name>
            <data_types>bit</data_types>
        </address>
        <address>
            <name> Byte data </name>
            <data_types>integer,decimal,string</data_types>
        </address>
    </start_addresses>
    <bits>
        <bit_data>
            <name> {a string} </name>
            <offset> {offset of address} </offset>
            <type> {turnout / sensor / ...} </type>
            <direction> { in/out/both } </direction>
        </bit_data>
        ....
    </bits>
        <integer_data>
            <name> {a string} </name>
            <offset> {offset of address} </offset>
            <size> {num bytes} </size>
            <encoding> {binary / string} </encoding>
            <direction> { in/out/both } </direction>
        </integer_data>
        ....
    </integers>
    <decimals>
        <decimal_data>
            <name> {a string} </name>
            <offset> {offset of address} </offset>
            <encoding> {binary / string} </encoding>
            <size> {num bytes if encoding=string; 4 bytes if encoding=binary} </size>
            <decimals> {num decimals if encoding=string} </decimals>
            <direction> { in/out/both } </direction>
        </decimal_data>
        ....
    </decimals>
    <strings>
        <string_data>
            <name> {a string} </name>
            <offset> {offset of address} </offset>
            <ending> {pad_zero, pad_space} </ending>
            <size> {num bytes} </size>
            <encoding> {ascii7, utf8, …} </encoding>
            <direction> { in/out/both } </direction>
        </string_data>
        ....
    </strings>
</multi_io_device>
```

Here, each bit_data, integer_data, and so on is one I/O. So there may be several of each of these. The start_addresses section is to define which start addresses are relevant. For LocoNet, I think turnouts and sensors should be used for bit data and OPC_PEER_XFER byte transfer for byte data. But for other protocols, other addresses may be recommended instead.

Does this seems reasonable or am I out on deep water?

bobjacobsen, 2019-03-29:
Sounds like a very neat capability!

But there's one thing I don't understand:

> Each MultiIOCard will be a named bean and there will be a manager for them and a beantable

What's the value in having `MultiIOCard` be a `NamedBean` etc?

Our usual approach is to manage that kind of thing via data stored in the preferences file and maintained via preferences-type panes.  C.f. the C/MRI setup.  Since the configuration of the board is a very basic thing, that ensures that it doesn't change during operation, etc.

danielb987, 2019-03-29:
Are you suggesting configuring MultiIOCards the same way as LocoNet LocoIO devices (menu choice LocoNet / Configure LocoIO (Legacy)) ?

If I have one LocoIO card on the layout, that works just fine. But if I have three, four och five LocoIO cards, I start to get into trouble since they only have addresses, not names. As soon as I have several of the same card on the layout, I want to give the devices their own names in order to remember which is which.

If I store the information in the preferences, will I then be able to store the name of the card?

danielb987, 2019-03-30:
@bobjacobsen 
I apologize. I haven't worked with C/MRI or any other system except LocoNet so I wasn't aware of what you where talking about. But I will look into C/MRI and its configuration. It may work fine for me.

bobjacobsen, 2019-03-30:
No apology necessary!  Thanks for thinking about this.  Sounds like a good capability. 

There are a couple of different ways to configure things like this within JMRI, and each has advantages and disadvantages.  Preferences main advantage is that they’re in place _early_; you’re guaranteed that they’re set up with the connection and therefore before panel files or scripts can try to reference (or create) Sensors, Turnouts, etc. But they’re a bit of a pain to set up.

> I apologize. I haven't worked with C/MRI or any other system except LocoNet so I wasn't aware of what you where talking about. But I will look into C/MRI and its configuration. It may work fine for me.

pabender, 2019-03-31:
> Are you suggesting configuring MultiIOCards the same way as LocoNet LocoIO devices (menu choice LocoNet / Configure LocoIO (Legacy)) ?
>
> If I have one LocoIO card on the layout, that works just fine. [...]

I think you may be looking at the wrong model for configuration.

Instead of looking at LocoIO devices, look at how the rr-circuits switch-man and other similar devices are configured.  This uses a programmer and stores the configuration like you would store configuration of any roster entry.

Alternately, look at how OpenLCB node configuration is stored in the node itself as XML data.

Where I am having trouble with all of this is you seem to be trying to commingle the configuration parameters of the node with the operation of that device.  That seems to be an overly complicated solution when if you break it into the two pieces, we have existing solutions that should work for what you want to do.

closed 2021-04-14 by stale[bot]