JMRI/JMRI #6744: Line Separator Character Handling Problems in Mixed OS JMRI Setups

mesheets, 2019-03-29:
In a JMRI setup with multiple devices running a mix of operating systems, it appears some problems arise when software running on one OS platform attempts to communicate with software running on another OS platform.

For example, in [java/src/jmri/jmrit/withrottle/DeviceServer.java line 110](https://github.com/JMRI/JMRI/blob/e3bf93e81312917a29144c3b10911c73bbd75e77/java/src/jmri/jmrit/withrottle/DeviceServer.java#L110), the line separator character sequence is set based on the platform on which that code is executing, such as the following: `System.getProperty("line.separator")`

While this works fine when all of the JMRI-connected systems are running on a similar platform (i.e. platforms that all define the same line separator character sequence), in a mixed platform setup, 

Attached is a [log file from a JMRI server instance with WiThrottle running on a Linux system](https://github.com/JMRI/JMRI/files/3024817/jmriSystemConsole.log), which shows connections from Windows devices being rejected.

If the JMRI server instance with WiThrottle is instead running on a Windows device, then the _**exact**_ same Windows devices whose connections were rejected above are able to connect _without_ being rejected by the JMRI server instance running on Windows.

If JMRI is going to use the line separator character sequence defined by the system on which it is running (which might not be an issue in and of itself), then when receiving and processing input, it should be able to handle any of the three most common line separator character sequences (‘\r\n’ or ‘\n’ or ‘\r’) that it might receive.  This would probably also be the best approach for maintaining compatibility with existing systems (as opposed to forcing a standardization on one particular line separator character sequence).

I am not yet sure how broad of an issue this might be in JMRI.  While many usages of  appear to be (understandably) related to the user interface, I did find a usage with the [CAN CbusSend](https://github.com/JMRI/JMRI/blob/1ed51f3e68c4e343c78d8f7048f53a62ecda434a/java/src/jmri/jmrix/can/cbus/CbusSend.java#L26) functionality.

BHoffman351, 2019-03-30:
>  it should be able to handle any of the three most common line separator character sequences (‘\r\n’ or ‘\n’ or ‘\r’) that it might receive.

It should already behave that way. DeviceServer does not mandate a specific line termination but uses readLine() to handle the incoming packets.
From BufferedReader.readLine():
Reads a line of text. A line is considered to be terminated by any one of a line feed ('\n'), a carriage return ('\r'), or a carriage return followed immediately by a linefeed.

Regards,
Brett

mesheets, 2019-03-30:
> It should already behave that way.

That's what I would expect, and that is what the [documentation](http://www.jmri.org/help/en/package/jmri/jmrit/withrottle/Protocol.shtml) would also seem to indicate:
> Each of these commands is plain text, with a newline character at the end (0A hex).

and
> Other implementations of the WiThrottle protocol will terminate the command with a carriage return (0D hex) followed by a newline.

However, use “System.getProperty("line.separator")” noted above at the link to the source code seems to indicate that the system will behave differently depending on the platform on which it is running.

Additionally (and as can be traced from the log file), an IOException is thrown and then [caught by the IOException handler](https://github.com/JMRI/JMRI/blob/e3bf93e81312917a29144c3b10911c73bbd75e77/java/src/jmri/jmrit/withrottle/DeviceServer.java#L369).

With the identical input, JMRI instances running on Linux are throwing this IOException but JMRI instances running on Windows are not.

BHoffman351, 2019-03-30:
> However, use of “System.getProperty("line.separator")” noted above at the link to the source code seems to indicate that the system will behave differently depending on the platform on which it is running.

That separator is only appended onto packets being sent from JMRI to another device. It is not used to identify incoming message separation points.

> With identical input, JMRI instances running on Linux are throwing this IOException, but JMRI instances running on Windows are not.

That line separator has been in use since the inception of the WiThrottle protocol. It has not been an issue with Linux systems in the past and appears to behave properly on the Pi-Sprog. Unless some method behavior has changed, JMRI sends the system-preferred line ending, but using readLine() it receives packets separated by any line ending.

Regards,
Brett

danielb987, 2019-03-30:
@mesheets 
> Attached is a [log file from a JMRI server instance with WiThrottle running on a Linux system](https://github.com/JMRI/JMRI/files/3024817/jmriSystemConsole.log), which shows connections from Windows devices being rejected.

What device is you using to communicate with JMRI and which software on this device? WiThrottle is on iPhone and EngineDriver is on Andriod, unless any of them has been ported to Windows.

danielb987, 2019-03-30:
Sorry, I found it.
https://www.microsoft.com/en-us/p/train-throttle/9nblggh4vf5j

mesheets, 2019-03-30:
From the "[Installing JMRI on Raspbian](http://jmri.org/install/Raspbian.shtml)" install doc page, I am working with the [M Steve Todd image](https://mstevetodd.com/rpi) and using that as the JMRI+WiThrottle server.

As linked on the default port "12080" website, I am using the "[Train Throttle](https://www.microsoft.com/store/apps/9NBLGGH4VF5J)" app, which is available for Windows.

mesheets, 2019-03-30:
@BHoffman351 

>>However, use of “System.getProperty("line.separator")” noted above at the link to the source code seems to indicate that the system will behave differently depending on the platform on which it is running.

>That separator is only appended onto packets being sent from JMRI to another device. It is not used to identify incoming message separation points.

The documentation notes the following regarding responses from JMRI:
> lines sent back from JMRI are terminated by a newline.

Should client or embedded devices connected to JMRI be setup to handle potential protocol ambiguity depending on which platform the JMRI is running?  From the documentation, my impression based on the context (including the preceding sentence noting that a newline character is “0A hex”) is that JMRI should only be sending a ‘\n’ as the line separator character sequence, regardless of the host platform on which it is running.

Have some clients that have been developed and tested against JMRI instances running Windows systems perhaps come to expect a CR/LF (instead of just the LF that the documentation seems to describe)?

mesheets, 2019-04-03:
Another console log from JMRI on Linux, with WiThrottle "debug"-level turned on ("log4j.category.jmri.jmrit.withrottle=DEBUG" as per [here](http://www.jmri.org/help/en/package/jmri/jmrit/withrottle/Protocol.shtml)), though it doesn't seem to reveal much more than the log file posted earlier:
  - [jmri-SystemConsole-Debug.log](https://github.com/JMRI/JMRI/files/3036852/jmri-SystemConsole-Debug.log)

.
Wireshark files showing communication differences with JMRI when running on Linux versus when running on Windows, which appears to confirm that what a connecting client might receive from JMRI will differ depending on JMRI's host platform:
  - [jmriOnLinux-Wireshark.pcapng.zip](https://github.com/JMRI/JMRI/files/3036868/jmriOnLinux-Wireshark.pcapng.zip)
  - [jmriOnWindows-Wireshark.pcapng.zip](https://github.com/JMRI/JMRI/files/3036869/jmriOnWindows-Wireshark.pcapng.zip)

BHoffman351, 2019-04-03:
> Wireshark files showing communication differences with JMRI when running on Linux versus when running on Windows, which appears to confirm that what a connecting client might receive from JMRI will differ depending on JMRI's host platform:

That is correct.

> Should client or embedded devices connected to JMRI be setup to handle potential protocol ambiguity depending on which platform the JMRI is running? 

Yes.
Not ambiguity, but clients should handle any variety of newline - \n, \r, or \r\n. "Newline" can refer to "line feed", "carriage return", or a combination such as "carriage return - newline".

> From the documentation, my impression based on the context (including the preceding sentence in that documentation, which notes a newline character is “0A hex”) is that JMRI should only be sending a ‘\n’ as the line separator character sequence, regardless of the host platform on which it is running.

JMRI is sending the system specific newline.  This has worked fine since November of 2009. You quoted the document as:

>> Other implementations of the WiThrottle protocol will terminate the command with a carriage return (0D hex) followed by a newline.

This states that "\r\n" is valid.

> Have some clients that have been developed and tested against JMRI instances running on Windows systems perhaps come to expect a CR/LF (instead of just the LF that the documentation seems to describe)?

Both Engine Driver and WiThrottle respect all newline options.

pabender, 2019-04-03:
Something that was drilled into my head when I was first learning networking was this principle:
“Be liberal in what you accept,  but conservative in what you send out.”

What that means on this particular case is that JMRI really should be accepting any variation of end of line that may be sent to it, but it should only send out ‘\n’ if the documentation says ( as was quoted earlier) “lines sent back from JMRI are terminated by a newline.“

mesheets, 2019-04-03:
@BHoffman351 
> This states that "\r\n" is valid.
That part of the documentation refers to throttle clients connecting to JMRI, _not_ to JMRI+WiThrottle server itself.  Thus, per the documentation, a "\r\n" from the JMRI+WiThrottle server should _not_ be sent.

The following the part of the documentation that refers to JMRI+WiThrottle server:
> lines sent back from JMRI are terminated by a newline.

In turn, a "newline" is defined by the sentence that immediately precedes the above quote in the documentation:
> a newline character at the end (0A hex)

Also,
> This has worked fine since November of 2009
I'm not quite as certain.  I've seen reports of problems going back to at least 2016, but I don't think the connection was ever made regarding hybrid platform scenarios and that that might actually be part of the problem.

icklesteve, 2019-04-03:
> While many usages of appear to be (understandably) related to the user interface, I did find a usage with the [CAN CbusSend](https://github.com/JMRI/JMRI/blob/1ed51f3e68c4e343c78d8f7048f53a62ecda434a/java/src/jmri/jmrix/can/cbus/CbusSend.java#L26) functionality.

This is purely for UI, feeding back a String to a JTextArea, with no effect on the CAN frame sent,
Steve.

BHoffman351, 2019-04-03:
> Thus, the documentation does _not_ appear to read "\r\n" as valid from JMRI+WiThrottle server; I am interpreting that based on the the distinction between "Throttle" and "JMRI" that is stated earlier in the documentation: "Throttles (or other applications) can locate JMRI…" and "Your application then sends commands to JMRI", thus seeming to read that JMRI should be able to accept either "\n" or "\r\n" from a connecting throttle client. Perhaps there might be opportunity to more clearly articulate some of this documentation?

The documentation was written by an outside source.  I am attempting to clarify that currently.

We seem to be having a drawn-out argument over semantics.  What is it you are looking to have happen?  WiThrottle and Engine Driver are extensively tested with JMRI on Windows, Mac, and Linux.  They are also tested against the LNWI and MRC Prodigy WiFi.  It sounds like you are having an issue with something outside of this range.

pabender, 2019-04-03:
Sent from my iPhone

> On Apr 3, 2019, at 11:15 AM, Brett Hoffman <notifications@github.com> wrote:
> We seem to be having a drawn-out argument over semantics.
> 

If we are arguing over semantics, then that means the documents are ambiguous.  The goal is to make the documentation of the protocol as unambiguous as possible so that others get the clearest picture possible so they can implement their own client or server.

The best approach is typically to be as specific as possible in the documentation.  In this case, state the specific line ending expected and use that line ending explicitly.  I.e.  do not allow operating system specific line endings to be used.

danielb987, 2019-04-03:
@pabender 
I agree on what you are saying. But if JMRI changes the output to only use \n and the clients starts to assume that, then that may cause problems with other Withrottle servers that are using another line ending.

Since different line endings has been used a long time, it seems better to change the documentation to say that all three line endings must be accepted by the client.

pabender, 2019-04-03:
> @pabender
> I agree on what you are saying. But if JMRI changes the output to only use \n and the clients starts to assume that, then that may cause problems with other Withrottle servers that are [...]

And that is the second part of the philosophical statement on this, ”be liberal in what you accept”.

The point being you should take the strictest interpretation when you send data out, but the most liberal when you take data in.  In other words, change the code in JMRI so the operating system doesn’t impact the exact character sequence we send, but leave in lace the code that interprets any 

We can’t control what other software does, but we can control what JMRI does.

Even if the documentation does change, changing the operating system should not change the end of line character we send in any network protocol.

bobjacobsen, 2019-04-03:
This Issue isn't about documentation.  It's about "Line Separator Character Handling Problems in Mixed OS JMRI Setups"   So far, we haven't actually established what those "Problems" are.  

So let's go back to what matters here:  @Mesheets You've pointed to "Windows" devices trying to connect as an issue.  What Windows devices?  Running what code?

danielb987, 2019-04-03:
> So let's go back to what matters here: @mesheets You've pointed to "Windows" devices trying to connect as an issue. What Windows devices? Running what code?

The answer is here:
https://github.com/JMRI/JMRI/issues/6744#issuecomment-478198483

> From the "Installing JMRI on Raspbian" install doc page, I am working with the M Steve Todd image and using that as the JMRI+WiThrottle server.

> As linked on the default port "12080" website, I am using the "Train Throttle" app, which is available for Windows.

BHoffman351, 2019-04-04:
It appears that the real issue is that the Train Throttle app does not accept variations on the newline (end of line) sequence.  It only accepts \r\n as valid.  Running JMRI on a Windows system will send a newline sequence of \r\n which will work in the Train Throttle app.  Running JMRI on a *nix system, the newline is sent as \n which is not being recognized by the Train Throttle app.  If we change the newline in JMRI to a fixed \n character, Train Throttle goes from working when connected to a Windows host to working on nothing.

I did not write the protocol document that is being referenced.  When I read through it in the past, this did not seem ambiguous to me because I know what it does.  Sometimes it takes a different set of eyes to interpret the description in a different way.  Now that I see that it is being interpreted outside of how it functions, I have made a pull request #6766 that (hopefully) includes clarification.

I am not arguing the point:

> “Be liberal in what you accept, but conservative in what you send out.”

I do agree with that statement.  If I were to do it over again, I probably would have used a single fixed character - likely the "line feed" (\n).  However, changing the protocol at this point will simply escalate the problem.

pabender, 2019-04-04:
@BHoffman351 

My real point is that JMRI needs to be consistent in what is sent regardless of operating system.

If we change the JMRI server implementation so that it uses `\r\n` That makes the TrainThrottle app work everywhere doesn’t it?

`\r\n` Is certainly used by other protocols, such as http, for end of line delimiters, so that works as far as I am concerned.

BHoffman351, 2019-04-04:
> If we change the JMRI server implementation so that it uses `\r\n` That makes the TrainThrottle app work everywhere doesn’t it?

How do we know that Windows-only hosted server was not the intended function of Train Throttle?

bobjacobsen, 2019-04-04:
Nobody knows how many other client implementations out there are just looking for \n. They would break if JMRI on their Linux or macOS hosts changed. 

I think we should document that JMRI can send any sequence and that clients need to accept that, then move the onus over to the client programs. 

Given the installed base, I wouldn’t change what it’s sending now.

pabender, 2019-04-04:
Sent from my iPhone

> On Apr 3, 2019, at 9:16 PM, Brett Hoffman <notifications@github.com> wrote:
> 
> If we change the JMRI server implementation so that it uses \r\n That makes the TrainThrottle app work everywhere doesn’t it?
> 
> How do we know that Windows-only hosted server was not the intended function of Train Throttle?
> 
How does that even matter?

It very well could be the developer never bothered to test the functionality with a non-windows server, so it was an oversight and not intentional.

The simple fact is we have users who want to use Train Throttle with a non-windows server, and our being inconsistent based on the operating system prevents that.  Regardless of choices some other developer made, we need to fix that.

bobjacobsen, 2019-04-04:
> Regardless of choices some other developer made, we need to fix that.

Do you have a way to do that at this point without risking breaking others?

pabender, 2019-04-04:
>> Regardless of choices some other developer made, we need to fix that.
> Do you have a way to do that at this point without risking breaking others?

Is there an absolutely foolproof way to ensure we don’t break anything, no, but the situation we have now is untenable. 

The current situation is that JMRI implements the reference implementation of the Withrottle Server and that reference implementation produces different results based on what environment it is running in.

Any way we look at this, there are going to be future compatibility issues with line endings.  If we make a choice of line endings, those compatibility issues eventually go away.  Not making a choice allows new client software to be introduced that is incompatible with one operating system or another.

From a long term support view, making a choice of line ending is preferable, even if there is a some short term pain involved.

BHoffman351, 2019-04-04:
> > How do we know that Windows-only hosted server was not the intended function of Train Throttle?

> How does that even matter?

Because if it was, then you are trying to fix something that the developer does not consider a problem.

> It very well could be the developer never bothered to test the functionality with a non-windows server, so it was an oversight and not intentional.

Likely, yes.

> The simple fact is we have users who want to use Train Throttle with a non-windows server, and our being inconsistent based on the operating system prevents that. Regardless of choices some other developer made, we need to fix that.

No we don’t.  If any of the documentation was followed, the newline the Train Throttle app appears to be looking for would not have been the \r\n.  The argument was made that the documentation leads to the use of \n.  JMRI should not be changed.  How do you propose changing the MRC Prodigy WiFi or the Digitrax LNWI?  The correct fix is for Train Throttle app’s developer to make the change - _IF THAT IS WHAT HE WANTS TO DO._  It would be impolite to insist.

mesheets, 2019-04-04:
> How do we know that Windows-only hosted server was not the intended function of Train Throttle?

The only known obstacle to using Train Throttle with any JMRI instance running on any platform was the unknown difference that JMRI behaved differently depending on which platform it was running.  If behavior differs depending on the platform, then testing becomes a more complex proposition, as it now needs to be tested against multiple platforms.

For commercial endeavors, such testing becomes part of the cost of doing business.  However, for developers who might be involved with JMRI at more of a hobbyist level, is there an expectation that they should now have to the equipment and setups necessary to regression against multiple platforms?

BHoffman351, 2019-04-04:
> From a long term support view, making a choice of line ending is preferable, even if there is a some short term pain involved.

Spoken by the person who does not have to provide the support.

pabender, 2019-04-04:
> > From a long term support view, making a choice of line ending is preferable, even if there is a some short term pain involved.
> 
> Spoken by the person who does not have to provide the support.

I provide support for this as well.

There is risk either way.  The risk is just different if we make a choice vs not making a choice.

mesheets, 2019-04-04:
For either a commercial or hobbyist developer, if moving forward there is a need to regression against multiple platforms instead of just a single, standard server instance, I would assign a greater cost/pain to the variable behavior.

bobjacobsen, 2019-04-04:
> the situation we have now is untenable.

No situation that has existed for years and many, many uses is made untenable by the introduction of a new, uncoordinated actor.  Your rhetoric should be better than that.

> The current situation is that JMRI implements the reference implementation of the Withrottle Server and that reference implementation produces different results based on what environment it is running in.

It's true that JMRI produces different results.  And the LnWi does something specific.  And other implementations probably do other things.

There isn't a single reference implementation of a server for this protocol any more, and we certainly don't get to _change_ it here.  We can only change what JMRI does.

> Any way we look at this, there are going to be future compatibility issues with line endings. 

Overstated. People need to implement to a spec.  (I agree that needs to be updated)  There are lots of variations in implementations of that spec, not just line endings.  

Since new implementations will have to work with existing ones, some of which are hardware that's not going to change, the cat is already out of the bag.

> If we make a choice of line endings, those compatibility issues eventually go away. 

Maybe.  I'd be interested in your thoughts of how long "eventually" is, and how many things will break _NOW_.  

You're thinking along the "just update it" software lines.  Existing model railroad PCs are not like that.  When they're running, they often stay running. And there's hardware....

> Not making a choice allows new client software to be introduced that is incompatible with one operating system or another.

That's a non-sequitur. Since not every implementation is the same _already_, making a choice also allows that, and might make it worse.

> From a long term support view, making a choice of line ending is preferable, even if there is a some short term pain involved.

Perhaps. But you haven't made that case.  What choice do you make?  The one that makes this new client work?  The one compatible with the installed base of LnWi devices?  The one compatible with servers we don't even know about?

bobjacobsen, 2019-04-04:
> For either a commercial or hobbyist developer, if moving forward there is a need to regression against multiple platforms instead of just a single, standard server instance, I would assign a greater cost/pain to the variable behavior.

Why would you do that? Seriously, why do you feel the need to check a `c == '\n' || c == '\r'` against actual hardware?

We're well into hypothetical spin-up mode on this one.  This stuff works, it works well, and we just need to 
1. update the docs and 
2. figure out who the Train Throttle developer is and let him know to check for another character.

Not hard. Nothing else breaks.

mesheets, 2019-04-04:
> Why would you do that? Seriously, why do you feel the need to check a c == '\n' || c == '\r' against actual hardware?

The platform difference has been missed already, hence this thread.  With the line termination sequence coupled to the platform, it a matter of test coverage best practices and not assuming anything.

bobjacobsen, 2019-04-04:
> The platform difference has been missed already, hence this thread. With the line termination sequence coupled to the platform, it a matter of test coverage best practices and not assuming anything.

Perhaps I misunderstood the point about "_regression_ testing".  

You certainly don't need hardware to regression test the input to a parsing routine. 

Do you know how to contact the Train Throttle author?

pabender, 2019-04-04:
> It's true that JMRI produces different results. And the LnWi does something specific. And other implementations probably do other things.
>
> There isn't a single reference implementation of a server [...]

And my real argument is that JMRI should make a choice, and it doesn’t matter to me which choice we make in JMRI.

> Overstated. People need to implement to a spec. (I agree that needs to be updated) There are lots of variations in implementations of that spec, not just line endings.
>
> Since new implementations [...]

Agreed, and the protocol can continue to specify different choices for line endings, which maintains compatibility with other server implementations.

> > If we make a choice of line endings, those compatibility issues eventually go away.
> 
> Maybe. I'd be interested in your thoughts of how long "eventually" is, and how many things will break _NOW_.

Likely not many because we are only talking about making JMRI consistent accross operating systems.  The largest client base is WiThrottle and EngineDriver and we know both of those work with JMRI regardless of operating system.

That said...

I don’t have a vested interest in which choice is made, only that not making a choice goes against the conservative send/liberal receive philosophical view of networking.  That philosophy is in place because protocol documentation is ambiguous and different implementations result due to those ambiguities.  We are not currently being conservative in our outgoing data.

Clients still need to test themselves against different possible implementations of the protocol, regardless of what we decide to do, or not do, as a result of this discussion.

I’ll yield the floor on this discussion.

mesheets, 2019-04-04:
> Do you know how to contact the Train Throttle author?
Yes, there is contact information posted in the app's page in the Microsoft Store.  I am not the author, but I can certainly handle reaching out to the author.

KenC57, 2019-04-04:
When anyone mentions "those compatibility issues eventually go away", you
have to remember our users. Model railroaders are some of the worst at
insisting on using something until the day it dies, or longer. So this makes
the idea of "eventually" be a very, very, long time in much likelihood. Tie
in the other implementations out there, in hardware, that would support the
idea of "eventually" being a very long time.

-ken c

mesheets, 2019-04-04:
Given the understandable sensitivity to maintaining compatibility with existing/legacy apps and devices, might it be worthwhile to consider the following:

**Synopsis**: Create a WiThrottle server "advanced" option that would allow the user to specify the line ending character used by the system.

**Modification Description**: Since DeviceServer.java (as linked in the initial post) already performs a lookup to determine what line separator character sequence to use, this would simply modify that existing lookup to determine whether the user has specified a particular line separator character sequence; if the user has not specified such, this lookup would default to its current behavior, i.e. System.getProperty("line.separator").

**Problem Scenario**: Suppose a user does have one of these potentially unknown legacy devices (c.f. the post of @KenC57 above) that is looking for a singular, specific line separator character sequence.  Who knows, perhaps it could even be someone on an old Mac that uses '\r'.  (Just a "for instance," not sure whether that configuration is actually possible.)  Regardless of what platform they were running, though, they update their hardware and JMRI install to a new platform (say, they decide to go for a smaller, dedicated device like the Raspberry Pi), only to discover that their beloved device no longer works with JMRI.

At the present, such a user wishing to continue use of their device would be stuck and potentially forced into a situation of not being able to change the host platform (assuming they would be able to know and/or find out enough to realize that is the problem).  With an advanced WiThrottle setting, not only would the user be able to continue using this special throttle device/app, it might also prompt a better awareness of, "Hey, if your existing device doesn't work on your new JMRI setup, try this and see if it works for you" (assuming that not all users are going to have the ability dive into the code and/or pull out Wireshark to ascertain the issue).

**Risk**: The potential for device manufacturers, app builders, etc. to abuse this setting and—instead of fixing their product—require end users to switch this setting from the default to one of the platform-specific values.  That's not what we want—we want manufacturers, app builders, etc. to handle any potential line separator character sequence, in this case to avoid a potential "mutually exclusive option" scenario, where one device/app only works with one setting, and another device/app only works with a different setting.

**Summary thoughts**: Given the current scenario, I would not expect this option to make matters worse for anyone (as it will still default to the current behavior), but it would grant users potentially greater flexibility in switching host platforms without fear of losing compatibility with their existing device/app throttles.

bobjacobsen, 2019-04-04:
Unnecessary configuration items have a cost. Each time somebody has a connection problem on the JMRI users list, we’ll have to go through whether this option is set right. That’s a poor user experience and a waste of time. And it doesn’t fix the earlier issues raised: it still requires design work because vendors will want their products to work robustly despite the setting; whatever the test concern was, it’s still there, etc. 

If the documentation had been correct, this entire issue would have been resolved in one post here: Train Throttle is broken and needed to be fixed.  

We can debate what the protocol should be, but that doesn’t change what it is. Implementations can change, including what subset of the protocol they use (including line ends they send), but that doesn’t change the protocol nor what’s required to implement it properly.  Changing protocols that are in wide use is a lot like moving a hole: you can’t do that, you can only make a new one. 

The documentation is now fixed (#6766) and the problem is now clear: Train Throttle is broken and needed to be fixed to properly implement the protocol.

Pugwash1, 2019-04-04:
I am very confused. 
The Train Throttle app v1.11 beta running on wjndows 10 works fine against a pi. Linux, 4.14.98,  Jmri server 4.15.5ish 2019-04-02 AM.
 It starts, auto configures, downloads the roster, select the loco. Drive the train. Backwards, forwards, functions etc. 
No problemo.

mesheets, 2019-04-04:
@bobjacobsen Understand completely regarding the configuration pit, which is why I didn't mention it earlier; however, based on the discussion yesterday, and as there seemed to be _much_ concern about potentially breaking known existing devices, it was merely a suggestion if there was perchance a similar level of concern about platform lock-in when using these unknown legacy devices that might only function properly with one specific platform's line separator character.

@Pugwash1 Based on the JMRI version, that would appear to be a rolling dev build from earlier this week?  The test setup from which the logs and traces were captured is as follows:

> From the "[Installing JMRI on Raspbian](http://jmri.org/install/Raspbian.shtml)" install doc page, I am working with the [M Steve Todd image](https://mstevetodd.com/rpi) and using that as the JMRI+WiThrottle server.

Per M. Steve Todd's website, that image is based on the "[June 2018 Raspbian Stretch Image](https://www.raspberrypi.org/blog/raspbian-stretch/), which includes Java 8."

In turn, Raspbian Stretch refers to [this link](https://www.debian.org/releases/stable/amd64/release-notes/ch-whats-new.html) for the technical details of the Debian release notes.

In those Debian release notes, the Linux kernel image for version 9 (Stretch) is listed as the 4.9 series, but I have run some updates.

From one of the log files attached above:
> PanelPro version 4.14+Rd060e0b starts under Java 1.8.0_65 on Linux arm v4.14.98-v7+

One of the key source files in question, DeviceServer.java, is showing as last modified 14 days ago, with commit comments for that and the preceding commit referring to threading-related changes.

Curious as to what could potentially have mitigated all the readLine IOExceptions thrown and logged when connecting cross-platform.  The JMRI version running on Windows (and with which Train Throttle worked without issue) is the same 4.14 "Production" release as that which is running on the Raspbian Linux noted here.

Pugwash1, 2019-04-04:
@mesheets 
You do have a roster with at least one valid entry, cause that app don't like empty rosters, it spits out warnings about readline from device.

Pugwash1, 2019-04-04:
@mesheets  In 4.14 it works fine, providing you have at least one entry in the roster.

mesheets, 2019-04-04:
@Pugwash1 Thank you much for that lead.  On Windows, though, I neither had nor have any roster entries, either.

On Windows, the roster-config section in roster.xml for the active profile is the following:
`<roster-config...>
  <roster />
</roster-config>`.  If I open DecoderPro: All Entries, or if I navigate to File > Print Preview Roster > List, there are no roster entries listed in either place.

On the Raspbian image, I neither had or have a roster, either; however, under the profile subfolders within `/home/pi/.jmri`, there is not a "roster.xml" file in _any_ of those profile folders (including the folder for my active profile).  Within the root of the individual profile folders are subfolders such as networkServices, profile, programmers, signal, and throttle.

In the Raspbian image, it does appears that a default roster.xml file (matching the roster.xml file in the active Windows profile, i.e. `<roster-config...>
  <roster />
</roster-config>`) under /home/pi/JMRI_UserFiles.  I copied that roster.xml to the active profile folder, and then cross-platform connectivity succeeded (!).

In short, from a roster perspective, the Windows install included a default roster.xml file with no roster entries, while the Raspbian image lacked a roster.xml file.

Does it make a difference to JMRI if within the active profile folder there is an entryless roster.xml file versus having no roster.xml file at all?

Pugwash1, 2019-04-05:
> Does it make a difference to JMRI if within the active profile folder there is an entryless roster.xml file versus having no roster.xml file at all?

I am not sure., but an empty roster, created by adding and deleting the entry, gave an error.

mesheets, 2019-04-06:
In taking a deeper look at the issue, it appears the difference is whether or not WiThrottle sends the "Initial Connection" information documented on the [WiThrottle protocol page](http://www.jmri.org/help/en/package/jmri/jmrit/withrottle/Protocol.shtml).

For the sake of comparison, all tests noted here were with JMRI+WiThrottle server running on Raspbian.

Off the top, I will grant that the documentation say that initial connection information "may" (implying to me that there is a possibility that it might not be sent).

Without a roster.xml file present in the active profile folder, the initial connection information always seems to be sent; however, _if_ a roster.xml file is present, the initial connection information usually does not seem to be sent (this was observed on both Linux and Windows).

(I say _usually_, because if I would click "Quit" on PanelPro, rename the empty roster file from roster.xml.bak back to roster.xml, and then restart PanelPro, sometimes it would still keep sending the initial connection information anyway.  However, perhaps some other things were going on in the background related to the rename?  In normal situations [i.e. when I don't keep stopping/restarting JMRI and changing whether or not the roster.xml file exists], though, it seems quite consistent that the initial connection information is _sent_ when roster.xml does _not_ exist and is _not sent_ when roster.xml _does_ exist.)

Again, I understand that "may" implies not to absolutely expect that information.

The "VN" (version) and other initial connection information appears to be sent in the constructor of DeviceServer (c.f. [here](https://github.com/JMRI/JMRI/blob/90c45b0c1696232a0cc63f61545c1ed5f2afd5b2/java/src/jmri/jmrit/withrottle/DeviceServer.java#L158)), so I haven't quite yet figured out why or how there is a difference in what is sent.

Train Throttle seems to respond best when the initial connection information isn't sent and fail when initial connection information is sent.  I'm not here to suggest that behavior is correct; just trying to understand the difference in how JMRI+WiThrottle decides what to send in what (at least to me) would seem to be the same situation: no roster has been defined.

[Wireshark files covering three different scenarios discussed here.](https://github.com/JMRI/JMRI/files/3050689/jmriOnLinux-variations-on-rosterXML-Wireshark.zip)

Pugwash1, 2019-04-06:
@mesheets 
Have you tried adding an entry to the roster. I know it doesnt work without at least one valid entry.

mesheets, 2019-04-07:
It seems to appear at least for the moment, that in the situation I was describing, the presence or absence of a "entryless" roster.xml file might have been a red herring.

Every once in a while, with the JMRI Raspbian image running on a Raspberry Pi 3B+, WiThrottle doesn't always detect the ethernet interface when PanelPro is automatically launched on upon bootup.  In those situation, I simply do WiThrottle > Stop Server, followed by WiThrottle > Start Server.

Apparently, when I first tried copying in a roster.xml file to the active profile folder under Raspian, that also happened to be a time that I had restarted WiThrottle.

After further testing, it now appears that JMRI+WiThrottle initializes communication much differently before being restarted versus after being restarted, and Train Throttle connects after restarting WiThrottle (but not before restarting).  Additionally, I have been able to reboot Raspbian and then follow these same steps repeat this behavior, so this seems to be more reproducible.

.
**Communication initialization flow _before_ restart of WiThrottle server**:
4. Client:  "NMyDeviceName"  (sets the device name)
5. Server: ACK
6. Server: "VN2.0"  (WiThrottle protocol version)
7. Client: ACK
8. Server:  "RL0"  (There are zero entries in the roster list)
9. Client: ACK
10. Server:  "PPA2 \n \n PTT]\[Turnouts}|{Turnout]\[Closed}|{2]\[Thrown}|{4"  (**Line 1**: Track power state unknown;  **Line 2**: "Closed" turnout label value = 2; "Thrown" turnout label value = 4)
11. Client: ACK
12. Server: "PRT]\[Routes}|{Route]\[Active}|{2]\[Inactive}|{4"  ("Active" route label value = 2; "Inactive" route label value = 4)
13. Client: ACK
14. Server: "RCC0 \n \n PFT1554551202<;>1.0 \n \n PW12080"  (**Line 1**: Zero consist list entries;  **Line 2**: Fast clock time synchronization, with a ratio of 1.0;  **Line 3**: Port of webserver is 12080)
15. Client:  "\*-"  (turns heartbeat monitoring off)
16. Server:  "\*10"  (server sends a heartbeat interval of 10 seconds, _after the client requests to turn off heartbeat monitoring_?)
17. Client:  "MTA*<;>qR"  (query all throttles in the "T" MultiThrottle for the direction)
18. Server:  ACK
19. Client:  RST, ACK  (Communication reset)

.
**Communication initialization flow _after_ restart of WiThrottle server**:
4. Client:  "NMyDeviceName"  (sets the device name)
5. Server: ACK  (From the documentation, it appears the server is supposed to send a heartbeat interval (in seconds) in response to an "N" command; however, Wireshark doesn't seem to show that.)
6. Client:  "\*-"  (turns heartbeat monitoring off)
7. Client:  "MTA*<;>qR"  (query all throttles in the "T" MultiThrottle for direction)
8. Client: "MTA*<;>qR"  (repeated)
9. Client: "MTA*<;>qV"  (query all throttles in the "T" MultiThrottle for velocity)
10. Server:  ACK
11. Server:  ACK
12. Server:  ACK
13. Server:  ACK
14. Client:  "MTA*<;>r"  (release [dispatch] all locomotives under the control of the current MultiThrottle)
15. Server:  ACK
16. Client:  PSH, ACK  (Push data through to the receiver for processing)
17. Server:  ACK
18. Client:  FIN, ACK  (Last packet from sender)
19. Server:  ACK

From a protocol perspective, might there be a particular reason for this change in behavior after WiThrottle server is restarted?

BHoffman351, 2019-04-07:
Perhaps it is time to close this issue.  We are 47 comments into chasing the wrong problem.  The real cause of the issue would be more appropriately discussed with the developer of the Train Throttle app.

It is correct that information may not be sent.  As an example, the earliest versions of the WiThrottle Server did not access the roster at all and therefore nothing was sent.  An app also needs to allow for a missed packet, or a feature that is not available on the server.  There is no situation where an app should be waiting for any response - even to a request that the app makes.

mesheets, 2019-04-07:
I just had another post 1 second before you....

mesheets, 2019-04-07:
Communication initialization comparisons when _not_ resetting the WiThrottle server

**Communication initialization flow with JMRI on Linux _without_ a roster.xml file** (with Wireshark frame number):
4. Client:  `NMyDeviceName`  (sets the device name)
5. Server:  ACK
6. Server: `VN2.0`  (WiThrottle protocol version)
7. Client:  ACK
8. Server:  `RL0`  (There are zero entries in the roster list)
9. Client:  ACK
10. Server:  `PPA2 \n \n PTT]\[Turnouts}|{Turnout]\[Closed}|{2]\[Thrown}|{4`  (**Line 1**: Track power state unknown;  **Line 2**: "Closed" turnout label value = 2; "Thrown" turnout label value = 4)
11. Client:  ACK
12. Server: `PRT]\[Routes}|{Route]\[Active}|{2]\[Inactive}|{4`  ("Active" route label value = 2; "Inactive" route label value = 4)
13. Client:  ACK
14. Server: `RCC0 \n \n PFT1554551202<;>1.0 \n \n PW12080`  (**Line 1**: Zero consist list entries;  **Line 2**: Fast clock time synchronization, with a ratio of 1.0;  **Line 3**: Port of webserver is 12080)
15. Client:  `*-`  (turns heartbeat monitoring off)
16. Server:  `*10`  (server sends a heartbeat interval of 10 seconds, _after the client requests to turn off heartbeat monitoring_?)
17. Client:  `MTA*<;>qR`  (query all throttles in the "T" MultiThrottle for the direction)
18. Server:  ACK
19. Client:  RST, ACK  (Communication reset; app exits)

.
**Communication initialization flow with JMRI on Linux with an "entryless" roster.xml file** (with Wireshark frame number):
4. Server: `VN2.0`  (WiThrottle protocol version)
5. Client:  `NMyDeviceName`  (sets the device name)
6. Server:  PSH, ACK
7. Client:  ACK
8. Server:  `RL0 \n \n PPA2 \n \n PTT]\[Turnouts}|{Turnout]\[Closed}|{2]\[Thrown}|{4 \n \n PRT]\[Routes}|{Route]\[Active}|{2]\[Inactive}|{4`  (**Line 1**: There are zero entries in the roster list;  **Line 2**: Track power state unknown;  **Line 3**: "Closed" turnout label value = 2; "Thrown" turnout label value = 4;  **Line 4**: "Active" route label value = 2; "Inactive" route label value = 4)
9. Client:  Client:  `*-`  (turns heartbeat monitoring off)
10. Server:  `RCC0 \n \n PFT1554554816<;>1.0 \n \n PW12080`  (**Line 1**: Zero consist list entries;  **Line 2**: Fast clock time synchronization, with a ratio of 1.0;  **Line 3**: Port of webserver is 12080)
11. Client:  ACK
12. Server:  `*10`
13. Client: ACK
14. Client:  `MTA*<;>qR`  (query all throttles in the "T" MultiThrottle for the direction)
15. Server:  ACK
16. Client:  RST, ACK  (Communication reset; app exits)

.
**Communication initialization flow with JMRI on Windows with an "entryless" roster.xml file** (with Wireshark frame number):
4. Client:  `NMyDeviceName`  (sets the device name)
5. Server: `VN2.0`  (WiThrottle protocol version)
6. Server:  `RL0`   (There are zero entries in the roster list)
7. Client:  ACK
8. Server:  `PTT]\[Turnouts}|{Turnout]\[Closed}|{2]\[Thrown}|{4`  ("Closed" turnout label value = 2; "Thrown" turnout label value = 4)
9. Client:  ACK
10. Server: `PRT]\[Routes}|{Route]\[Active}|{2]\[Inactive}|{4`  ("Active" route label value = 2; "Inactive" route label value = 4)
11. Client:  ACK
12. Server:  `PFT1554115511<;>1.0 \r\n \r\n PW12080`  (**Line 1**: Fast clock time synchronization, with a ratio of 1.0;  **Line 2**: Port of webserver is 12080)
13. Client:  ACK
14. Server:  `*10`  (server specifies a heartbeat interval of 10 seconds)
15. Client:  ACK
16. Client:  `*-`  (client turns heartbeat monitoring off)
17. Client:  `MTA*<;>qR`  (query all throttles in the "T" MultiThrottle for the direction)
18. Server:  ACK

**High-level differences**
On Windows
- `PPA2` message is not sent (when on Linux, this is sent between the `RL0` and `PPT` messages)
- `RCC0` message is not sent (when on Linux, this is sent between the `PRT` and `PFT` messages)

On Linux
- Server sends a heartbeat interval of 10 seconds following a client request to turn it off; however, given the way the code appears to flow, it doesn't seem like this should matter, as new messages won't be read until the `N` command sent at connection initiation has fully processed and sent its messages.
- Connection is reset after client sends `MTA*<;>qR` message, whereas on Windows the connection is maintained.

On Both
- RL0 is sent  (There are zero entries in the roster list)
- Overall, apart from these and the line separator character differences, these communication initialization flows seem very similar (though not identical) across the platforms.

.
**Communication initialization flow _after_ restart of WiThrottle server**:
4. Client:  `NMyDeviceName`  (sets the device name)
5. Server: ACK  (From the documentation, it appears the server is supposed to send a heartbeat interval (in seconds) in response to an `N` command; however, Wireshark doesn't seem to show that.)
6. Client:  `*-`  (turns heartbeat monitoring off)
7. Client:  `MTA*<;>qR`  (query all throttles in the "T" MultiThrottle for direction)
8. Client:  `MTA*<;>qR`  (repeated)
9. Client:  `MTA*<;>qV`  (query all throttles in the "T" MultiThrottle for velocity)
10. Server:  ACK
11. Server:  ACK
12. Server:  ACK
13. Server:  ACK
14. Client:  `MTA*<;>r`  (release [dispatch] all locomotives under the control of the current MultiThrottle)
15. Server:  ACK
16. Client:  PSH, ACK  (Push data through to the receiver for processing)
17. Server:  ACK
18. Client:  FIN, ACK  (Last packet from sender)
19. Server:  ACK

In this case, it appears that the server never sends the `*10` heartbeat interval message in response to receipt of the `N` message (not to mention the absence of the informational messages that "may" be sent). Thus, if the client were waiting for the `*10` heartbeat interval message response to the `N` message, it would end up hanging.  Among other questions, is that heartbeat interval message response optional (in which case the documentation needs updated), or should clients expect that response (in which case WiThrottle might not be responding properly after being stopped and restarted).

greenwizard88, 2019-04-17:
I'm the developer of Train Throttle.

I'm sending data via 
`streamWriter.WriteLineAsync("STRING_TO_SEND");`
`streamWriter.FlushAsync();`

And listening via
`string response = await streamReader.ReadLineAsync();`
`if (String.IsNullOrWhiteSpace(response) == false) {`
`    parse(response);`
`}`

The `ReadLineAsync()` documentation doesn't say much, but the documentation for `ReadLine()` says that `\r`, `\n`, and `\r\n` are all valid newline characters. The application should be quite hardy in that sense.

Train Throttle uses WriteLineAsync() exclusively, although it's possible that a roster with an engine name that has a newline in it might present a problem... I'll look into that tonight.

I develop Train Throttle against JMRI running on a Raspberry Pi I set up some time ago.

I suspect that the difference between `\r`, `\n`, and `\r\n` is also a red herring, although I have no idea as what the root cause would be.

The error in the log
`2019-03-26 12:04:18,964 withrottle.DeviceServer               WARN  - readLine from device 'Windows 10 Professional' failed, consecutive error # 25 [DeviceServer-2]` Is on the JMRI side. So what's listed as a readLine warning in the log is actually something that my app is sending via writeLine.

The only place my code uses qR and qV are in the same function, right next to each other
`            await sendCommand(THROTTLE + "A*<;>" + "qR");`
`            await sendCommand(THROTTLE + "A*<;>" + "qV");`

If there's an issue after the qR is sent, it's in that code block. sendCommand is a wrapper for the streamWriter.writeLineAsync() that every command uses.

I'm pretty fresh out of ideas. @mesheets would you be able to send me your config/roster files? I'd like to try and reproduce the issue at home.

https://docs.microsoft.com/en-us/dotnet/api/system.io.streamwriter.writelineasync?view=netframework-4.7.2#System_IO_StreamWriter_WriteLineAsync

https://docs.microsoft.com/en-us/dotnet/api/system.io.streamreader.readlineasync?view=netframework-4.7.2

mesheets, 2019-04-23:
I have used a rather unmodified JMRI configuration.  Part of what has been puzzling is that an "out of the box" setup works on Windows but does not on Linux/Raspbian.

Based on the suggestion of @Pugwash1 , I did then check into potential differences related to roster.xml and did find a difference there between the platform distributions.  (As described earlier, the Windows install included an "entryless" roster.xml file in the active profile folder, while Linux/Raspbian image did _not_ have a roster.xml file in the active profile folder.)

On Linux/Raspbian, I have attempted two different configurations.
1. Without a roster.xml file in the named profile folder under /home/pi/.jmri:  This is the default startup configuration for the JMRI Linux/Raspbian image.
2. With a roster.xml file in the profile folder that has been copied from /home/pi/JMRI_UserFiles/roster.xml to the root of the named profile folder under /home/pi/.jmri.  The contents of this roster.xml file are as follows:

```xml
<?xml version="1.0" encoding="UTF-8"?>
<?xml-stylesheet href="/xml/XSLT/roster2array.xsl" type="text/xsl"?>
<roster-config xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:noNamespaceSchemaLocation="http://jmri.org/xml/schema/roster.xsd">
  <!--Written by JMRI version 4.12+Rb6a9bb1 on Fri Jul 06 06:08:01 EDT 2018-->
  <roster />
</roster-config>
```
NOTE: I'm not sure why the roster.xml file reports JMRI version 4.12; in WiThrottle server GUI, the version is reported as 4.14_Rd060e0b under Help > About.

In both of the above cases, the only uncommented line in /home/pi/.jmri/jmri.conf is the following:
```conf
default_options="-Dpi4j.linking=dynamic"
```

.
On Windows, the roster.xml file has been as installed by default, as follows:
```xml
<?xml version="1.0" encoding="UTF-8"?>
<?xml-stylesheet href="/xml/XSLT/roster2array.xsl" type="text/xsl"?>
<roster-config xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:noNamespaceSchemaLocation="http://jmri.org/xml/schema/roster.xsd">
  <!--Written by JMRI version 4.14+Rd060e0b on Thu Apr 04 20:26:19 EDT 2019-->
  <roster />
</roster-config>
```

Perhaps I'm not looking in the correct places, but I'm not finding a jmri.conf file under Windows.

.
In addition to the roster.xml installation/image differences noted above between platforms, are there any other potential _configuration_ differences between Windows and Linux/Raspbian that might potentially contribute to the noted behavioral differences between Windows and Linux/Raspbian?

Furthermore, there is the difference in behavior that has been discovered when the JRMI WiThrottle server first starts up versus when the JMRI WiThrottle server has been restarted (as noted [here](https://github.com/JMRI/JMRI/issues/6744#issuecomment-480562592)), so it currently does not seem possible to test different roster.xml scenarios by simply stopping and restarting the JMRI WiThrottle server via WiThrottle server's menu options.

bobjacobsen, 2019-04-23:
The “written by JMRI” lines are what they say: the time the was written by a program running that particular JMRI version. 

Note that the Windows date is well after the release creation date. Maybe it was created during the installation, or by something you did after the installation.

mstevetodd, 2019-04-23:
@greenwizard88 
I installed Train Throttle and did a bit of testing. I found that:
1) Train Throttle always crashes if I have no roster entries, indicated by WiThrottle Server returning "RL0".
2) Train Throttle does not crash if I have at least one roster entry, indicated by WiThrottle Server returning "RL1]\[VRE53}|{53}|{S" or similar.
I verified this on Windows 10, Raspbian and Ubuntu, by removing all roster entries, attempting to connect (crash), then adding a roster entry and connecting successfully.
I found no differences in behavior on the 3 platforms, nor with empty roster.xml vs. no roster.xml.

My _guess_ is that your code isn't handling the absence of roster entries properly.
Please remove all roster entries and test your application.

--SteveT

mesheets, 2019-04-24:
> Note that the Windows date is well after the release creation date. Maybe it was created during the installation, or by something you did after the installation.

Yes, perhaps during install?  I performed a clean install of JMRI for testing purposes, which might explain the more recent timestamp in the version of the roster.xml file on Windows.

.
> Please remove all roster entries and test your application.

Testing recommendation: I would advise **_against_** using the following sequence when testing, as it will yield different results that will appear to work:
1. Switch to the WiThrottle window
2. In the WiThrottle window, select WiThrottle menu > Stop Server
3. Add/remove/edit the roster.xml file
4. In the WiThrottle window, select WiThrottle menu > Start Server

If stopping and restarting the JMRI WiThrottle server as listed like that, the connection initialization sequence from the JMRI WiThrottle server will be different (c.f. the Wireshark traces posted earlier).  I'm not saying whether that difference in behavior is expected (I can't speak to that, myself); just that a connection between JMRI WiThrottle and Train Throttle will then establish successfully (_even without a roster.xml file present_) and one might in turn be left wondering what all the discussion is about.

I had sometimes been following the above sequence, and it was really starting to confuse me when testing, until I realized that manually stopping and starting the JMRI WiThrottle server seems to induce a different connection initialization sequence.

rhwood, 2019-04-24:
The roster.xml file is created when a JMRI application runs, not at installation time.

Once the roster.xml file read, it is retained in memory, such that manual changes while a JMRI application is running will have any effect; constantly rereading that from disk would have significant negative performance impacts otherwise.

mesheets, 2019-04-24:
@rhwood 
> manual changes while a JMRI application is running will have any effect

For my clarification, and based on the context of the negative performance impact of constant disk rereads, should that perhaps read, "manual changes while a JMRI application is running will _not_ have any effect"?

That was somewhat my expectation of what the behavior might be, which is why I would stop and restart the WiThrottle server service, until I discovered that WiThrottle behaves differently once it has been stopped and then restarted.  Now, I close JMRI by clicking the "Quit" button on PanelPro and then relaunch PanelPro, and sometimes for good measure it seems I need to reboot.

greenwizard88, 2019-04-28:
There's an update for Train Throttle out, v1.13beta, which should correct the issue. Can anyone who was able to reproduce the issue, confirm for me that it's fixed?

mstevetodd, 2019-04-28:
@greenwizard88 I tested it on the same configuration which crashed before, and it no longer crashes. Thanks!

[1 more reply]