JMRI/JMRI #6750: GitHub reporting a security issue with

bobjacobsen, 2019-03-30:
See Issue #6613 and PR #6614 for background. 

Looks like https://github.com/patternfly/angular-patternfly/issues/769 has been resolved.  Could somebody who a nom-compatible Linux please try the steps recommended by @rhwood in #6614? Thanks in advance.

The [current security warning](https://github.com/JMRI/JMRI/network/alert/web/app/package-lock.json/bootstrap/open) (link might not work for some people?) is about `bootstrap` and recommends updating to 3.4.1 or later.

[image]

rhwood, 2019-03-30:
Only owners and admins see the alerts by default. See https://help.github.com/en/articles/managing-alerts-for-vulnerable-dependencies-in-your-organizations-repositories

rhwood, 2019-04-12:
Did #6774 resolve this?

bobjacobsen, 2020-01-08:
Fixed by @rhwood's #6774

closed 2020-01-08 by bobjacobsen