JMRI/JMRI #6775: How should DCC Signal Heads be initialized?

bobjacobsen, 2019-04-05:
DccSignalHead (hence DSH) objects hold a mapping of an integer value, used as the number to send over DCC to command changes, and an appearance name:  Dark, Red, etc.  By default, the zero value is associated with Dark.

Right now, DSH objects start up with appearance value at zero. i.e. usually dark.

They don't send anything to the layout  at start up. This is intentional:  There was concern about how much DCC traffic would be sent as a panel file initializes a bunch of signal heads all at once.

But a jmriusers thread has pointed out a problem:  Not all hardware starts up dark.  In particular, NCE Light It devices remember what they were.  And setting it to Dark again doesn't send anything, because the SignalHead doesn't send duplicate state settings. This is a problem with a signal system that sets the head to Dark at startup (instead of something else), but the Light It stays with its previously remembered value.

There are a couple solutions, and maybe others exist:

1. Start at 0 (usually Dark) and above, and send that as a DCC command at startup.

2. Add an Unknown state with value, and start up at that.  This has to be visible in the Signal Head Table, but otherwise can't be set, commanded to, won't be set to the layout, etc.  The first time an appearance is set, that new value will go out because it won't be the same as Unknown.

3.  Always send the command, even if it's duplicative.  (Our signal logic tends to recompute a _lot_ of aspects and appearances, so I don't favor this)

4. Have a special "first appearance setting _always_ goes" special case, so the first setting bypasses the duplicate check.

I tend to prefer (4), as I think it has the least chance of unexpected consequences, and will still solve the problem.

rhwood, 2019-04-06:
Another solution is to store state outside the panel file itself (possibly in a cache file (we already have a class for caches, but I can't look it up right now)) and set the DSH to the state in the cache file. The cache file could be written to nearly constantly without affecting panel file contents.

pabender, 2019-04-08:
> I tend to prefer (4), as I think it has the least chance of unexpected consequences, and will still solve the problem.

I would lean towards either 4 or 2.

There are three reasons I lean towards 2.
1) 2 really includes 4, it just makes the initial state visible to users.
2) Unknown is a common named bean state, so users are familiar with what it means already
3) Unless there is feedback telling us what the current state is, we really don't know what the state is on the layout, so showing something other than unknown doesn't reflect our knowledge of the layout state.

dheap, 2019-04-08:
I'd lean towards 2.

closed 2021-04-14 by stale[bot]