JMRI/JMRI #6794: Accessory from 1 to 2044 with Expressnet

ohojvat, 2019-04-14:
As stated in V3.0 and 3.6 the range for accssory address is from 1 to 1023

As accesory range in NMRA is 1 to 2044 , and we (LDHtrenes) make the command station and the hand-held unit, I add a bit in the message to the command station for the full 9 bits addressing.
from Expressnet 3.0 in Engllsh
2.2.18 Accessory Decoder operation request 
Header byte | Data Byte 1 |  Data Byte 2
          0x52  |   Address     |  0x80 + DBBD        (hex)
I have used bit6 of Data Byte2 for sending the bit 9 of accessory address
For address 1024 to 2044    it becomes  0xC0 + DBBD
If JMRI people can use this idea , the full NMRI address range can be used 
As its was recently done I can change my code if other solution arises.

Waiting some replay, Regards,Osvaldo Hojvat

pabender, 2019-04-14:
Is there a way we can differentiate your command station from a Lenz command station?

Does your command station have a built in computer interface, or are you using one of the existing XpressNet interfaces?

The problem is not really can JMRI do this, but what does a Lenz command station do with it.  Since we automatically identify connected XpressNet command stations at start up, I really need your command station to report something other than the existing combinations of software and hardware reported by the existing command stations.

We can also force a configuration for your specific system.  This is easiest when your system has its own interface, but it isn’t absolutely required.

At least some of the Lenz computer interfaces actively filter commands that are not explicitly allowed.  If your command station has its own interface, we don’t need to worry about that.

ohojvat, 2019-04-15:
Thank you for your reply.

We are making a PC serial interfase similar to LI100 and can do some modifications.

I will check the information at the serial (rs232) PC port at starting 
Also when accesory order was sent. with address  below and over 1034.

I will inform you soon

Best regards, Osvaldo Hojvat

-------------------------------------

We had some email contact at august 2018 about Accessory  PoM

ohojvat, 2019-04-16:
Hello Paul
 
Our system was adjusted to reply messages similar to LENZ for JMRI compatibility.

We can modify messages as needed.
Request Command station version  ->   our reply  63 21 38 00 7A
we change recently from   3.6 to 3.8 software version
00 for command station model can be modifyed to a number that Lenz may never use

-----------------------------------------------------------
 
For PC interfase we set in edit/preferences LI100 and in a few cases LI100F for 19200 baud rate

It is seted by a jumper

 The number for ExpressNet is 30   

The reply from PC interfase hardware and soft version is 02 30 01 33
I  can change 30  and  01 to the required value

--------------------------

Our  name is LDH and our decoders are listed in JMRI definitions.  
We can use LDH-M28E as the command station name (or LDH alone)

We have not a special name for the PC interfase.     (may be LDH-PC)

We are a very small, only 2 people "factory".  

  our page:  http://www.LDHtrenes.com.ar

--------------------------------------------------------------------------------

I mostly use a PC with WindowsXP and JMRI   PanelPro3.10.1-r2837

but I also have a slow Windows10 notebook with a new JMRI version.

Best Regards, Osvaldo Hojvat

silverailscolo, 2019-05-28:
Excuse me for changing/reverting, now I guess this name is confusing for a purpose?

pabender, 2019-05-28:
I had forgotten about this issue.

I recently was sent a Lenz LZV200 which supports XpressNet version 3.8.  Lenz tells me it does support accessory decoders up to 2048.  Lenz has yet to release the updated a XpressNet documentation

pabender, 2019-05-28:
> Excuse me for changing/reverting, now I guess this name is confusing for a purpose?

It was correct as you changed it.  Unfortunately there are many people who add an E to the front, even though that is not proper.  I have learned to ignore it.

ohojvat, 2019-05-28:
I don't understand your comment.  I see the title remains the same. 

There are two purposes in this issue:
- Ask JMRI to allow the full accesory range 1 to 2044 ( original Epressnet only 1 to 1024 )
- If other manufacturer likes it , they can use this idea. 

I am waiting a new reply from Paul Bender.

ohojvat, 2019-05-28:
Sorry for the mistake - I will take care of it in the future. I know that there are not an E before the X

Please let me know the new XpresNet command, (perhaps it may be the same as I was stated ?) 

I can change my software is it may be easy for JMRI

pabender, 2019-05-28:
> I don't understand your comment. I see the title remains the same.

Your original title had ExpressNet instead of XpressNet.  @silverailscolo changed it to the proper XpressNet then changed it back.

> There are two purposes in this issue:
> 
> * Ask JMRI to allow the full accesory range 1 to 2044 ( original Epressnet only 1 to 1024 )
> * If other manufacturer likes it , they can use this idea.

Lenz’s new version of the protocol includes support for accessory decoders from 1 to 2048.  Unfortunately Lenz has not released the updated version ( 3.8 ) of the protocol.  Once that happens, we will add support for the official commands.

ohojvat, 2019-05-28:
Sorry again      (XpressNet - XpressNet - XpressNet)

ohojvat, 2019-05-28:
Please let me know if you get and use in JMRI the new protocol information , so I may adjust my code for better compatibility.
Best Regards

closed 2021-04-14 by stale[bot]