JMRI/JMRI #6795: Switchboard not using saved connection data

KenC57, 2019-04-15:
It seems the switchboard is not using the saved connection when you open the panel file again. So it flips back to Internal. Here is the users file. It shows "N" for the nce connection but builds the switchboard on opening as Internal. I was using 4.14.5 to validate what they saw.
-ken c
[nce turn outs.xml.txt](https://github.com/JMRI/JMRI/files/3078065/nce.turn.outs.xml.txt)

bobjacobsen, 2019-04-15:
Works for me. Are you testing with an “N” connection?

KenC57, 2019-04-15:
Using NCE simulator, set for N. When the panel is displayed, it all shows
the IT* values instead of the NT*.
Bob, were you on 4.14.4 or newer?
-ken c

silverailscolo, 2019-04-16:
Will check the code. The connection (prefix) is indeed read from the xml, seems too late in the process while switches are already being created.

pabender, 2019-04-16:
Sent from my iPad

> On Apr 16, 2019, at 1:43 AM, Egbert Broerse <notifications@github.com> wrote:
> Will check the code. The connection (prefix) is indeed read from the xml, seems too late in the process while switches are already being created.
> 

The panel file(s) should be loading after the connections are all established.  If they are not, then there is an ordering problem in the initial configuration.

KenC57, 2019-04-16:
I was manually load the panel file. I used NCE Simulator. Original user was
on a SB5.
-ken c

silverailscolo, 2019-05-17:
Fixed with PR #6802

closed 2019-05-17 by silverailscolo