JMRI/JMRI #6809: OpenJDK and JMRI

wimorrison, 2019-04-18:


wimorrison, 2019-04-18:
I want to run JMRI using OpenJDK but when I install it from https://adoptopenjdk.net/ and uninstall Java 8 JMRI will not sun saying that it cannot find Java, yet other Java programs run happily without Java 8. Event trying to run the jmri.jar file fails

Haw can I amke JMRI run under OpenJDK to allow me and others to move away from Oracle and their demands?

bobjacobsen, 2019-04-18:
We routinely test JMRI itself against OpenJDK, so this sounds like a launcher issue. 

What OS are you using? What does `java -version` on the command line do?

wimorrison, 2019-04-18:
I am using Windows 10 Build 1809 and the information is

openjdk 12 2019-03-19
OpenJDK Runtime Environment AdoptOpenJDK (build 12+33)
Eclipse OpenJ9 VM AdoptOpenJDK (build openj9-0.13.0, JRE 12 Windows 10 amd64-64-Bit Compressed References 20190320_32 (JIT enabled, AOT enabled)
OpenJ9   - caeb51f87
OMR      - 33a33ff2
JCL      - 419f571019 based on jdk-12+33)

does this help?

devel-bobm, 2019-04-18:
If you are running any JMRI version before JMRI 14.5.4, there may be issues with the JMRI launcher for Windows, depending on which OpenJDK version you are running.  

Pull request #6652 addressed the issues seen at that time, and those fixes are in the JMRI 14.5.4 test release.

devel-bobm, 2019-04-18:
Correction: 4.15.4 is the version which has the OpenJDK fixes.

bobjacobsen, 2019-04-18:
That’s showing OpenJDK 12, which we haven’t started testing against yet.

It’s probably not the cause of your problem, but you might want to try OpenJDK 11 or earlier.

wimorrison, 2019-04-19:
I have tried the 4.15.4 version however I cannot install it without installing Java 1.8 or later :(

is there a variable that I can defined which will make the installer routing use the Java version that is installed?

wimorrison, 2019-04-19:
I installed Java8 to let me install 4.15.4 then reinstalled OpenJDK but JMRI still wont run, unless it is using Java8. I then ran installtest.bat and the results can be seen here

https://abload.de/img/javaissue11k7r.png

It says that it has tested and found Java, but then falls over :(

ckier, 2020-03-19:
See related workaround in issue #8258

closed 2021-04-14 by stale[bot]

Deepfat, 2021-06-16:
I have a similar issue I am running our [openJDK](https://www.microsoft.com/openjdk) and java version returns 
openjdk version "11.0.11"

However I get the java not found when installing JMRI 4.22
[image]
Any help appreciated @deepfat  by night and andrew.fryer@microsoft.com by day