JMRI/JMRI #6811: Travis: No output has been received in the last 10m0s

danielb987, 2019-04-19:
Travis headless=false frequent gives the error:

```
No output has been received in the last 10m0s, this potentially indicates a stalled build or something wrong with the build itself.
Check the details on how to adjust your build configuration on: https://docs.travis-ci.com/user/common-build-problems/#Build-times-out-because-no-output-was-received

The build has been terminated
```

The last lines in the log are:
```
[INFO] Running jmri.jmrit.operations.rollingstock.cars.CarColorsTest
[INFO] Tests run: 2, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.036 s - in jmri.jmrit.operations.rollingstock.cars.CarColorsTest
[INFO] Running jmri.jmrit.operations.rollingstock.cars.CarManagerXmlTest
[INFO] Tests run: 1, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.017 s - in jmri.jmrit.operations.rollingstock.cars.CarManagerXmlTest
[INFO] Running jmri.jmrit.operations.rollingstock.cars.CarRoadsTest
[INFO] Tests run: 2, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.045 s - in jmri.jmrit.operations.rollingstock.cars.CarRoadsTest
[INFO] Running jmri.jmrit.operations.rollingstock.cars.CarEditFrameTest
```

I'm not sure, but I think I have seen CarEditFrameTest as the last test with this error several times before. May it be a problem with that test? This test do `new CarEditFrame()` 27 times.

pabender, 2019-04-19:
Sent from my iPad

> On Apr 18, 2019, at 8:12 PM, Daniel <notifications@github.com> wrote:
> 
> Travis headless=false frequent gives the error:
> 
> No output has been received in the last 10m0s, this potentially indicates a stalled build or something wrong with the build itself.
> Check the details on how to adjust your build configuration on: https://docs.travis-ci.com/user/common-build-problems/#Build-times-out-because-no-output-was-received
> 
> The build has been terminated
> The last lines in the log are:
> 
> [INFO] Running jmri.jmrit.operations.rollingstock.cars.CarColorsTest
> [INFO] Tests run: 2, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.036 s - in jmri.jmrit.operations.rollingstock.cars.CarColorsTest
> [INFO] Running jmri.jmrit.operations.rollingstock.cars.CarManagerXmlTest
> [INFO] Tests run: 1, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.017 s - in jmri.jmrit.operations.rollingstock.cars.CarManagerXmlTest
> [INFO] Running jmri.jmrit.operations.rollingstock.cars.CarRoadsTest
> [INFO] Tests run: 2, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.045 s - in jmri.jmrit.operations.rollingstock.cars.CarRoadsTest
> [INFO] Running jmri.jmrit.operations.rollingstock.cars.CarEditFrameTest
> I'm not sure, but I think I have seen CarEditFrameTest as the last test with this error several times before. May it be a problem with that test? This test do new CarEditFrame() 27 times.
> 

Basically, what this tells us is that the last test before the end of the output is hanging.

That test and a similar one for engines seem to be the biggest culprits right now.  

We need to fix that so we can turn the verbose output off.  It makes it very difficult to look at test reports on mobile devices.

danielb987, 2019-04-22:
Output from travis headless=false
```
[INFO] Running jmri.jmrit.roster.swing.RosterTableTest
[INFO] Tests run: 1, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.018 s - in jmri.jmrit.roster.swing.RosterTableTest
[INFO] Running jmri.jmrit.roster.swing.RosterEntrySelectorPanelTest
[INFO] Tests run: 1, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.009 s - in jmri.jmrit.roster.swing.RosterEntrySelectorPanelTest
[INFO] Running jmri.jmrit.roster.swing.RosterFrameTest

No output has been received in the last 10m0s, this potentially indicates a stalled build or something wrong with the build itself.
Check the details on how to adjust your build configuration on: https://docs.travis-ci.com/user/common-build-problems/#Build-times-out-because-no-output-was-received
```

danielb987, 2019-04-28:
Output from travis headless=false
```
[INFO] Running jmri.jmrix.nce.NceConnectionStatusTest
[INFO] Tests run: 1, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0 s - in jmri.jmrix.nce.NceConnectionStatusTest
[INFO] Running jmri.jmrix.nce.NceConsistTest
Exception in thread "Read Consist 3" Exception in thread "Read Consist 3" java.lang.NoClassDefFoundError: Could not initialize class jmri.jmrix.nce.NceBinaryCommand
	at jmri.jmrix.nce.NceConsist$NceReadConsist.readConsistMemory(NceConsist.java:368)
[INFO] Tests run: 12, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.049 s - in jmri.jmrix.nce.NceConsistTest
[INFO] Running jmri.jmrix.nce.swing.NceMenuTest
[INFO] Tests run: 1, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0 s - in jmri.jmrix.nce.swing.NceMenuTest
[INFO] Running jmri.jmrix.nce.swing.NceComponentFactoryTest
[INFO] Tests run: 1, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0 s - in jmri.jmrix.nce.swing.NceComponentFactoryTest
[INFO] Running jmri.jmrix.nce.swing.NceNamedPaneActionTest

No output has been received in the last 10m0s, this potentially indicates a stalled build or something wrong with the build itself.
Check the details on how to adjust your build configuration on: https://docs.travis-ci.com/user/common-build-problems/#Build-times-out-because-no-output-was-received
```

danielb987, 2019-04-30:
```
[INFO] Running jmri.jmrit.roster.swing.RosterEntrySelectorPanelTest
[INFO] Tests run: 1, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.012 s - in jmri.jmrit.roster.swing.RosterEntrySelectorPanelTest
[INFO] Running jmri.jmrit.roster.swing.RosterFrameTest
WARN  - testIdentify3ViaDecoderId(jmri.jmrit.roster.swing.RosterFrameTest) : run  1 failed, RetryRule repeats [main] util.junit.rules.RetryRule.evaluate()

No output has been received in the last 10m0s, this potentially indicates a stalled build or something wrong with the build itself.
Check the details on how to adjust your build configuration on: https://docs.travis-ci.com/user/common-build-problems/#Build-times-out-because-no-output-was-received
```

danielb987, 2019-05-07:
```
[INFO] Running jmri.jmrit.roster.swing.RosterEntrySelectorPanelTest
[INFO] Tests run: 1, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.017 s - in jmri.jmrit.roster.swing.RosterEntrySelectorPanelTest
[INFO] Running jmri.jmrit.roster.swing.RosterFrameTest

No output has been received in the last 10m0s, this potentially indicates a stalled build or something wrong with the build itself.
Check the details on how to adjust your build configuration on: https://docs.travis-ci.com/user/common-build-problems/#Build-times-out-because-no-output-was-received
```

danielb987, 2019-05-07:
@pabender 
I have three times there RosterFrameTest is timing out. See above. Maybe timeout and retry rules should be added to this test?

danielb987, 2019-06-23:
@pabender 
I frequent get this error from jmri.RunCucumberTest on Travis headless=false on my branch when I commit:

```
[INFO] Running jmri.script.ScriptFileChooserTest
[INFO] Tests run: 1, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.017 s - in jmri.script.ScriptFileChooserTest
[INFO] Running jmri.script.ScriptOutputTest
[INFO] Tests run: 1, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.004 s - in jmri.script.ScriptOutputTest
[INFO] Running jmri.TimebaseRateExceptionTest
[INFO] Tests run: 2, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.004 s - in jmri.TimebaseRateExceptionTest
[INFO] Running jmri.RunCucumberTest

No output has been received in the last 10m0s, this potentially indicates a stalled build or something wrong with the build itself.
Check the details on how to adjust your build configuration on: https://docs.travis-ci.com/user/common-build-problems/#Build-times-out-because-no-output-was-received
```

closed 2019-11-01 by danielb987