JMRI/JMRI #6820: Clock rate setting to non-integer rate does not work

newHeiko, 2019-04-19:
Quick problem description:

Running JMRI 4.14 with a Loconet Simulator, I cannot set the Fast Clock to a non-integer rate even though I'm selecting "Internal Computer Clock" as the time source.

Running JMRI 4.14 with a real Loconet connection (PR3 + DCS51) this works for some arcane reason. Running master, this does not work either.

My use case for this is to provide fast clock time via the webserver, totally unrelated to the loconet, while at the same time providing a withrottle server which feeds the loconet connection.

Of course I could run two separate instances of JMRI, and this may be the path to go, but so far it worked...

Glancing through the code and the issues here (particularly #6256) I believe a general re-factoring of our clock system (and its configuration) could be in order (and I will add a few ideas there), but to fix this quickly, I created PR #6819 to allow setting non-integer rates again. And I'm not that deep into JMRI internals to understand all the InstanceManager-stuff going on in there...

All the best,
Heiko

P.S: This is my first PR, so please let me know what I can do better next time :)

devel-bobm, 2019-04-19:
There are a couple of challenges in supporting integral fast clock rates with LocoNet systems:
- LocoNet does not support non-integer fast-clock rates.
- LocoNet does not support updates of fast-clock time to tetherless "simplex" throttles.

The first issue above means that the only obvious and practical way to implement non-integer fast clock rates would be to send the new time every fast-minute.  The second issue above means that tetherless "simplex" throttles won't see those messages.  (I am not sure about tetherless "duplex" throttles.)

The above two issues make this kind of solution practical _only_ for users who do not have "simplex" (or duplex?) throttles which are capable of showing the fast clock, or for layouts where users are properly trained to ignore the fast-clock time displayed on the throttle display.

For those users who want/need tetherless "simplex" LocoNet throttles to "follow" fast-clock time, having fractional fast-clock rates is problematic, and very likely result in non-trivial end-user support issues.

I would like to propose an alternate approach for your consideration.  Perhaps the WiThrottle Server can be configured to get its "fast clock" source from something other than LocoNet, even though it is "serving" locos via LocoNet.  I am not familiar with WiThrottle Server, so cannot assess the feasibility of such an approach.  But such an approach could leave LocoNet code "untouched" and could potentially give increased "fast-clock"  flexibility for all WiThrottle Server users, no matter what system they use.

newHeiko, 2019-04-19:
> I would like to propose an alternate approach for your consideration. Perhaps the WiThrottle Server can be configured to get its "fast clock" source from something other than LocoNet, even though it [...]

I knew I probably wasn't too clear on the issue - that's exactly what I'm trying to do... I set the very first drop-down in this window: http://jmri.org/help/en/package/jmri/jmrit/simpleclock/SimpleClockFrame.shtml to Internal Computer Clock and still can't set non-integer rates if there is a loconet connection anywhere in the same configuration profile.

If a loconet clock is selected as the clock source, my PR still rejects non-integer rates.

Heiko

devel-bobm, 2019-04-19:
I suspect that this is getting bogged down by the `instanceManager.getDefault()` call.  Exactly how that returns a "clock manager" will determine the behavior of the `SimpleClockFrame`.  I believe that order of connections can influence the behavior of `getDefault`.

Alternately, perhaps this "problem" justifies inclusion of an additional "default" setting - for the "fast-clock source".

FYI - I don't think that "rounding" a non-integer rate would be the best choice.  Assuming that the fast clock "master" updates the fast minute every minute, I would use the next integer rate which is slower than the non-integer rate (i.e. "truncate").  In other words, 1.5:1 on the master translates to 1:1 on the slave.  That way any fast clock "slave" will be be slower than the master, and will be "forced" to the new, correct fast minute, before its local time has moved to that minute.

newHeiko, 2019-04-21:
> FYI - I don't think that "rounding" a non-integer rate would be the best choice. Assuming that the fast clock "master" updates the fast minute every minute, I would use the next integer rate which [...]

That sounds like a good plan to me.

Now summer is coming, so I'm not really looking forward to more coding, but if I have some time on trains or similar, I'll try to come up with some more generic refactoring "AdvancedTimeBase" and "AdvancedClockFrame"... but that's beyond the scope if this issue.

PR #6819 (2019-05-08): Plaster over clock rate setting bug...
...when hardware provides a clock but user selects internal computer
clcok as time source.

Fixes #6820
commit 2ebbb9bf (2019-05-08): Merge branch 'master' of https://github.com/JMRI/JMRI into fixComputerClockRateSetting

Add this comment to properly link with issue: Fixes #6820
closed 2019-05-08 by bobjacobsen