JMRI/JMRI #6821: Clock rate textfield is over-written every fast minute on master

newHeiko, 2019-04-19:
While diving into #6820 I found another issue:

On master, the textfield used for entering the fast clock rate gets overwritten every fast clock minute by SimpleClockFrame/propertyChange -> updateRate()

This does not happen on 4.14.

This makes actually changing the clock rate difficult unless the clock is stopped first (or the rate is small enough that you have the time to enter a new value and push the "Set Rate" button before the next minute)

I don't have a clear idea how to fix this yet, it might also be part of a new clock system, but I'm going to forget if I don't open an issue here :)

Happy Easter,
Heiko

devel-bobm, 2019-04-19:
This is _likely a feature_ of the LocoNet FastClock implementation.  LocoNet messaging only allows setting both the time _and_ the rate.  There is no LocoNet message to set the fast-clock time without setting the fast-clock rate.  And there is no LocoNet message to set the fast-clock rate without setting the fast-clock time.

Other connection types could have similar issues if their messaging supports only a single message for both fast-rate and fast-time.  I do not have directly-related experience with other systems, so can only mention it as something to watch out for.

danielb987, 2019-04-20:
Is there a need to trigger a propertyChange event if the fast clock rate is set to the current rate?

Isn't it reasonable to have the code something like:

```
if (abs(newRate-oldRate) > 0.001) {
    // trigger propertyChange
}
```

pabender, 2019-04-20:
Sent from my iPhone

> On Apr 19, 2019, at 8:59 PM, Daniel <notifications@github.com> wrote:
> 
> Is there a need to trigger a propertyChange event if the fast clock rate is set to the current rate?
> 
You are only supposed to fire a property change if there is a change.
> Isn't it reasonable to have the code something like:
> 
> if (abs(newRate-oldRate) > 0.001) {
>     // trigger propertyChange
> }

If you are checking for equality, then you want to do do with tolerance, so it is reasonable for inequality as well.

newHeiko, 2019-04-20:
I believe - with my limited grasp of how the clock system works - the change triggering the propertyChange event is the change in fast clock time, which happens every fast clock minute.

One idea would be to expand the ClockControl interface with a RateChange and a TimeChange event, the other to keep a shadow copy of the fast clock rate input field and only overwrite the text input field if the new rate from the propertyChange event is different than the shadow copy.

But I'd like to start somewhere different, with a more straightforward system for the clock configuration, making clearer which configuration settings belong to which part.

Trying to follow #6371 as well and contribute to it. And maybe then this issue will also be solved in the process.

Heiko

rhwood, 2019-04-20:
A listener can listen for all property change events from the listener to object, or only select events. In 4.15.5, the `time` event was added to `SimpleTimeBase` but so was the ability to listen to only select events from that class. It’s entirely possible that when that happened, a listener was not adjusted to either not listen to the `time` or `minute` events, or was not adjusted to only change the rate in the UI if the `rate` event is sent.

rhwood, 2019-04-20:
The propertyChange method on line 774 of SimpleClockFrame updates the rate field in the UI for *every* PropertyChangeEvent fired by SimpleTimeBase; it needs to be updated to only set the rate field if a `rate` PropertyChangeEvent is fired and to update the running button only if a `run` PropertyChangeEvent is fired. See the PropertyChangeEvent Javadoc for the API for getting the property name from an event.

newHeiko, 2019-04-21:
> A listener can listen for all property change events from the listener to object, or only select events. In 4.15.5, the `time` event was added to `SimpleTimeBase` but so was the ability to listen to [...]

Thanks for pushing my nose to the fact that the event originates from SimpleTimeBase, I thought it came from deeper within the clock system.

Since for some reason I didn't get any notification from your PR, I also created  (less invasive, probably your attempt is the first step in a larger refactoring).

I leave it to you guys to decide which way to go with the code

PR #6830 (2019-04-23): Fix: SimpleClockFrame only updates changed properties
Fix #6821 by ensuring the SimpleClockFrame only updates UI elements changed by a changed property when listening to changes in SimpleTimeBase.

Also removes use of some deprecated methods.
commit fce3625a (2019-04-23): fix: act only on changed property

Fixes #6821
closed 2019-04-23 by bobjacobsen