JMRI/JMRI #6826: Web Server - MultiSensor - not change state / icon - 4.15.5

sidlo64, 2019-04-20:
MultiSensor works normally in Panel Editor.
In Web Server panels the same sensor not change state / icon.
All other icons - turnouts, sensors, signal masts - are OK.
Affected version 4.15.4 and next.
Production version 4.14 is OK.

This issue was reported in #6765 and expected to be fixed in version 4.15.5, but the bug still persists.

rhwood, 2019-04-20:
Please provide a test profile showing the problem with specific steps to reproduce.

sidlo64, 2019-04-20:
[BugMultiSensor.jmri.zip](https://github.com/JMRI/JMRI/files/3100121/BugMultiSensor.jmri.zip)
JMRI 4.15.5
In zip file is TestProfile.xml. 
Connection Digitrax LocoNet Simulator (no meaning).
Open TestProfile.xml and start Web Server.
In Panel Editor click buttons and levers follows buttons.
Open browser and Test Profile panel.
Click buttons - but **levers does not follow buttons**.
When you refresh web page (F5) the lever take correct position.

PR #6832 (2019-04-23): Fix: Use SystemNames for Sensors in MultiSensorIcon in web panel
Fixes #6826 by ensuring that Sensors in a MultiSensorIcon in a web panel are created with SystemNames instead of UserNames.

Also:
- for any passed in name for a named bean in the JSON Streaming API, the name is switched to a system name for further processing
- remove some duplication of code when creating XML for web panels
commit 22efae69 (2019-04-23): fix: force JSON name to system name when processing

Fixes #6826 by ensuring that, for any passed in name for a named bean, the name is switched to a system name for further processing.
closed 2019-04-23 by bobjacobsen