JMRI/JMRI #6827: openlcb.PackageTest leaves too many threads

bobjacobsen, 2019-04-20:
When run as part of comprehensive tests,`openlcb.PackageTest` leaves hundreds of threads running. This overwhelms various diagnostic tools (i.e. from ctrl-\ to NetBeans stack tracer) during later parts of a test run.

[image]

pabender, 2019-04-20:
I would actually go one step further.

I think the JMRI OpenLCB tests start more threads than are required for unit testing.  When last I dealt with this, we reduced the number of residual threads from 

There also are at least a few OpenLCB test classes that test the pieces of the OpenLCB library.  Those really shouldn’t be in the JMRI tests.

( also I just spotted that https://github.com/JMRI/JMRI/blob/master/java/test/jmri/jmrix/openlcb/CanConverterTest.java doesn’t actually test anything in the JMRI.jmrix.openlcb subtree, and one of the things it creates is an object from the library....)

pabender, 2019-04-23:
@bobjacobsen that step up between 1:40 and 1:50 is 30 or 40 threads identified as BlockBossLogic.  We should probably figure out how to terminate those threads.  There are also a couple BlockBossLogic threads created pretty early in the process, before the OpenLCB threads start.

bobjacobsen, 2019-04-23:
@pabender  I think that group is left over from the XML file loading tests. Those don't really clean up created objects except to re-init the InstanceManager.  I'll take a look at making that smarter.

bobjacobsen, 2020-04-22:
Somehow, the OpenLCB tests are firing off _new_ threads long after they OLB tests themselves are complete.  Perhaps the threads or timers left behind spawn new ones?  Some of these are deep enough in other tests that they may be causing intermittent problems.  

Can we move this to the "run separately" category?  That's a by-test annotation, unfortunately.

Running with these options:
```
JMRI_OPTIONS=-Duser.country=US -Duser.language=en -Djmri.util.JUnitUtil.checkSetUpTearDownSequence=true -Djmri.util.JUnitUtil.checkRemnantThreads=true -Djmri.util.JUnitUtil.printSetUpTearDownNames=true -Djmri.util.JUnitUtil.checkSequenceDumpsStack=true
```

gives you a trace of when new threads are present at the end of tests:

```
>> Starting test in jmri.jmrix.openlcb.swing.send.OpenLcbCanSendPaneTest
<<   Ending test in jmri.jmrix.openlcb.swing.send.OpenLcbCanSendPaneTest
WARN  - Found remnant thread "OpenLCB Memory Configuration Service Retry Timer" in group "main" after jmri.jmrix.openlcb.swing.send.OpenLcbCanSendPaneTest [main] jmri.util.JUnitUtil.lambda$checkThreads$4()
WARN  - Found remnant thread "OpenLCB-datagram-timer" in group "main" after jmri.jmrix.openlcb.swing.send.OpenLcbCanSendPaneTest [main] jmri.util.JUnitUtil.lambda$checkThreads$4()
WARN  - Found remnant thread "OpenLCB Mimic Node Store Timer" in group "main" after jmri.jmrix.openlcb.swing.send.OpenLcbCanSendPaneTest [main] jmri.util.JUnitUtil.lambda$checkThreads$4()
WARN  - Found remnant thread "OpenLCB LoaderClient Timeout Timer" in group "main" after jmri.jmrix.openlcb.swing.send.OpenLcbCanSendPaneTest [main] jmri.util.JUnitUtil.lambda$checkThreads$4()
WARN  - Found remnant thread "Olcb-Pool-3-Thread-0" in group "main" after jmri.jmrix.openlcb.swing.send.OpenLcbCanSendPaneTest [main] jmri.util.JUnitUtil.lambda$checkThreads$4()
WARN  - Found remnant thread "Olcb-Pool-3-Thread-1" in group "main" after jmri.jmrix.openlcb.swing.send.OpenLcbCanSendPaneTest [main] jmri.util.JUnitUtil.lambda$checkThreads$4()
WARN  - Found remnant thread "OpenLCB NIDaAlgorithm Timer" in group "main" after jmri.jmrix.openlcb.swing.send.OpenLcbCanSendPaneTest [main] jmri.util.JUnitUtil.lambda$checkThreads$4()
>> Starting test in jmri.jmrix.openlcb.swing.send.OpenLcbCanSendPaneTest
<<   Ending test in jmri.jmrix.openlcb.swing.send.OpenLcbCanSendPaneTest
WARN  - Found remnant thread "Olcb-Pool-4-Thread-0" in group "main" after jmri.jmrix.openlcb.swing.send.OpenLcbCanSendPaneTest [main] jmri.util.JUnitUtil.lambda$checkThreads$4()
WARN  - Found remnant thread "OpenLCB-datagram-timer" in group "main" after jmri.jmrix.openlcb.swing.send.OpenLcbCanSendPaneTest [main] jmri.util.JUnitUtil.lambda$checkThreads$4()
WARN  - Found remnant thread "Olcb-Pool-4-Thread-1" in group "main" after jmri.jmrix.openlcb.swing.send.OpenLcbCanSendPaneTest [main] jmri.util.JUnitUtil.lambda$checkThreads$4()
WARN  - Found remnant thread "OpenLCB Memory Configuration Service Retry Timer" in group "main" after jmri.jmrix.openlcb.swing.send.OpenLcbCanSendPaneTest [main] jmri.util.JUnitUtil.lambda$checkThreads$4()
>> Starting test in jmri.jmrix.openlcb.swing.send.OpenLcbCanSendPaneTest
<<   Ending test in jmri.jmrix.openlcb.swing.send.OpenLcbCanSendPaneTest
WARN  - Found remnant thread "Olcb-Pool-5-Thread-0" in group "main" after jmri.jmrix.openlcb.swing.send.OpenLcbCanSendPaneTest [main] jmri.util.JUnitUtil.lambda$checkThreads$4()
WARN  - Found remnant thread "OpenLCB Memory Configuration Service Retry Timer" in group "main" after jmri.jmrix.openlcb.swing.send.OpenLcbCanSendPaneTest [main] jmri.util.JUnitUtil.lambda$checkThreads$4()
WARN  - Found remnant thread "OpenLCB-datagram-timer" in group "main" after jmri.jmrix.openlcb.swing.send.OpenLcbCanSendPaneTest [main] jmri.util.JUnitUtil.lambda$checkThreads$4()
WARN  - Found remnant thread "Olcb-Pool-5-Thread-1" in group "main" after jmri.jmrix.openlcb.swing.send.OpenLcbCanSendPaneTest [main] jmri.util.JUnitUtil.lambda$checkThreads$4()
>> Starting test in jmri.jmrix.openlcb.swing.send.OpenLcbCanSendPaneTest
<<   Ending test in jmri.jmrix.openlcb.swing.send.OpenLcbCanSendPaneTest
WARN  - Found remnant thread "Olcb-Pool-6-Thread-0" in group "main" after jmri.jmrix.openlcb.swing.send.OpenLcbCanSendPaneTest [main] jmri.util.JUnitUtil.lambda$checkThreads$4()
WARN  - Found remnant thread "Olcb-Pool-6-Thread-1" in group "main" after jmri.jmrix.openlcb.swing.send.OpenLcbCanSendPaneTest [main] jmri.util.JUnitUtil.lambda$checkThreads$4()
WARN  - Found remnant thread "OpenLCB Memory Configuration Service Retry Timer" in group "main" after jmri.jmrix.openlcb.swing.send.OpenLcbCanSendPaneTest [main] jmri.util.JUnitUtil.lambda$checkThreads$4()
WARN  - Found remnant thread "OpenLCB-datagram-timer" in group "main" after jmri.jmrix.openlcb.swing.send.OpenLcbCanSendPaneTest [main] jmri.util.JUnitUtil.lambda$checkThreads$4()
>> Starting test in jmri.jmrix.openlcb.swing.send.OpenLcbCanSendPaneTest
<<   Ending test in jmri.jmrix.openlcb.swing.send.OpenLcbCanSendPaneTest
WARN  - Found remnant thread "Olcb-Pool-7-Thread-0" in group "main" after jmri.jmrix.openlcb.swing.send.OpenLcbCanSendPaneTest [main] jmri.util.JUnitUtil.lambda$checkThreads$4()
WARN  - Found remnant thread "OpenLCB Memory Configuration Service Retry Timer" in group "main" after jmri.jmrix.openlcb.swing.send.OpenLcbCanSendPaneTest [main] jmri.util.JUnitUtil.lambda$checkThreads$4()
[... cut ...]
WARN  - Found remnant thread "olcbCanInterface.initialize" in group "main" after jmri.jmrix.loconet.Ib1ThrottleTest [main] jmri.util.JUnitUtil.lambda$checkThreads$4()
>> Starting test in jmri.jmrix.loconet.Ib1ThrottleTest
<<   Ending test in jmri.jmrix.loconet.Ib1ThrottleTest
WARN  - Found remnant thread "olcbCanInterface.initialize" in group "main" after jmri.jmrix.loconet.Ib1ThrottleTest [main] jmri.util.JUnitUtil.lambda$checkThreads$4()
>> Starting test in jmri.jmrix.loconet.Ib1ThrottleTest
<<   Ending test in jmri.jmrix.loconet.Ib1ThrottleTest
```

pabender, 2020-04-23:
@bobjacobsen the olcbCanInterface.initialize thread is actually one that is created in jmri.jmrix.openlcb.OlcbConfigurationManager

i.e. it is in our code, not the Olcb library.  The ThreadPools are created by the Olcb library.

closed 2021-04-14 by stale[bot]