JMRI/JMRI #6828: Threading issue in CbusSlotMonitorDataModel

bobjacobsen, 2019-04-20:
`CbusSlotMonitorDataModel` is an `javax.swing.table.AbstractTableModel`.   As such, it needs to fire notifications to Swing components for display, etc as part of an MVC pattern.

When run as part of the `ant logalltest` target used by JaCoCo, it emits messages like:

```
    [junit] WARN  - Thread was holding monitor jmri.jmrix.can.cbus.swing.cbusslotmonitor.CbusSlotMonitorDataModel@2f03702e from jmri.jmrix.can.cbus.swing.cbusslotmonitor.CbusSlotMonitorDataModel.createnewrow(CbusSlotMonitorDataModel.java:325) [main] jmri.util.ThreadingUtil.?()
    [junit] WARN  - Thread was holding monitor jmri.jmrix.can.cbus.swing.cbusslotmonitor.CbusSlotMonitorDataModel@2f03702e from jmri.jmrix.can.cbus.swing.cbusslotmonitor.CbusSlotMonitorDataModel.processploc(CbusSlotMonitorDataModel.java:527) [main] jmri.util.ThreadingUtil.?()
```

These come from `jmri.util.ThreadingUtil` when doing i.e. a switch to the GUI thread via a `ThreadingUtil.runOnGUI` call.  For example, `CbusSlotMonitorDataModel.processploc` does:

```
    private synchronized void processploc(boolean messagein, int session, DccLocoAddress addr,
        int speeddir, int fa, int fb, int fc) {
        // log.debug( Bundle.getMessage("CBUS_CMND_BR") + Bundle.getMessage("CNFO_PLOC",session,locoid));
        
        int row = provideTableRow(addr);
        setValueAt(session, row, SESSION_ID_COLUMN);
        setValueAt(speeddir, row, LOCO_COMMANDED_SPEED_COLUMN);
        processdfun( messagein, session, 1, fa);
        processdfun( messagein, session, 2, fb);
        processdfun( messagein, session, 3, fc);

    }
```

That's a synchronized method, so it takes a monitor on the object.  But then (via the setValueAt method at line 268, which calls `updateGui` on line 310) it does a `ThreadingUtil.runOnGUI` _before_ releasing the monitor. 

I can't tell from an initial read of the code why the `synchronized` is necessary.  If not, it should be removed.  (If that flags inconsistent synch, then perhaps the others can be removed instead of putting this one back, but maybe not)  

If this does have parts that need to be synchronized, then separating those out from the GUI calls would be a solution.

icklesteve, 2019-04-20:
Well done for spotting @bobjacobsen , I had a look over it yesterday but had missed that.

The synchronized has been there since initial upload so potentially a legacy copy / paste, none of the other processXXX private routines have it.

I'll just run a few tests then upload,
Steve

bobjacobsen, 2019-04-20:
Looks like `createnewrow` (line 318) is the other `synchronized` call.  The only thing I see there that might require synchronization is the add to `_mainArray`.  If that's the case, switching to a self-synchronized list (line 66) will fix it:
```
_mainArray = = Collections.synchronizedList( new ArrayList<CbusSlotMonitorSession>() );
```
But not the caution about iterating in the [Javadoc](https://docs.oracle.com/javase/7/docs/api/java/util/Collections.html#synchronizedList(java.util.List)) I don't see any iterations over `_mainArray`, so this should be OK.

bobjacobsen, 2019-04-20:
Closed by #6829

closed 2019-04-20 by bobjacobsen