JMRI/JMRI #6837: NodeConfigToolPaneTest causing exception in AllTest

icklesteve, 2019-04-22:
Hi All,

In latest separate tests log at
http://builds.jmri.org/jenkins/job/Development/job/Separate_Tests/376/consoleFull
there is 1 test with an exception, jmri/jmrix/can/cbus/swing/nodeconfig/NodeConfigToolPaneTest

The line it's failing on is a standard JTable.getRowCount() which indicates a particular table isn't initialising OK which I'll upload a fix for v soon.

The test itself isn't one of my best, there's no assert, instead assuming any exceptions will be flagged, something I'll change.

I was under the impression though, that a test would auto fail with an uncaught exception?

```
jmri/jmrix/can/cbus/swing/nodeconfig/NodeConfigToolPaneTest
Picked up JAVA_TOOL_OPTIONS: -Dfile.encoding=UTF-8
.Using org.netbeans.jemmy.drivers.DefaultDriverInstaller driver installer
.Exception in thread "AWT-EventQueue-0" java.lang.NullPointerException
	at jmri.jmrix.can.cbus.node.CbusNodeTableDataModel.getRowCount(CbusNodeTableDataModel.java:125)
	at javax.swing.table.TableRowSorter$TableRowSorterModelWrapper.getRowCount(TableRowSorter.java:265)
	at javax.swing.DefaultRowSorter.getViewRowCount(DefaultRowSorter.java:815)
	at javax.swing.JTable.getRowCount(JTable.java:2660)
) .. .. .. ..

Time: 1.092

OK (2 tests)
```

bobjacobsen, 2019-04-22:
The top of that Exception traceback shows that it started in the AWT/Swing thread, running through Swing code itself (it didn't start in the test thread).  Swing's handling of exceptions during UI processing is a little complicated, and we don't always catch Exceptions that happen there during testing.  

Might be an interesting project to figure out how to install a complete set of error handlers to fix that.

icklesteve, 2019-04-23:
Thanks Bob, that makes sense now.
That sounds like a worthwhile project but a bit beyond myself at present!

Issue not resolved by #6838 as per console log at http://builds.jmri.org/jenkins/job/Development/job/Separate_Tests/378/consoleFull
so further attempt at #6845

icklesteve, 2019-05-05:
Confirmed this particular exception resolved after checking
http://builds.jmri.org/jenkins/job/Development/job/AllTest/7/consoleFull

closed 2019-05-05 by icklesteve