JMRI/JMRI #6871: XML test passes for bad Logix systemNames

KenC57, 2019-04-28:
Very old panel files did not use the IX prefix in the Logix systemName field. The Validate XML test does not detect this and says 'Ok'. Later parts of loading the file then causes the Logix create to fail and then later errors of the conditionals not finding a Logix to bind with.
-ken c

pabender, 2019-04-28:
> On Apr 28, 2019, at 7:59 AM, Ken Cameron <notifications@github.com> wrote:
> Very old panel files did not use the IX prefix in the Logix systemName field. The Validate XML test does not detect this and says 'Ok'. Later parts of loading the file then causes the Logix create to fail and then later errors of the conditionals not finding a Logix to bind with.
> 
How old?

Do we need to add a test that catches this?

Do we need to ask the user to load the file in an intermediated version that can read the old format but writes the new one?

Paul

bobjacobsen, 2019-04-28:
@KenC57 Could you please check if the release-note instructions to read and write the file with JRMI 4.12 fix this?

If not, please append the file (might have to add .txt to the name)

KenC57, 2019-04-28:
Paul,
The version he was on was 3.0 or 3.10, something in the 3's. I'm not sure
which version started writing with the IX or how that transition went. I
guess it would be good to document which was the last version that read
without the IX but wrote with the IX.

At least the XML should fail if it won't read the file successfully.
-ken c

dsand47, 2019-04-28:
The lack of IX is not the issue.  It was a numeric system name, but jmri.Manager requires that the first character be a letter.  XML validation was successful because it only requires at least two characters.

The systemNameType pattern in general.xsd should be changed.

```
  <xs:simpleType name="systemNameType">
    <xs:annotation>
        <xs:documentation>
          General definition of "systemName" attribute type.
          Centralized so we can later enforce naming, linking, etc.
        </xs:documentation>
    </xs:annotation>
    <xs:restriction base="xs:token">
        <!-- "[A-Z]([0-9]*)[A-Z].*" is more correct, but fails lots of short signal head names -->
        <!-- type letter pattern can also be constrained: ABCHLMOPRSTX but also Y? -->
        <!-- the following is "at least two characters" -->
             <xs:pattern value="..*"/>
    </xs:restriction>
  </xs:simpleType>
```
Setting the pattern to `[A-Z].*` will satisfy the jmri.Manager requirement.

bobjacobsen, 2019-04-29:
@dsand47: I've got a branch with a full set of validations set for after 4.16 is released.  

Note that there's no requirement that the 1st character be an upper case letter; lower case is fine.

@KenC57:
> I guess it would be good to document which was the last version that read
without the IX but wrote with the IX.

That's why I asked if converting through JMRI 4.12 fixed this.  That's the version that's supposed to be the conversion line.  Did it work?

KenC57, 2019-04-29:
I haven't had a chance to test with 4.12, my usual systems don't go back
that far. The original file is 17M with about 16M of the repeating block
routing bug. I'll see what I can find.

dsand47, 2019-04-29:
As far as I can tell, there is no automatic conversion for bad Logix system names.  

At 4.10 a bad name loses it conditionals but the Logix itself survives.  At 4.12 a
jmri.NamedBean$BadSystemNameException occurs and the Logix and its conditionals no longer exist.

dsand47, 2020-01-13:
> As far as I can tell, there is no automatic conversion for bad Logix system names.
> 
PR #7894 fixes bad Logix system names during loading by using getSystemNamePrefix().

bobjacobsen, 2020-08-19:
Fixed by #7894

closed 2020-08-19 by bobjacobsen