JMRI/JMRI #6876: CTC Windows use custom storage mechanisms

rhwood, 2019-04-29:
The CTC Windows do not extend JmriJFrame and consequently use custom mechanisms for storing all preferences and settings like window size and position that JMRI provides standard mechanisms for handling that are not used.

The CTC system may also have other settings that the `jmri.UserPreferencesManager` or `jmri.util.prefs.JmriPreferencesProvider` could be used to store in a way that allows the CTC windows to behave in ways similar to other JMRI windows.

dsand47, 2019-04-29:
The CTC Editor windows are built and maintained using the NetBeans IDE and pre-date integration into JMRI.  This also applies to the ProgramProperties.xml file.

dsand47, 2021-03-30:
TODO:  Review using user-interface.xml tools instead of AwtWindowsProperties.txt.

dsand47, 2021-04-23:
PR #9671 moved the window location and size to the standard JMRI process.

closed 2021-04-23 by dsand47