JMRI/JMRI #6877: Throttle Steal / Share Implementation

icklesteve, 2019-04-29:
Hi All,

I've been thinking about how best to introduce the Steal or Share Throttle feature in MERG CBUS.

Please see proposed addition to ThrottleListener flow diagram.

Would also add to throttle preferences a "Silent Share" boolean option, complementing "Silent Steal".
This would be the preferred way for ThrottleListener scripts like AutoDispatcher to respond to the option.

Thoughts appreciated,
Steve.

[image]

Text here if anyone wants to propose tweaks, edited with https://www.planttext.com/
```
 @startuml
  participant ThrottleListener
  participant ThrottleManager
  participant Throttle
 
  == Quick Failure Scenario ==
 
  ThrottleListener-> ThrottleManager : requestThrottle(address,ThrottleListener)
  group If Throttle Request Cannot Continue
     ThrottleListener <-- ThrottleManager : return false 
     deactivate ThrottleManager
     note over ThrottleListener : Request ends here
  end
  deactivate ThrottleListener
 
  == Normal Request Scenario ==
  
  ThrottleListener-> ThrottleManager : requestThrottle(address,ThrottleListener)
 
  group If Throttle Request Can Continue
     ThrottleListener <-- ThrottleManager : return true
     note over ThrottleListener : Waits for callback to proceed.
     group If Throttle Object Does not exist
        note over ThrottleManager : Throttle Manager start system specific actions requred to create a throttle.
     == Optional Steal ==
        ThrottleListener <-- ThrottleManager : notifyStealThrottleRequired(address)
        note over ThrottleListener : To steal or not is determined by the throttle Listener. 
        group If the Listener does not wish to steal
            ThrottleListener --> ThrottleManager : stealThrottleRequest(address info, false )
            note over ThrottleListener : Request ends here
        end
        group If the Listener wishes to steal
            ThrottleListener --> ThrottleManager : stealThrottleRequest(address info, true)
                
                note over ThrottleManager : Throttle creation continues normally.
            end
 
 
      == Optional Steal or Share PROPOSAL==
        ThrottleListener <-- ThrottleManager : notifyStealOrShareThrottleRequired(address)
        note over ThrottleListener : To steal or share is determined by the throttle Listener.\nIdeally the listener should check for either isSilentSteal() or isSilentShare() being set.

        group If the Listener does not wish to steal or share
            ThrottleListener --> ThrottleManager : shareThrottleRequest(address info, false )
            note over ThrottleListener : Request ends here
        end
        
        group If the Listener wishes to steal
            ThrottleListener --> ThrottleManager : stealThrottleRequest(address info, true)
            note over ThrottleManager : Throttle creation continues normally.
        end
        
        group If the Listener wishes to share
            ThrottleListener --> ThrottleManager : shareThrottleRequest(address info, true )
            note over ThrottleManager : Throttle creation continues normally.
        end
 
 
     == Creating the throttle  ==
        ThrottleManager --> Throttle : new Throttle(address)
        group If the Throttle creation fails
           ThrottleListener <-- ThrottleManager: notifyFailedThrottleReqest(address)
           note over ThrottleListener : Request ends here
        end                    
        group If the Throttle creation request succeeds
           ThrottleManager <-- Throttle: return new Throttle
        end
     end
     
  ThrottleListener <-- ThrottleManager: notifyThrottleFound(Throttle)
  
  note over ThrottleListener : Throttle can now be controlled by ThrottleListener or a delegate.
  
  ThrottleListener-> Throttle : Registers as propertyChangeListener
  

  @enduml
```

pabender, 2019-04-29:
@icklesteve 

Most systems don't actually support the concept of stealling.  

As for sharing, we generally do what physical throttles do, but on most systems we just don't have any idea if another throttle is in control of the address.  Also, a single JMRi instance does not share or steal with/from itself.  JMRI is to be considered a single physical throttle for this purpose.

LocoNet brings the steal concept out to the user interface and actually prompts the user for what to do.  The current steal implementation was added to handle this single use case.  As expressed in user documentation, Digitrax never mentions sharing, but we know it happens.

XpressNet does not really have either a steal or a share concept.  The command station will notify a device that some other device is now in control, but the throttle itself doesn't have a choice in the matter (Physical throttles give an indication they are not in control by blinking the address.  JMRi polls the command station periodically to check for current status ).

I don't believe that either CVP or NCE systems will tell us what is going on with respect to other throttles.

I don't know how other systems work in this respect.

Does MERG support either stealing or sharing on physical throttles?

bobjacobsen, 2019-04-29:
If we're rethinking this, can I make a different suggestion? Instead of relying on multiple callbacks, just specify the operation from the start:

 - In addition to the existing `requestThrottle(address, ThrottleListener)` method, add a new `requestThrottle(address, ThrottleListener, ThrottleManager.RequestType)` method, where `ThrottleManager.RequestType` is an `enum` specifying one of three request behaviors if the throttle is already in use:  `STEAL`, `SHARE`, and `DROP`.
- The existing `requestThrottle(address, ThrottleListener)` method would have the same behavior with a system-specific default for the `ThrottleManager.RequestType` argument.
 - Remove the `notifyStealThrottleRequired` et al callbacks
 - Provide a way to ask the Throttle whether it's in "SHARE" or "EXCLUSIVE" or "UNKNOWN" mode (not all hardware can tell; note that this is a time-varying thing, hence a bound parameter, since somebody _else_ can share the throttle later)

This would eliminate the need to program callbacks (which are a pain for scripting beginners), and make it really clear what the request is to do.

pabender, 2019-04-29:
@bobjacobsen for scripting, we already hide a lot of this from the user because of the notifyThrottleFound behavior.

in particular, the process started by getThrottle() in jmri.jmrit.automat.AbstractAutomaton uses an automatic steal implementation.
(getThrottle predates the addition of steal...)

pabender, 2019-04-29:
The only issue I have with eliminating the callbacks is how do you know for certain if the steal needs to be handled if the command station doesn't tell you unless you ask for a throttle in use?

The callbacks correspond to the traffic on the network.

bobjacobsen, 2019-04-29:
Why does the throttle-requesting code need to know if a steal has to be done?  The requestor can just specify how it wants that to be handled, should it be needed, and leave it at that.

pabender, 2019-04-29:
> Why does the throttle-requesting code need to know if a steal has to be done? The requestor can just specify how it wants that to be handled, should it be needed, and leave it at that.

Strictly speaking, it isn't, but the thought is that on systems where stealing is possible, we want the user to make the choice.  We don't want the user to be asked the question if it isn't required.

rhwood, 2019-04-29:
The DT402 throttles check if an address is in use and prompt the user to confirm they want to steal the address if it is use, and just take the address if the user does not need to be prompted. That's the behavior the steal callbacks allow JMRI to emulate.

icklesteve, 2019-04-29:
The steal / share mechanism has been standard for MERG handsets since about 2013.

Currently, trying to using it at the same time as JMRI causes major errors though this is more to do with the mechanism for cancelling sessions in dynamic stacks, something I'll deal with in another thread.

All of the non-UI ThrottleListeners that I've looked at default to Steal in the steal scenario.
That would not be the preferred behaviour for steal / share, if the user has set the option for it, use it, else default to share?

Plan was to reduce visibility of the feature by either hiding the silent share tickbox in Throttle Preferences Pane, or disabling it,
new abstracted throttle manager boolean showSilentShareOption() method, defaulting to false..

As per CBUS dev. guide 6b, pg. 20  https://www.merg.org.uk/merg_wiki/lib/exe/fetch.php?media=public:cbuspublic:developer_6b.pdf the procedure is roughly

Listener > Throttle         Request Throttle Loco 1
Throttle  > Layout          Request Session Loco 1
Layout > Throttle           Session Denied, loco in use

Where the current JMRI implementation ends

Throttle > Listener         Denied but can steal or share
Listener > Throttle         Popup steal / share, if user preference unset
Throttle > Layout           Request share ( or steal ) for loco 1, no request sent if cancelled.
Layout > Throttle           Here's your session
Throttle > Listener         Here's your throttle

So CbusThrottleManager wouldn't know if steal / share is available until after the initial ( normal ) request.

I'm in the process of creating a UML for throttle creation / destruction, so perhaps our thinking may change depending on what that shows!

Pugwash1, 2019-04-30:
@rhwood To expand on this, the new Digitrax command stations allow the true stealing of a throttle, the aquierer is prompted if the want to steal, and the throttle they steal from is notified and can no longer control the loco.
Our Club/Group is very interested in applying this to WiFi throttles.Currently we use a custom build that has loconet as not having shareable throttles.
We think @icklesteve idea is well worth the building of the prototype to see how it works, and then tweaking if it's worth while.

pabender, 2019-04-30:
> All of the non-UI ThrottleListeners that I've looked at default to Steal in the steal scenario.

Right, we put into place an automatic steal behavior. for most of these applications when we put the steal mechanism in place.

Part of this had to do with not disrupting the user experience.

An example occurs with consisting. In order to add/remove a locomotive from a consist we manipulate the throttle (at a minimum to set the direction, but sometimes to set the speed also).  

On a loconet system, we have to have control over the slot while setting up the consist, but the a) the consist tool itself doesn't know anything about stealing (it shouldn't, that's the throttle's job) and b) The user just wants the consists to work. 

As with many things we have in JMRI, the consist tool was designed to simplify the user's life by hiding the details from them.  This is a use case where @bobjacobsen's suggestion to eliminate the callback would work well.

icklesteve, 2019-04-30:
As to what type of initial request to create a Throttle should be made, I'm not keen on introducing extra logic to ThrottleListeners to make that decision at creation time ( or querying the ThrottleManager for a suggested request prior to a request ).
I think the main request throttle should remain as-is, it currently generally works as expected, and custom hardware methods can deal with it.
Any proposed changes to this should be considered a side add-on to existing, not a revamp!

I've worked out that in CBUS to calculate if a new Throttle request loco address is steal / share capable prior to asking the command station for a session number would require min 6 CAN frames, max 70 CAN frames ( in a typical 32 session, though I'm aware of a 128 session CBUS command station implementation that would require 260ish CAN frames ), would be a best guess at what the command station is likely to do, and would be out of date by the time the final frame was received.  
It's not exactly a slow protocol, but I really don't want to second-guess command stations with something like Throttle creation if I can help it!
Only after that could a reasonably valid decision on what sort of request should be made initially.

I'm also wanting to avoid potential to break existing / future steal-only ( share only? ) hardware types, along with keeping custom Jython changes minimal.
I'm thinking an extra small-ish method in ThrottleListeners for the extra notification would be a lot easier for users to add to any custom Throttle scripts than a complete re-vamp of the initial request!

I'm also thinking that if the user has set something in JMRI preferences, it should be honoured as close as possible ( though I appreciate why some scripts default to try a silent steal if they're running into trouble, especially for headless installs. )

isSilentShare() would be false by default, matching behaviour of isSilentSteal()

The extra ThrottleListener method would be something like

Non-UI
```
@Override
public void notifyStealOrShareThrottleRequired( address) {
    if isSilentSteal() {
        return requestToSteal ...
    }
    else {
        return requestToShare ...
    }
}
```

With UI
```
@Override
public void notifyStealOrShareThrottleRequired( address) {
    if isSilentSteal() {
        return requestToSteal ...
    }
    else if isSilentShare() {
        return requestToShare ....
    }
    else { 
        // Pop a share / steal dialogue to ask the question, 
        // I would update code in jmri.jmrit.AddressPanel

        // if UI for a ThrottleListener isn't developed for a share / steal question but is for steal question
        // , ask the steal question, it's better than nothing and the same callback would work!
 }
```

I've had a look at the general Throttle operation / close process, and have come up with a 1st draft of how things roughly work.
After running into the same issue as #4884 , I've highlighted in the ThrottleManager where an improvement would be really useful for resetting Throttles. ( I may as well tackle it in this thread as it does have Steal in the title! )

Note proposal for new Throttle propertyChangeEvents DispatchEnabled and ReleaseEnabled for any Command Stations with changing throttle Dispatch availability. ( for updating UI only )
This saves any logic for what is currently valid being held in the Throttle, NOT the listener, making it more hardware future-proof and shouldn't break anything existing.
This makes implementation in something like jmri.jmrit.AddressPanel fairly easy, eg. add to within the existing Throttle propertyChangeListener

```
if ("DispatchEnabled".compareTo(evt.getPropertyName()) == 0) {
    log.debug("propertyChange: Dispatch Button Enabled {}" , (Boolean) evt.getNewValue() );
    dispatchButton.setEnabled( (Boolean) evt.getNewValue() );
}
if ("ReleaseEnabled".compareTo(evt.getPropertyName()) == 0) {
    log.debug("propertyChange: release Button Enabled {}" , (Boolean) evt.getNewValue() );
    releaseButton.setEnabled( (Boolean) evt.getNewValue() );
}
```

I also think there's potential to standardise on a "notify throttle no longer valid" propertyChangeEvent.
Though I'd need to look at the current throttleListeners to see what was most commonly used ( especially in any Jython scripts ), go for that and depreciate the other.

[... 103 lines cut ...]
    
    ThrottleManager --> ThrottleListener: If somehow still listeners via Throttle propertyChangeEvent throttleRemoved\nbut can't see how this is possible, Throttle won't be taken off the HashTable if there are still listeners.
    
      ThrottleManager-> Throttle : throttleDispose()
    
    
      end
    
    group Command Station Error condition or Throttle has been Stolen by something exterior to JMRI
    
    
    note over ThrottleManager : PROPOSED super protected forceReleaseThrottle(Throttle)\nremoves Throttle from the core Hashtable,\notherwise new attempts to request a throttle for a locoaddress\nreturn the disposed Throttle and cause an exception\nAlternately make HashTable protected instead of private\nAlternately a get method for the HashTable
    
    
      ThrottleManager-> Throttle : throttleDispose()
      Throttle -> ThrottleListener : Property Change Event ThrottleConnected:false
      
      Throttle --> ThrottleListener : use Property Change Event throttleRemoved instead?
      
    ThrottleListener-> Throttle : removePropertyChangeListener(ThrottleListener)
    
     end

  @enduml
```

pabender, 2019-04-30:
> I also think there's potential to standardise on a "notify throttle no longer valid" propertyChangeEvent.
> Though I'd need to look at the current throttleListeners to see what was most commonly used ( especially in any Jython scripts ), go for that and depreciate the other.
> 

There isn't a general "no longer valid" property at this point in time.

The throttleListener interface should only be used for the acquisition process.  

There is no requirement that the throttleListener actually be able to control the throttle.  It is allowed for the throttle listener to pass the throttle object on to some other object for control.

There is also no requirement that an object controlling a throttle listen to property changes from that throttle.  (The same is true for every other object type we support).

> Unfortunately the https://www.planttext.com/ renderer has a maximum image height, otherwise I'd attempt to combine the Throttle use diagram below with the one posted above ;-)
> Recommendations for alternative no max height UML renderers very welcome!
> 

Try to make each diagram only have one use case.  The diagram you attached is unreadable.

The neither the throttle listener or throttle manager are involved in normal operation of a throttle.  That is between the controlling object (UI,script,etc) and the throttle object.

The throttle listener shouldn't be involved in the dispose process.

Also, you need to remember we treat JMRI as one big throttle, so there may actually be more than one object that holds a reference to that throttle. 

> Also Consisting isn't explored in the flow-chart, again due to max. diagram height, though something I'll explore, though to properly implement in CBUS I'd like to get #4884 sorted first, along with a standardised "Throttle invalid" propertyChangeEvent.
> 

Consisting is separate from the throttle.  Whether or not the throttle is used in creation of the consist is a system dependent detail.

BTW, this isn't a flow chart.  It is a sequence diagram.

KenC57, 2019-05-01:
You may want to look at the LCC implementation for further clues. Here is the path and basic notes.
-ken c

The traction protocol draft is in https://github.com/openlcb/documents/blob/master/specs/drafts/TractionWN.pdf
The protocol contains support for both of the operating modes: exclusive and cooperative. It is up to the train node to define whether they want exclusive or coop mode. Train nodes are also allowed to have further access restrictions if somebody wants to have a product like that on the market, For example your train node could be configured to only allow your throttle to drive it. This might be helpful at garden railroad meets.

The train nodes I have programmed are operating in coop mode without any access restrictions.
Coop mode has two components:
- the train node accepting drive commands simultaneously from all throttles.
- a listener connection that makes the train node echo all drive commands to all listeners -- this is how the different throttles can update their screen.

The listener connection is the same concept as what is used for consisting. Consisted locos are just listeners to the lead loco, where the lead loco echoes the commands to them.

icklesteve, 2019-05-04:
Thanks for the comments all, have certainly helped my thinking.

Good tips from @pabender , will focus on ThrottleListeners for now so have removed Throttle from the ThrottleListener sequence diagram to focus on that process.

After looking through all of the ThrottleListeners, there was a clear split into 2 types.
The "please just show me the Throttle, I could well be headless" type, eg.
- jmris.AbstractThrottleServer
- jmrit.dispatcher.AutoActiveTrain
- jmrit.logix.Warrant
- jython/Jynstruments/ThrottleWindowToolBar/DCCThrottle.jyn\DCCThrottle.py

and the more exacting Listeners, "I have a developed mechanism for handling steal questions, please leave this decision to me, I might ask the user if I'm not sure" type, eg.

- jmrit.throttle.AddressPanel
- jmrit.withrottle.ThrottleController
- jmrit.withrottle.MultiThrottleController

The please just show me the Throttle ThrottleListener types really shouldn't need to go to the trouble of checking user prefs. so in this updated proposal, would request using
requestThrottle(address,ThrottleListener,**false**)

The more exacting ThrottleListeners would use 
requestThrottle(address,ThrottleListener,**true**)
to say to the ThrottleManager that any steal / share questions should go through them.

I think this is similar to how @bobjacobsen was thinking but without adding the same logic to the majority of ThrottleListeners, the ThrottleManager takes care of that for them.

I've also considered the need for a share / cancel option.
It doesn't make sense to introduce a method for asking "steal / share / cancel" but not "share / cancel".
ThrottleManagers should choose the correct question based on Hardware attributes or operating modes ( ie. majority would use just REQUEST_TYPE.STEAL )

If in future someone needs a hardware specific extension to this, eg "Force Move Slot?" ( ?? ), no extra callback functions are needed and would not disrupt the ThrottleListeners who just want a throttle..

Existing process here : http://jmri.org/JavaDoc/doc/jmri/ThrottleListener.html

Thoughts appreciated,
Steve

[image]

icklesteve, 2019-05-11:
Latest proposal, retaining creation by BasicRosterEntry -
[image]

pabender, 2019-05-11:
Why the distinction about the basic roster entry?

The throttle is either requested by a LocoAddress or a BasicRostEntry.  The throttle just needs to retain what it needs to communicate with the layout hardware.

The front end may need more information, but the backend typically doesn’t.

icklesteve, 2019-05-12:
Good point, 
Have also updated to show the ThrottleListener in the question response
Also added a CANCEL enum to differentiate from a normal cancel request.

[image]

icklesteve, 2019-05-26:
Closed as implemented on #6932

closed 2019-05-26 by icklesteve